MichaIng/DietPi #2091: Error removing libmpdclient2

dynobot, 2018-09-23:
#### Details:
- Date           | Sun 23 Sep 16:09:50 BST 2018
- Bug report     | N/A
- DietPi version | v6.16
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1144 SMP Tue Sep 18 17:34:46 BST 2018
- Distro         | stretch (index=4)
- Command        | dpkg -P mpd libmpdclient2
- Exit code      | 1
- Software title | DietPi-Software

#### Steps to reproduce:

uninstall and purge all selected

#### Actual behaviour:

- ...mpd and mympd show up in the Uninstall software list, selected to uninstall, did not complete, failed.

#### Extra details:

manual uninstall worked...
root@DietPi:~# apt remove libmpdclient2
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages will be REMOVED:
  libmpdclient-dev libmpdclient2
0 upgraded, 0 newly installed, 2 to remove and 0 not upgraded.
After this operation, 420 kB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 27350 files and directories currently installed.)
Removing libmpdclient-dev (2.9-1) ...
Removing libmpdclient2 (2.9-1) ...
root@DietPi:~# 

#### Additional logs:
```
Log file contents:
dpkg: warning: ignoring request to remove mpd which isn't installed
dpkg: dependency problems prevent removal of libmpdclient2:
 libmpdclient-dev depends on libmpdclient2 (= 2.9-1).

dpkg: error processing package libmpdclient2 (--purge):
 dependency problems - not removing
Errors were encountered while processing:
 libmpdclient2
```

MichaIng, 2018-09-23:
@dynobot 
Thanks for your report.

Ah, `dpkg -P` fails when facing dependency issues. Anyway I think we should not use this, but `G_AGP` (apt-get purge) instead, which calls APT with options to purge all dependants as well and skip non-existent packages.

Fourdee, 2018-09-24:
@MichaIng 

Works a charm 👍 

Marking as completed.

closed 2018-09-24 by Fourdee