MichaIng/DietPi #2092: Samba-Client creates file system owned by 'Dietpi' not 'Root'

dynobot, 2018-09-23:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| `6.16`
- Distro version	| 9.4
- Kernel version	|Linux DietPi 4.14.70-v7+ #1144 SMP Tue Sep 18 17:34:46 BST 2018 armv7l GNU/Linux
- SBC device		| RPi 3 Model B (armv7l)
- Power supply used	| (5v
- SDcard used		|patriot 8g
#### Additional Information (if applicable):
- Software title	| (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?
- `dietpi-bugreport` ID

#### Steps to reproduce:

1. ...connect to a network samba share using the config-manager
2. ...ip, name, owner, pw, pw, samba file destination [ok].....

#### Expected behaviour:

- ...Normally it creates a folder owned by root with permissions for all users

#### Actual behaviour:

- ...created a file share with owner as dietpi
[image]

#### Extra details:

- ...

MichaIng, 2018-09-23:
@dynobot 
Thanks for your report.

This is the expected behaviour since v6.16. We changed it to allow other software titles access the samba mount, e.g. music players to access shared music folder, downloaders to move their files into shared download folder and such.

Since root still has full access, this should not cause any issues, or do you face any related issue?

dynobot, 2018-09-23:
Yes the issue I am having is when I use MPD it can not access the files.  MPD runs under user mpd and is only in the group 'audio'. When I run MPD as user dietpi, I have no problems accessing the files -but- myMPD points to user mpd running MPD. If you want to change samba-client to use dietpi instead of root then MPD needs to be in the same group. Looks like dietpi user does not share to 'all users' with permissions. BTW, I never had a problem sharing when root controlled the samba directories....root also had different permissions though, as it allowed read and execute permissions to all-users.

dynobot, 2018-09-23:
Okay that seemed to work. But how about giving dietpi the same permissions root had before?

Read and Execute privileges for all users.

That way all the other programs that might run under some 'other' user won't have this problem....just a thought.

Fourdee, 2018-09-23:
@dynobot 

> Okay that seemed to work. But how about giving dietpi the same permissions root had before?

Would be nice, however, the issue is that all installed software no longer runs as root (minus a few, eg: desktop). This is in an effort to increase "security" for our users.

So to allow access to networked dietpi-userdata, we must use the 'dietpi' group to ensure those programs can function:
https://dietpi.com/phpbb/viewtopic.php?f=11&t=4595#p14192

And samba doesnt support changing individual folder permissions, so this is the best solution we can provide at this time.

dynobot, 2018-09-23:
Okay then I guess dietpi it is...no worries.

Fourdee, 2018-09-23:
Completed, resolved for v6.17.

closed 2018-09-23 by Fourdee