MichaIng/DietPi #2093: v6.16 | Known issues and fixes

Fourdee, 2018-09-23:
## Currently known issue with v6.16 release, which will be fixed for v6.17:
#### Unable to run ```dietpi-config```
- ```[FAILED] DietPi-Config | DEBUG: /var/lib/dietpi/dietpi-wifi.db does not exist, unable to continue. If you did not manually remove this file, please report this issue to DietPi.```

**Fix**: If you see the above error, please see below for solution:
https://github.com/Fourdee/DietPi/issues/2087#issuecomment-423836528

---

#### MyMPD service does not run after installation:
**Fix**: If you see the above error, please see below for solution:
https://github.com/Fourdee/DietPi/issues/2088#issuecomment-423749547

Phil1988, 2018-09-24:
I found an issue with the VMware Image of v6.16.

I am not able to install "VWware Tools" and in addition there are several options not availible in VMware Workstation 14.
For example I cant connect my USB devices to the VM.
(wanted to add an USB HDD for Nextcloud)

Tested on Windows 10 and thought this would be the problem, but also on Windows 7 (both x64) this problem exists.
The image of v6.14 (that I am using) does work perfectly on all tested PCs while the v6.16 has this behaviour.

Hope you have some hints and can work the issue out and how to solve it :)

Fourdee, 2018-09-24:
@Phil1988 

Hi, thanks for the report 👍 

If i'am reading this correctly, VMware tools/support can be installed via APT:
https://www.takistmr.com/index.php/how-to-install-vmware-tools-in-debian-9/
```
G_AGI  open-vm-tools open-vm-tools-desktop
```

This is another method, which allows for the mount of the VMware tools image to install:
https://www.debiantutorials.com/how-to-install-vmware-tools/

---

> For example I cant connect my USB devices to the VM.

Ah yes, this may be my fault. I removed the USB support from the VMware image. I believe @MichaIng must of left this in for previous releases.

You can add the USB hardware support back in, via the hardware options menu for the VM:
[image]

Phil1988, 2018-09-24:
Thanks for your fast reply and thanks for the last point and picture.

On the "old" VMware Image (v6.14) I have this:
[image]

While the new one (v6.16) has only this:
[image]

In words: 
CD/DVD, USB Controller and soundcard are missing.

The missing USB Controller is causing the issue that no USB flash drive is availible and the missing CD/DVD is causing the problem not beeing able to install VMware Tools out of VMware Workstation/Player.

Thanks for you commend and to show me the "Add" function of VMware (I am not that familiar with that software yet).

I just added these and everything is working :)
I would say adding theses "Features" in future VMware images should be better to start for everyone.

Thanks a lot @Fourdee for your great work! (and your help of course)

MichaIng, 2018-09-24:
@Phil1988 
Indeed I also actively removed all non-essential hardware from the VM to keep the image as slim as possible, according to: https://github.com/Fourdee/DietPi/issues/1505

Since, I guess everyone anyway wants to adjust VM hardware, core count, drive space and such, it should be okay to leave it like that. But yeah, quick start and being able to plug&play any USB device directly from VM window, is an argument to keep it.

Hmm we could theoretically pre-install WMware tools on the guests. Enables RTC, shared folder and some other stuff. But I guess they could be incompatible, when e.g. most current VMware Tools are installed, but VM runs on an earlier WMware player version?

MichaIng, 2018-09-25:
@Fourdee
Still an issue on Jessie systems: https://github.com/Fourdee/DietPi/issues/2075

Fourdee, 2018-09-26:
@MichaIng 

> But yeah, quick start and being able to plug&play any USB device directly from VM window, is an argument to keep it.

Yep agree, we'll add the USB support back in by default for next batch of images.

> But I guess they could be incompatible, when e.g. most current VMware Tools are installed, but VM runs on an earlier WMware player version?

It should be fine to install the openvmtools through APT, however, we'd need to test this on the next batch of images.

> Still an issue on Jessie systems: #2075

Yep, i'll set a Jessie system when I can, run some tests.

MichaIng, 2018-09-26:
@Fourdee @Phil1988 
I updated the creation procedure to not remove USB on both, VMware and VBox: https://github.com/Fourdee/DietPi/issues/1505

No hit to remove CD/DVD drive/controller anyway.

I would not mess with pre-installed repo vm tools. Better leave this to end user and repo version can be very outdated. WMware and VBox both support a more less automated install method, that assures a fully compatible/featured version. And some users might not need it.

Phil1988, 2018-09-27:
sounds good to me :)

thanks for taking this so serious :D

MichaIng, 2018-10-02:
@kevkha 
Thanks for report and solution.

Just wanted to fix it, but Fourdee was faster and already fixed it days ago 😄: https://github.com/Fourdee/DietPi/commit/c8dcef0f629c84ecc59558adabdbbad6b52946e6#diff-93b9c3871f032c905ec8b4505d408ca9

kevkha, 2018-10-02:
@MichaIng Thanks. I did code-review and add comments there.

kevkha, 2018-10-02:
@MichaIng Upgrade from 6.14 to 6.16 yields `NULL  | Use dietpi-config to setup a connnection`. This was on LAN adapter eth0 so suspect upgrade did not put the network info back?

MichaIng, 2018-10-02:
@kevkha 
If I see correct above, you have a Stretch image, right? 🤔

Only other idea, what we changed about Ethernet network: https://github.com/Fourdee/DietPi/blob/testing/rootfs/etc/network/if-up.d/dietpi-disable_offload

Could you try to run this command manually?
`/etc/network/if-up.d/dietpi-disable_offload && echo "Exit code: $?"`
- `systemctl status networking` showed error in case this fails, but not sure if this effects actual network connection then.

Fourdee, 2018-10-03:
Locking this thread to keep it clean. 

If you have an issue related (or not) to the above. Please check existing tickets, or, create a new one.

Fourdee, 2018-10-25:
Completed, pending release of v6.17 today.

closed 2018-10-25 by Fourdee