MichaIng/DietPi #2094: Cloudflare dns client ( for dns over https ) and Pihole configuration

breneser, 2018-09-25:
## Creating a software request:

Hi is it possible to add Cloudflare dns client: https://developers.cloudflare.com/argo-tunnel/downloads/
Would be fantastic to be able to make it work with Pihole out of the box

#### Give us some formal software information:
- Software title |  Cloudflare dns over https
- Short description | secure dns over https integrated with pihole?
- GitHub URL (if available) | 
- External URL (if available) | https://developers.cloudflare.com/1.1.1.1/dns-over-https/
- (Official) install documentation | https://developers.cloudflare.com/1.1.1.1/dns-over-https/cloudflared-proxy/

#### Are there similar/alternative software titles available with DietPi-Software?
- No?

#### What makes your requested software better than the above solutions, if available?

- ...

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?
Following url maybe helpful ?
https://scotthelme.co.uk/securing-dns-across-all-of-my-devices-with-pihole-dns-over-https-1-1-1-1/

#### Can you provide the installation steps that you would suggest DietPi-Software to do?
1. Install Cloudflare deamon https://developers.cloudflare.com/argo-tunnel/downloads/
2. Install Pihole
3. Configure pihole

Atavic, 2018-09-28:
Cloudflare is a massive tracking enviroment, but if you want to use it, see:
https://github.com/pi-hole/pi-hole/pull/2075#issuecomment-377965650

breneser, 2018-09-28:
@Atavic Thanks for reminding me good mr 👍 I needed that 🎩

breneser, 2018-09-29:
@Atavic Having checked this blog, they seem to be promising no logging, no tracking? Can I ask what your take on this is ? https://blog.cloudflare.com/announcing-1111/

Atavic, 2018-09-29:
Cloudflare is a bad CDN for the user's privacy:

https://github.com/privacytoolsIO/privacytools.io/issues/374#issuecomment-354477178
https://github.com/MoonchildProductions/UXP/issues/798#issuecomment-425661657
https://github.com/esdiscuss/esdiscuss.org/issues/285
https://github.com/bromite/bromite/issues/98
https://github.com/guardianproject/haven/issues/320#issue-348213762
and https://github.com/ipfs/ipfs/issues/362

You can see how much content comes from Cloudflare with: https://github.com/privacytoolsIO/privacytools.io/issues/518

Related issue: https://github.com/JasonYao/dotfiles/issues/3#issuecomment-425670059

JasonYao, 2018-09-29:
Not sure why @Atavic is using my personal dotfiles as an example here lol- pretty sure he's just a troll that did a search for any issues referencing `1.1.1.1` just to spout off.

Given a choice between Google's DNS where they actively and openly datamine, and Cloudflare's DNS where logs are deleted after 24 hours and anonymized before being sent to APNIC for research purposes, I'd be going with Cloudflare

Atavic, 2018-09-30:
There are other alternatives other than Google and Cloudflare.

MichaIng, 2018-10-05:
dietpi.com uses cloudflare as will for some weeks :thinking:. I think it's true that such is always a trade-off between security and privacy. There might be better solutions, but I would leave such decision to end user. See no reason to not offer that as on option into DietPi.