MichaIng/DietPi #2100: NanoPi Neo Air

troyane, 2018-10-01:
## Creating an image request:

#### Give us some formal device information:
- Device name | FriendlyElec's NanoPi Neo Air (LTS)
- Official product URL | https://www.friendlyarm.com/index.php?route=product/product&path=69&product_id=151
- Official WIKI |  http://wiki.friendlyarm.com/wiki/index.php/NanoPi_NEO_Air
- GitHub resource URL (if available) | https://github.com/friendlyarm
- Image download URL:
  - Official Debian for NAS Dock (nanopi-neo-air_debian-nas-jessie_4.14.52_20180628.img) -- https://drive.google.com/open?id=1Jy7Onvq6e03RZnO2SRri8y9NfilDIF_p
  - Supported Armbian mainline kernel 4.14.y (IMO preferred) -- https://www.armbian.com/nanopi-neo-air/

#### Is the SBC officially supported by the Debian installer?
Not sure.

#### If not, is a reliable 3rd party Debian image available for this SBC?
- Supported Armbian -- https://www.armbian.com/nanopi-neo-air/

#### If not, are there install instructions for Debian available?
Not sure.

Fourdee, 2018-10-01:
@troyane 

Many thanks for the request 👍 

We could use kernel/uboot from ```nanopi-neo-air_sd_friendlycore-xenial_4.14_armhf_YYYYMMDD.img.zip```, and create a Stretch rootFS.

I'll pick this up when I can.

troyane, 2018-10-23:
@Fourdee Is there any help required? I'm ready to help you :)

Fourdee, 2018-10-25:
@troyane 

Apologies for the delay, been busy with v6.17.

> Is there any help required? I'm ready to help you :)

I appreciate the offer, however, you'll need to know how to transfer/replace rootfs to ```nanopi-neo-air_sd_friendlycore-xenial_4.14_armhf_YYYYMMDD.img.zip```, a little tedious 😄 

I'll flag this for v6.18.

troyane, 2018-10-25:
@Fourdee Yesterday I've found this folder https://drive.google.com/drive/folders/1lxLDeJbCDI8dceaimJhjOcyLK5mar_jY -- it looks like file `dietpi_nanopi-h3_20180711.img.7z1` was uploaded a ~week ago. Quick test of that file didn't went well (system stuck on loading). Today I've a got system loaded, here is log [on pastebin](https://pastebin.com/UYAx9Wv8). After few reboots, few tries to set wifi up, it worked. I've seen that it was 
```
[ INFO ] DietPi-Update | Current Version : 6.3
[ INFO ] DietPi-Update | Server Version  : 6.16
```
and I've seen all updates in the middle.

As a result, wifi works and that is great. Will test it further.

Fourdee, 2018-10-28:
@troyane 

Interesting find. Its not an official image from us, however, "if it works" 😄

frostasm, 2018-10-31:
I'm also interested in NanoPi Neo Air support

Fourdee, 2018-11-06:
Hmm, unable to get WiFi working:
- AP6212
```
root@DietPi:~# modprobe -r brcmfmac
root@DietPi:~# modprobe brcmfmac
root@DietPi:~# lsmod
Module                  Size  Used by
brcmfmac              172032  1
brcmutil               16384  1 brcmfmac
nls_ascii              16384  1
rtc_ds1307             24576  0
g_mass_storage         16384  0
ip_tables              24576  0
x_tables               20480  1 ip_tables
root@DietPi:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
root@DietPi:~# dmesg | grep brc
[   69.459762] brcmfmac: brcmf_fw_map_chip_to_name: using brcm/brcmfmac43430-sdio.bin for chip 0x00a9a6(43430) rev 0x000000
[   70.505777] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[   71.538246] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[   72.542199] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[  380.649935] brcmfmac: brcmf_fw_map_chip_to_name: using brcm/brcmfmac43430-sdio.bin for chip 0x00a9a6(43430) rev 0x000000
[  381.683221] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[  382.708675] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[  383.712344] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
```

troyane, 2018-11-06:
@Fourdee just in case, may be you can take a look at unofficial I'm going previously shared? May be there will be answer to your problem? 

Just in case is there any help required? I can test your img or something. 

P.S.: here I've tried to prepare DietPi img, but got strange situation: https://github.com/Fourdee/DietPi/issues/2210#issuecomment-435821836

Fourdee, 2018-11-24:
Our image is now live, completed.

closed 2018-11-24 by Fourdee