MichaIng/DietPi #2101: RoonBridge | Change CPU affinity of "interrupt 62"+"63" for smoother playback

MichaIng, 2018-10-01:
Reference: https://dietpi.com/phpbb/viewtopic.php?f=11&t=4585

I have really no idea about that, just a guess into the blue that the idea is to adjust affinity of those two playback related modules to different CPUs/threads instead of having just one core overloaded.

Looks like these are no processes, where affinity can be applied via `dietpi-process_tool`?

Fourdee, 2018-10-09:
62/63 IRQ appears to be USB bus.
```
root@DietPi:~# cat /proc/interrupts
           CPU0       CPU1       CPU2       CPU3
 62:          0          0          0          0       GIC  62  dwc_otg, dwc_otg                                                                             _hcd:usb2, dwc_otg_pcd
 63: 1447059107          0          0          0       GIC  63  dwc_otg, dwc_otg                                                                             _hcd:usb1
```

🈯️ 
```
root@DietPi:~# echo 4 > /proc/irq/63/smp_affinity
root@DietPi:~# cat  /proc/irq/63/smp_affinity
4
```

Following will have no effect as Odroid C2 only has 4 cores/logical threads:
```
echo 8 > /proc/irq/63/smp_affinity
```

---

Maybe we could add ```echo 4 > /proc/irq/63/smp_affinity``` if USB DAC is selected on Odroid C2.

MichaIng, 2018-10-09:
@Fourdee 
> Maybe we could add echo 4 > /proc/irq/63/smp_affinity if USB DAC is selected on Odroid C2.
Sounds good. This means that it will only use CPU3? Couldn't it be configured to allow usage of all CPUs/threads actually?

Better not mess with is via dietpi-process tool, I think it's okay to solve this for those devices only, where we know it's an issue.

Fourdee, 2018-10-23:
Note to self:
- If USB-DAC, apply this by default.

closed 2018-10-23 by Fourdee