MichaIng/DietPi #2102: DietPi-Drive_Manager | NVMe support/tests

Fourdee, 2018-10-02:
6 partitions?
```
dd if=/dev/zero of=/dev/nvme0n1 bs=4K count=4000

#reboot

nvme0n1      259:0    0 232.9G  0 disk
├─nvme0n1p1  259:1    0     4M  0 part
├─nvme0n1p2  259:2    0     4M  0 part
├─nvme0n1p3  259:3    0    12M  0 part
├─nvme0n1p4  259:4    0    32M  0 part
├─nvme0n1p5  259:5    0    32M  0 part
└─nvme0n1p6  259:6    0 232.5G  0 part
```

MichaIng, 2018-10-02:
Ah this confirms my question in the other topic. I will fix drive_manager accordingly, as currently it might try take `/dev/nvme0` as drive instead of `/dev/nvme0n1`.

Fourdee, 2018-10-02:
@MichaIng 

Just finished, will send commit.

Fourdee, 2018-10-02:
And we need to implement a Random R/W benchmark.
https://twitter.com/DietPi_/status/1047191607248936962

james-b-nz, 2018-10-08:
Hi,
I just installed DietPi on a NanoPC T4 with an Intel NVME 600p SSD. I can confirm that the 6 partitions exist by default for this configuration as well:

```
root@DietPiNanoPCT4:~# lsblk /dev/nvme0n1
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
nvme0n1     259:0    0 119.2G  0 disk
├─nvme0n1p1 259:1    0     4M  0 part
├─nvme0n1p2 259:2    0     4M  0 part
├─nvme0n1p3 259:6    0    12M  0 part
├─nvme0n1p4 259:8    0    32M  0 part
├─nvme0n1p5 259:9    0    32M  0 part
└─nvme0n1p6 259:10   0 118.9G  0 part /mnt/nvme
```

For anyone interested the results in the DietPi benchmark for the above drive is:

[image]

Fourdee, 2018-10-09:
@james-b-nz 

> I just installed DietPi on a NanoPC T4 with an Intel NVME 600p SSD. I can confirm that the 6 partitions exist by default for this configuration as well:

Thank you for testing and confirming 👍 

Ticket now resolved.

Fourdee, 2018-10-10:
Friendly ARM have confirmed this is an issue with the kernel configuration. They have fixed it here:
https://github.com/friendlyarm/kernel-rockchip/commit/a3c93d1b469401211e66fdec965810c1a41087f9

@carlosedp 

Any chance you'd be willing to update the NanoPC-T4 kernel for our images, to include the above fix?

Its not critical, as we have a workaround in DietPi code to compensate for block layout, however, would be nice to have it updated.

carlosedp, 2018-10-11:
Sorry about the delay guys, I'll build the latest Kernel from Rockchip, 4.4.154. Just finishing up some stuff on my build VM.

james-b-nz, 2018-10-12:
So the default kernel config has the Mali driver configured for Android only, I've rebuilt 4.4.143 with the correct options.
Installed the XFCE desktop from dietpi-software and logged in as below:
[image]

I included openvswitch and ip_tables for @carlosedp and similar setups although I was aiming to use it with a GUI. If anyone wants any other modules compiled in let me know and I can update it.

Files for testing with updated boot.img, resource.img, kernel.img and modules in a tar.gz to copy to rootfs: 
https://wetransfer.com/downloads/a358f75410b3caeb9ddf26dce1e9d25820181012002709/d2cb9426e31c00c862f39ccadb62eb2720181012002709/c95aab

Fourdee, 2018-10-16:
Completed.

closed 2018-10-16 by Fourdee