MichaIng/DietPi #2103: WiFi autoreconnect broken

frankdugan3, 2018-10-02:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| 6.16
- Distro version	| stretch
- Kernel version	| Linux DietPi 3.16.57+ #1 SMP PREEMPT Sun Aug 19 15:27:56 CEST 2018 aarch64 GNU/Linux
- SBC device		| Odroid C2 (aarch64)

#### Additional Information (if applicable):
- Locale for WiFi (and everything else) is US (english)
- Latest updates and firmware applied

#### Steps to reproduce:
1. Enable WiFi, add SSID configurations, and eneable autoreconnect

#### Expected behaviour:
- It should be able to connect to the saved SSID configurations

#### Actual behaviour:
- No connection

#### Extra details:
- WiFi was successfully connected until turning on autoreconnect
- `systemctl status dietpi-wifi-monitor` reports the following:

```
● dietpi-wifi-monitor.service - Monitors loss of WiFi connection and automatically reconnects
   Loaded: loaded (/etc/systemd/system/dietpi-wifi-monitor.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2018-10-02 13:32:35 EDT; 4min 59s ago
 Main PID: 636 (dietpi-wifi-mon)
   CGroup: /system.slice/dietpi-wifi-monitor.service
           ├─ 636 /bin/bash /var/lib/dietpi/dietpi-wifi-monitor.sh
           ├─1601 /sbin/wpa_supplicant -s -B -P /run/wpa_supplicant.wlan0.pid -i wlan0 -D nl80211,wext -c /etc/w
pa_supplicant/wpa_supplicant.conf
           ├─1616 /sbin/dhclient -4 -v -pf /run/dhclient.wlan0.pid -lf /var/lib/dhcp/dhclient.wlan0.leases -I -d
f /var/lib/dhcp/dhclient6.wlan0.leases wlan0
           └─1798 ping -I wlan0 -c 1 192.168.86.1

Oct 02 13:37:16 ScrumKiosk-01 dhclient[1605]: No working leases in persistent database - sleeping.
Oct 02 13:37:16 ScrumKiosk-01 bash[636]: No DHCPOFFERS received.
Oct 02 13:37:16 ScrumKiosk-01 bash[636]: No working leases in persistent database - sleeping.
Oct 02 13:37:16 ScrumKiosk-01 bash[636]: /var/lib/dietpi/dietpi-wifi-monitor.sh: line 16: G_DIETPI-NOTIFY: comma
nd not found
Oct 02 13:37:18 ScrumKiosk-01 dhclient[1616]: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 7
Oct 02 13:37:22 ScrumKiosk-01 dhclient[1616]: DHCPREQUEST of 192.168.86.45 on wlan0 to 255.255.255.255 port 67
Oct 02 13:37:22 ScrumKiosk-01 dhclient[1616]: DHCPOFFER of 192.168.86.45 from 192.168.86.1
Oct 02 13:37:23 ScrumKiosk-01 dhclient[1616]: DHCPACK of 192.168.86.45 from 192.168.86.1
Oct 02 13:37:23 ScrumKiosk-01 dhclient[1616]: bound to 192.168.86.45 -- renewal in 39930 seconds.
Oct 02 13:37:26 ScrumKiosk-01 bash[636]: /var/lib/dietpi/dietpi-wifi-monitor.sh: line 8: G_DIETPI-NOTIFY: comman
d not found

```

MichaIng, 2018-10-02:
@frankdugan3 
Thanks for your report.

Hui took a while to even find this service in the code, as I never use WiFi with DietPi 😄.
```
#!/bin/bash
{
URL_PING="$WIFI_GATEWAY"
ADAPTER="wlan$WIFI_DEV_INDEX"
TICKRATE=10
while true
do
G_DIETPI-NOTIFY 2 "Checking connnection for: \$ADAPTER"
ping -I \$ADAPTER -c 1 \$URL_PING
if (( \$? != 0 )); then
G_DIETPI-NOTIFY 2 "Detected connection loss: \$ADAPTER. Reconnecting"
ifdown "\$ADAPTER"
sleep 1
ifup "\$ADAPTER"
G_DIETPI-NOTIFY 0 'Completed'
else
G_DIETPI-NOTIFY 0 "Connection \$ADAPTER"
fi
sleep \$TICKRATE
done
```

> /var/lib/dietpi/dietpi-wifi-monitor.sh: line 16: G_DIETPI-NOTIFY: command not found

As we do not source DietPi-Globals, we cannot use `G_DIETPI-NOTIFY` here. Error our side. Either we need to source globals or use simple `echo`. As output should just go to journal, I would just do echo, also to reduce init time and RAM usage of the service.

Also currently it looks quite spammy, producing output on every tick. I would just create output, if connection was actually lost and after successful reconnection.
@Fourdee 
What do you think?

So but the errors should no break the script. Also your status output shows it is active and running.

Other topic when checking the code:
https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-config#L3451
- Shouldn't this be `$WIFI_DEV_INDEX` here?

https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-config#L2434
- `Network_Restart()` should include stop/start the wifi monitor service, so it does not interfere.

But besides that I do not find a reason WiFi should break and have no DietPi with WiFi device here to test. I hope Fourdee finds time to check.

frankdugan3, 2018-10-02:
@MichaIng Not sure where it breaks, but in the end something in the re-connection script ends up dropping the connection. `ip link` reports `wlan0` as down, and I can't ping anything outside localhost.

MichaIng, 2018-10-02:
@frankdugan3 
Strange, it should only drop WiFi interface, if it could not ping the gateway. Then it should bring up the same interface. So if `$ADAPTER` should somehow be wrong, it should not affect the right interface.

But if `$WIFI_GATEWAY` was wrong, it would be in a connection loop, perhaps breaking after some time.

Can you please do the following:
- Disable WiFi monitor, and make WiFi work.
- Get gateway via: `ip r | grep -m1 'default' | awk '{ print $3 }'`
- Ping that gateway: `ping -I wlan0 -c 1 "$(ip r | grep -m1 'default' | awk '{ print $3 }')"`
- Do you use DHCP or static IP? In case of static IP? In this case try to ping this static IP that you configured: `cat -n /tmp/net_interfaces | grep 'gateway ' | sed -n 2p | awk '{print $3}'`

frankdugan3, 2018-10-02:
OK, here's the full log from the time of login until getting a successful connection. It's long, so I'll comment up here.

I'm using DHCP. The reason my previous log shows connection is because I had it running for a long time before I checked the logs, and it apparently connected eventually.

When I have autoreconnect off, the WiFi connects **immediately** both on changing the settings via `dietpi-config` and on boot. 

The more I dig, the behavior is very strange. It becomes a very long connection process with a lot of false starts and stops when the reconnect script is starting the WiFi connection. It also times out WiFi during boot. Also, if I manually run `ifdown wlan0` and `ifup wlan0`, the behaves the same way as the reconnect script. Perhaps this comes down to my particular WiFi module, but something is different about connecting during boot or `dietpi-config` and simply running `ifup`. 

If I wait long enough, it does seem to randomly connect successfully. The issue is that it's so flaky it randomly drops the connection, and then randomly picks it back up after many false starts.

```
● dietpi-wifi-monitor.service - Monitors loss of WiFi connection and automatically reconnects
   Loaded: loaded (/etc/systemd/system/dietpi-wifi-monitor.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2018-10-02 16:32:33 EDT; 29s ago
 Main PID: 646 (dietpi-wifi-mon)
   CGroup: /system.slice/dietpi-wifi-monitor.service
           ├─ 646 /bin/bash /var/lib/dietpi/dietpi-wifi-monitor.sh
           ├─1423 ifup wlan0
           ├─1439 /sbin/wpa_supplicant -s -B -P /run/wpa_supplicant.wlan0.pid -i wlan0 -D nl80211,wext -c /etc/wpa_supplicant/wpa_supplicant.conf
           ├─1442 /bin/sh -c /sbin/dhclient -4 -v -pf /run/dhclient.wlan0.pid -lf /var/lib/dhcp/dhclient.wlan0.leases -I -df /var/lib/dhcp/dhclient6.wlan0.leases wlan0 
           └─1443 /sbin/dhclient -4 -v -pf /run/dhclient.wlan0.pid -lf /var/lib/dhcp/dhclient.wlan0.leases -I -df /var/lib/dhcp/dhclient6.wlan0.leases wlan0

Oct 02 16:32:58 ScrumKiosk-01 bash[646]: Listening on LPF/wlan0/74:da:38:ee:b3:db
Oct 02 16:32:58 ScrumKiosk-01 bash[646]: Sending on   LPF/wlan0/74:da:38:ee:b3:db
Oct 02 16:32:58 ScrumKiosk-01 bash[646]: Sending on   Socket/fallback
Oct 02 16:32:58 ScrumKiosk-01 bash[646]: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 6
Oct 02 16:32:58 ScrumKiosk-01 dhclient[1443]: Sending on   LPF/wlan0/74:da:38:ee:b3:db
Oct 02 16:32:58 ScrumKiosk-01 dhclient[1443]: Sending on   Socket/fallback
Oct 02 16:32:58 ScrumKiosk-01 dhclient[1443]: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 6
Oct 02 16:32:59 ScrumKiosk-01 wpa_supplicant[1439]: wlan0: SME: Trying to authenticate with 70:3a:cb:1f:96:25 (SSID='Duganators' freq=2462 MHz)
Oct 02 16:33:00 ScrumKiosk-01 wpa_supplicant[1439]: wlan0: SME: Trying to authenticate with 70:3a:cb:1f:94:15 (SSID='Duganators' freq=2462 MHz)
Oct 02 16:33:02 ScrumKiosk-01 wpa_supplicant[1439]: wlan0: SME: Trying to authenticate with 70:3a:cb:1f:96:25 (SSID='Duganators' freq=2462 MHz)
● dietpi-wifi-monitor.service - Monitors loss of WiFi connection and automatically reconnects
   Loaded: loaded (/etc/systemd/system/dietpi-wifi-monitor.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2018-10-02 16:32:33 EDT; 44s ago
 Main PID: 646 (dietpi-wifi-mon)
   CGroup: /system.slice/dietpi-wifi-monitor.service
           ├─ 646 /bin/bash /var/lib/dietpi/dietpi-wifi-monitor.sh
           ├─1439 /sbin/wpa_supplicant -s -B -P /run/wpa_supplicant.wlan0.pid -i wlan0 -D nl80211,wext -c /etc/wpa_supplicant/wpa_supplicant.conf
           ├─1629 /sbin/dhclient -4 -v -pf /run/dhclient.wlan0.pid -lf /var/lib/dhcp/dhclient.wlan0.leases -I -df /var/lib/dhcp/dhclient6.wlan0.leases wlan0
           └─1663 sleep 10

Oct 02 16:33:05 ScrumKiosk-01 wpa_supplicant[1439]: wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="Duganators" auth_failures=1 duration=10 reason=CONN_FAILED
Oct 02 16:33:09 ScrumKiosk-01 dhclient[1443]: No DHCPOFFERS received.
Oct 02 16:33:09 ScrumKiosk-01 bash[646]: No DHCPOFFERS received.
Oct 02 16:33:09 ScrumKiosk-01 bash[646]: No working leases in persistent database - sleeping.
Oct 02 16:33:09 ScrumKiosk-01 dhclient[1443]: No working leases in persistent database - sleeping.
Oct 02 16:33:10 ScrumKiosk-01 bash[646]: /var/lib/dietpi/dietpi-wifi-monitor.sh: line 16: G_DIETPI-NOTIFY: command not found
Oct 02 16:33:13 ScrumKiosk-01 dhclient[1629]: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 7
Oct 02 16:33:16 ScrumKiosk-01 wpa_supplicant[1439]: wlan0: CTRL-EVENT-SSID-REENABLED id=0 ssid="Duganators"
Oct 02 16:33:16 ScrumKiosk-01 wpa_supplicant[1439]: wlan0: SME: Trying to authenticate with 70:3a:cb:1f:94:15 (SSID='Duganators' freq=2462 MHz)
Oct 02 16:33:17 ScrumKiosk-01 wpa_supplicant[1439]: wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="Duganators" auth_failures=2 duration=20 reason=CONN_FAILED
● dietpi-wifi-monitor.service - Monitors loss of WiFi connection and automatically reconnects
   Loaded: loaded (/etc/systemd/system/dietpi-wifi-monitor.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2018-10-02 16:32:33 EDT; 1min 1s ago
 Main PID: 646 (dietpi-wifi-mon)
   CGroup: /system.slice/dietpi-wifi-monitor.service
           ├─ 646 /bin/bash /var/lib/dietpi/dietpi-wifi-monitor.sh
           ├─1711 ifup wlan0
[... cut ...]
Oct 02 16:37:29 ScrumKiosk-01 bash[646]: 64 bytes from 192.168.86.1: icmp_seq=1 ttl=64 time=7.52 ms
Oct 02 16:37:29 ScrumKiosk-01 bash[646]: --- 192.168.86.1 ping statistics ---
Oct 02 16:37:29 ScrumKiosk-01 bash[646]: 1 packets transmitted, 1 received, 0% packet loss, time 0ms
Oct 02 16:37:29 ScrumKiosk-01 bash[646]: rtt min/avg/max/mdev = 7.525/7.525/7.525/0.000 ms
Oct 02 16:37:29 ScrumKiosk-01 bash[646]: /var/lib/dietpi/dietpi-wifi-monitor.sh: line 20: G_DIETPI-NOTIFY: command not found

```

Fourdee, 2018-10-03:
@frankdugan3 

>  line 20: G_DIETPI-NOTIFY: command not found

Yep, needs globals loaded in.

> If I wait long enough, it does seem to randomly connect successfully. The issue is that it's so flaky it randomly drops the connection, and then randomly picks it back up after many false starts.

This appears to be working as intended? If no ping result, the service will drop and restart the WiFi adapter.
Its then down to the adapter to achieve a stable connection, which it has 10 seconds to achieve before next check.

Looks more like a slow DHCP + possible signal strength issue. 
Try using a static IP, under flaky connections it removes the need for DHCP lease/ack.

Also, whats the signal strength (dietpi-config)?

I'll setup my C2 with same adapter and run tests.

frankdugan3, 2018-10-03:
@Fourdee The signal should be pretty strong given that it's 3 ft away from the AP. I haven't had issues with slow DHCP with any other devices, and there's only a few things connecting to the AP. The same image is also being testing in a completely different location with a different brand of AP with another C2 and WiFi module (same model), and it's behaving exactly the same way. An older image with an older version of DietPi does not have this issue on the same hardware and network.

I'm starting to think this is coming down to a WiFi driver issue. @Fourdee are you using the same WiFi adapter as me (Edimax EW-7811Un)?

Fourdee, 2018-10-03:
@frankdugan3 

Yep, testing now.

```
root@DietPi:~# lsusb
Bus 001 Device 003: ID 7392:7811 Edimax Technology Co., Ltd EW-7811Un 802.11n Wireless Adapter [Realtek RTL8188CUS]
```

Fourdee, 2018-10-03:
So the issue appears to be, when both ```eth``` and ```wlan``` are enabled/active, with the wifi connection script:
- Ping from wlan fails:
```
root@DietPi:~# ping -I wlan0 -c 1 google.com
```

Causing the re-connection script to constantly reconnect.

Fourdee, 2018-10-03:
@frankdugan3 

- Are both eth and wlan connections on the same networks? eg: 192.168.0.*
- Do the problems still occur when eth is disabled and system rebooted?

frankdugan3, 2018-10-03:
@Fourdee They are on the same networks. By disabled, do you mean configuration or just pull the plug? It definitely has the issue when eth is unplugged.

Fourdee, 2018-10-03:
@frankdugan3 

> They are on the same networks. 

Thanks.

Interesting, what happens if you run these commands (please paste results):
```
systemctl stop dietpi-wifi-monitor
ifdown wlan0
ifup wlan0
sleep 5
ping -I wlan0 -c 1 $(ip route show 0.0.0.0/0 dev wlan0 | awk '{print $3}')
ip r
```

>  By disabled, do you mean configuration or just pull the plug

```dietpi-config``` > Networking > Eth > disable

frankdugan3, 2018-10-04:
@Fourdee Seems like if I disable eth, it has no problem disconnecting/reconnecting with `ifup wlan0`/`ifdown wlan0`.

The following was run with eth disabled and auto-reconnect enabled, and after a reboot:
```
root@ScrumKiosk-01:~# systemctl stop dietpi-wifi-monitor
root@ScrumKiosk-01:~# ifdown wlan0
Removed stale PID file
Internet Systems Consortium DHCP Client 4.3.5
Copyright 2004-2016 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan0/74:da:38:ee:b3:db
Sending on   LPF/wlan0/74:da:38:ee:b3:db
Sending on   Socket/fallback
DHCPRELEASE on wlan0 to 192.168.86.1 port 67
root@ScrumKiosk-01:~# ifup wlan0
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not supported.
Internet Systems Consortium DHCP Client 4.3.5
Copyright 2004-2016 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan0/74:da:38:ee:b3:db
Sending on   LPF/wlan0/74:da:38:ee:b3:db
Sending on   Socket/fallback
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 7
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 4
DHCPREQUEST of 192.168.86.41 on wlan0 to 255.255.255.255 port 67
DHCPOFFER of 192.168.86.41 from 192.168.86.1
DHCPACK of 192.168.86.41 from 192.168.86.1
bound to 192.168.86.41 -- renewal in 32984 seconds.
root@ScrumKiosk-01:~# sleep 5
root@ScrumKiosk-01:~# ping -I wlan0 -c 1 $(ip route show 0.0.0.0/0 dev wlan0 | awk '{print $3}')
PING 192.168.86.1 (192.168.86.1) from 192.168.86.41 wlan0: 56(84) bytes of data.
64 bytes from 192.168.86.1: icmp_seq=1 ttl=64 time=33.3 ms

--- 192.168.86.1 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 33.339/33.339/33.339/0.000 ms
root@ScrumKiosk-01:~# ip r
default via 192.168.86.1 dev wlan0 
192.168.86.0/24 dev wlan0 proto kernel scope link src 192.168.86.41 
```

Fourdee, 2018-10-04:
@frankdugan3 

> Seems like if I disable eth, it has no problem disconnecting/reconnecting with ifup wlan0/ifdown wlan0.

Many thanks for testing 👍 

Ok, so we need to rule out if this is a hardware specific issue. I'll run the same tests on RPi, see if we can replicate.

If the issue is limited to C2, we can confirm its a hardware/kernel limitation, outside of our control. If the issue is device wide, we'll try to find a fix :)

Fourdee, 2018-10-05:
Ok RPi tests:

#### Same network:
STATIC
- 🈯️ Pings fine
- 🈴 No gateway from ``` ip route```
```
root@DietPi:~# ping -I wlan0 -c 1 192.168.0.1
PING 192.168.0.1 (192.168.0.1) from 192.168.0.28 wlan0: 56(84) bytes of data.
64 bytes from 192.168.0.1: icmp_seq=1 ttl=64 time=4.85 ms

--- 192.168.0.1 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 4.859/4.859/4.859/0.000 ms
root@DietPi:~# ip route show dev wlan0
192.168.0.0/24 proto kernel scope link src 192.168.0.28
```

#### Different networks (hotspot)

Fourdee, 2018-10-05:
So issue appears device wide.

Seems solution is to setup a routing table:
https://askubuntu.com/questions/868942/how-to-configure-2-network-interfaces-with-different-gateways

However, I do not know enough about it, to implement it at this time.

The current solution, is to ensure only 1 hardware network device is enabled. EG: once WiFi is setup, disable the ethernet device via ```dietpi-config```.

frankdugan3, 2018-10-05:
OK, thanks for digging deep into it!

MichaIng, 2018-10-19:
@frankdugan3 @Fourdee 
To go forward with this, perhaps workaround for v6.17:
- It's not (just) `dietpi-wifi-monitor.service` that breaks, but generally WiFi + Ethernet being enabled the same time (Ethernet plugged + both interfaces up), that break WiFi connection then?
- Does the subnet play a role, so only issue if both are on different subnets?
- WiFi Hotspot install is not broken by this? Since for this usually both need to be up on different subnets, right?

As long as we cannot provide a real fix, perhaps then we should workaround:
- If required, when enabling WiFi via `dietpi-config`, disable Ethernet automatically and the other way round.
- If it depends on subnet: When enabling either WiFi or Ethernet, while the other one is already enabled, set subnet of the to-enable device to be the same then the other. Warn user, if subnet is about to be changed, that the other needs to/will be changed as well (if both enabled). But not sure how to solve this in case of DHCP then.

Fourdee, 2018-10-22:
@MichaIng 

> WiFi + Ethernet being enabled 

Yep, the issue is no routing table being applied once setup. So the system has no idea which device to use.
We used to install a package which allowed for both devices at the same time (cant remember name at the moment), however, this broke in Stretch I believe.

WiFi hotspot works, due to routing table being applied.

> If required, when enabling WiFi via dietpi-config, disable Ethernet automatically and the other way round.

This would be the ideal solution, however, will drop current SSH session midpoint, auto reboot?

twikedk, 2018-12-01:
I'm not sure if this is a solution and my test is on a RPi not a Ordroid C2 but according to this: https://www.raspberrypi.org/forums/viewtopic.php?p=1073973#p1072266 dhcpcd5 should be able to handle the situation. I replaced isc-dhcp-client with dhcpcd5 and have done a local test and am able to ping on both adapters. If I have misunderstood the issue (I didn't read into it all :-)) then just ignore my input.

miguelpruivo, 2023-12-29:
@MichaIng I've been using the DietPI with eth only in a particular subnet, but since I use vlans, I would take advantage of the wlan in the pi4 to connect to a different vlan and use that interface to a set of services, but I've been experiencing issues whule connecting (actually, I was never able to connect to wifi _at all_). Not sure if it's because I'm already using eth but the errors I am seeing are these:
```

Listening on LPF/wlan0/dc:a6:32:78:1f:68
Sending on LPF/wlan0/dc:a6:32:78:1f:68
Sending on Socket/fallback
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 4
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 7
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 15
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 16
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 11
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 8
No DHCPOFFERS received.

No working leases in persistent database - sleeping.
```
Thank you.

Joulinar, 2023-12-29:
@miguelpruivo pls stick to your forum post https://dietpi.com/forum/t/cant-connect-to-wifi-with-pi4/18775/3
It doesn't make sense to use this old topic. I doubt your issue is related.