MichaIng/DietPi #2106: Baikal install error

jundar2001, 2018-10-03:
#### Details:
- Date           | Wed  3 Oct 08:09:32 BST 2018
- Bug report     | N/A
- DietPi version | v6.16
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi B+ (armv6l) (index=1)
- Kernel version | #1123 Wed Jun 27 17:05:32 BST 2018
- Distro         | stretch (index=4)
- Command        | Connection test: http://raspbian.raspberrypi.org/raspbian
- Exit code      | 4
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Spider mode enabled. Check if remote file exists.
--2018-10-03 08:00:56--  http://raspbian.raspberrypi.org/raspbian
Resolving raspbian.raspberrypi.org (raspbian.raspberrypi.org)... 93.93.128.193, 2a00:1098:0:80:1000:75:0:3
Connecting to raspbian.raspberrypi.org (raspbian.raspberrypi.org)|93.93.128.193|:80... connected.
HTTP request sent, awaiting response... Read error (Connection timed out) in headers.
Giving up.
```

MichaIng, 2018-10-05:
@jundar2001
Did it work meanwhile?

jundar2001, 2018-10-07:
Tried a reinstall, works perfectly. Thanks for the help!

closed 2018-10-07 by jundar2001