MichaIng/DietPi #2108: PI ZERO NOT SUPPORTED

FuckingFedUp, 2018-10-04:
A: Pi Zero has no WiFi or LAN. You have to use a USB Dongle.
B: There are no USB dongles supported without downloading a driver
C: An internet connection is required to download or install anything
D: Goto A

Also, you need a working DietPi installation to answer your forums registration question.... I can't get my installation working, and can't register for help either!!!

This was the only way I could find to contact you because you hide your email addresses and don't have a contact form.

Any idea when you might fix the impossible scenario with the Pi Zero? The ability to install drivers from the "boot" partition instead of online would be a help.

MichaIng, 2018-10-04:
@FuckingFedUp 
Please check my answer in the forum: https://dietpi.com/phpbb/viewtopic.php?f=11&t=5113

Feature request included, to allow placing a deb package into /boot which will be installed on first run setup automatically.

MichaIng, 2018-10-04:
@F-ck-ngFedUp 
So, lets go on with discussion here, as the driver for at least the second dongle you tried is pre-installed and should work.

Please plug 0BDA:8176 (RTL8188CU/RTL8192CU) for the following and post outputs here:
- First how did you actually configure/test the dongle? Can you list the steps here?
- Please try (after existing hung 1st run setup, when it complains missing network) `/DietPi/dietpi/func/dietpi-set_hardware wificreds set`, open `dietpi-config` > `Network Adapters` and try to configure WiFi there.
- If no device is found, please check `dmesg`, as you mentioned power management failure messages? A WiFi dongle should take really nearly no power, so if the device ran stable with other OS, so you are sure that power supply is sufficient, this should not be the issue.
- But yeah, this leads me to an important message: Did you try the Pi Zero already with a different OS successfully, so you are sure the power supply is reliable sufficient?

F-ck-ngFedUp, 2018-10-04:
> First how did you actually configure/test the dongle? Can you list the steps here?

I connected things so many different ways now I can't remember every combination. Currently I have an official Pi power supply powering the Pi, and the powered USB hub powering everything else, which is currently only the keyboard and WiFi dongle.

I used Etcher to flash the img onto the card. I edited the text file in TextPad via the Windows machine I flashed the card with, and put my network credentials into the first network slot (0).

I tried letting it install, but after the initial changing of passwords etc. it tried to bring the WiFi up and that's where an error appears about power management, which vanishes before I get a chance to read it, followed by the blue screen telling you installation failed (Exit code 4 I think).

I tried this again with the various dongles from the box. They all seem to be RTL chipset and onlt eh RTL8188CU has had any effect. The others don't seem to be detected at all and it just acts as though there's no network dongle.

> Please try (after existing hung 1st run setup, when it complains missing network) /DietPi/dietpi/func/dietpi-set_hardware wificreds set, open dietpi-config > Network Adapters and try to configure WiFi there.

dietpi-config says there's a card available but not connected. Going into any of the options to scan for a network or try to connect looks like it's doing ifdown then ifup which then gives a power management error and dumps me back at the menu.

> If no device is found, please check dmesg, as you mentioned power management failure messages?

Nothing seemingly related in there. I've taken a photo of the errors from the command line when doing ifdown and ifup. https://imgur.com/a/wT1GLma

> Did you try the Pi Zero already with a different OS successfully, so you are sure the power supply is reliable sufficient?

It's the official power supply and it drives the Pi2, Pi3 and PiZero easily (obviously only one at a time).

Fourdee, 2018-10-04:
@F-ck-ngFedUp 

> Nothing seemingly related in there. I've taken a photo of the errors from the command line when doing ifdown and ifup. https://imgur.com/a/wT1GLma

Hi,

Looks like this device is working as intended.

Usually if DHCP fails, due to:
- Incorrect SSID/key entry
- Just outside a stable signal range
- Slow router which DHCP leases (try rebooting router)

If none of the above applies, does you key contain any of the following characters?
> ` " $

F-ck-ngFedUp, 2018-10-04:
The entries are correct, that was the first thing I thought - a mistype. They are absolutely correct though.

The router is 4 feet away. My laptop is further away and reports 92% signal strength if I put the same dongle into my Windows 7 laptop.

The router is fine. I've rebooted it anyway, but the media centre PC, laptop, Smart TV, NAS and my mobile phone are all connected to it and it's giving me a respectable Internet speed. The Smart TV is the far side of the next room but it's happy connecting to YouTube, NetFlix etc. even playing HD video.

Fourdee, 2018-10-04:
@F-ck-ngFedUp 

Very strange, does your WiFi key contain any of the following characters, these can cause issues which we have fixed in our next release?

> ` " $

F-ck-ngFedUp, 2018-10-04:
No, my WiFi key was (as I just changed it) 64FriedTomatoes

F-ck-ngFedUp, 2018-10-04:
I'm just downloading a fresh Raspbian Stretch Lite to see what happens...

F-ck-ngFedUp, 2018-10-04:
No hardware changes at all, even the same MicroSD card - boot up, entered network credentials, running update and it's downloading things already. *shrug*
https://imgur.com/a/RZUcW6l

F-ck-ngFedUp, 2018-10-04:
Just when I was about to give up for the night... I'VE GOT IT WORKING!!

I've also now done this three times, it's totally reproducible.

Installing the first time is complaining about the power management.
Running dietpi-config doesn't work - says connection available, but disconnected. Trying to go into setup crashes out with the power-management error.
Unplug USB hub, replug USB hub
Running dietpi-config now gives a different message about WiFi must be enabled - enable it, it compiles a few things, then connects!

It will only recognise the WiFi properly and enable it if you unplug, then replug, the Hub or the WiFi dongle while the Pi is running. Simply rebooting doesn't work.

Hope this helps somebody else, I've got a bald patch now, lol

MichaIng, 2018-10-05:
@F-ck-ngFedUp 
Okay, so definitely something wrong our end.

Perhaps we should set up a Raspbian system as well and investigate for differences in WiFi/network setup and possibly additional firmware packages. Although couldn't find a related one in Debian repo 🤔.

If Raspbian is still up:
- Could you paste result of: `dpkg --get-selections | grep '^firmware-'`

Went through DietPi-PREP to check what change could be related. Current RPi Stretch image is very fresh, there should be no difference of pre-image and current Raspbian.
- We remove `/etc/profile.d/wifi-country.sh`, but set country code our side in boot script. Still recheck could be beneficial.
`wifi-country.sh` does not even set something, just checks if WiFi country is set, otherwise prints warning:
```
(
        export TEXTDOMAIN=wifi-country

        . gettext.sh

        if [ ! -f /run/wifi-country-unset ]; then
                exit 0
        fi

        if ! /bin/grep -q '^Revision\s*:\s*[ 123][0-9a-fA-F][0-9a-fA-F][0-9a-fA-F]0[dD][0-9a-fA-F]$' /proc/cpuinfo ; then
                exit 0
        fi

        if /sbin/iw reg get | /bin/grep -q "country [A-Z][A-Z]:" ; then
                exit 0
        fi

        echo
        /usr/bin/gettext -s "Wi-Fi is disabled because the country is not set."
        /usr/bin/gettext -s "Use raspi-config to set the country before use."
        echo
)
```

Could not find anything else that might be related. I guess it's something about the WiFi config and/or how it is applied on DietPi vs. Raspbian then...

F-ck-ngFedUp, 2018-10-05:
I wiped the Raspbian image when I re-flashed DietPi.

Could there be something in Raspbian that forces it to re-scan the USB devices? It seems to be removal and reinserting that solves the issue on DietPi - maybe something in raspi-config triggers that when executed? I'm guessing, if this hasn't happened with others, it's something specific to this chipset combination.

mrmajik45, 2018-10-05:
@F-ck-ngFedUp The Pi zero doesn't have LOOPz for your cat picture, "And raspbian was made just for Pis"

Fourdee, 2018-10-07:
@MichaIng 

[raspi-config.zip](https://github.com/Fourdee/DietPi/files/2453649/raspi-config.zip)

- Seems they are using ```wpa_cli``` to set ```wpa_supplicant.conf```
Adds following entry to  ```wpa_supplicant.conf```
```
network={
        ssid="xxxxxx"
        psk="1234567890"
}
```
- Does not support scanning for SSID's, must manually type in SSID.

MichaIng, 2018-10-07:
@F-ck-ngFedUp 
> To be honest, since it started working, I didn't want to mess with the settings again, lol

Ah so the error does not occur on reboot any more, was just present on first boot?
Jep then leave it as it is!

> On Raspbian they've updated raspi-config - the second option is Network Options

Okay, seems I am on an outdated raspi-config repo, since there are not network options inside 😄:
https://github.com/asb/raspi-config/blob/master/raspi-config#L952-L960

Jep this one is the correct one, whoopsie 😅: https://github.com/RPi-Distro/raspi-config

Fourdee, 2018-10-09:
Marking as closed due to:
- Unable to replicate
- User reports issue now resolved
- Added auto back-slashing for SSID saving to ```wifi.db```, as per KEY.

Please reopen if required.

closed 2018-10-09 by Fourdee