MichaIng/DietPi #2110: Process tool problem

roelfoubert, 2018-10-04:
I tried updating transmission but it stops because of an error.
How can I fix this?

/DietPi/dietpi/.dietpi-process_tool: line 11: php-fpm: command not found
/DietPi/dietpi/.dietpi-process_tool: line 12: plexmediaerver: command not found
[ SUB2 ] DietPi-Process_tool > Apply
pgrep: only one pattern can be provided
Try `pgrep --help' for more information.
[FAILED] DietPi-Process_tool | Plex Media Server () : Affinity  0-3
[FAILED] DietPi-Process_tool | An issue has occurred

MichaIng, 2018-10-04:
@roelfoubert 
Thanks for your report.

Please do:
```
rm /DietPi/dietpi/.dietpi-process_tool
dietpi-process_tool 1
```

This should resolve it. Cause is a rework we did on process tool with DietPi v6.8 where we actually reset the above config file, but somehow, most likely due to Linux async disk writes, it did not survive the reboot. Strange that there is no way until now that the kernel forces all held back writes to disk before shutting down 🤔.

MichaIng, 2018-10-05:
Very strange, there are still the wrong entries in `.dietpi-process_tool`.

Could you please paste output of:
`cat /DietPi/dietpi/.dietpi-process_tool`

roelfoubert, 2018-10-05:
This is whatsapp I get

root@RaspberryPi:~# cat /DietPi/dietpi/.dietpi-process_tool
aria2c 0 0-3 SCHED_OTHER 0
avahi-daemon 0 0-3 SCHED_OTHER 0
cron 0 0-3 SCHED_OTHER 0
dropbear 0 0-3 SCHED_OTHER 0
JackettConsole.exe 0 0-3 SCHED_OTHER 0
proftpd 0 0-3 SCHED_OTHER 0
Radarr.exe 0 0-3 SCHED_OTHER 0
NzbDrone.exe 0 0-3 SCHED_OTHER 0
transmission-daemon 0 0-3 SCHED_OTHER 0
lighttpd 0 0-3 SCHED_OTHER 0
php-fpm 0 0-3 SCHED_OTHER 0

MichaIng, 2018-10-05:
@roelfoubert 
Yeah that is the oid style. And if you revive this file and rerun process tool, it looks the same again?

roelfoubert, 2018-10-06:
I always get the same error

roelfoubert, 2018-10-07:
I'm using the latest version 6.16

roelfoubert, 2018-10-07:
changing the line in /DietPi/dietpi/dietpi-process_tool fixed my probem

MichaIng, 2018-10-07:
@roelfoubert 
> changing the line in /DietPi/dietpi/dietpi-process_tool fixed my problem

I am confused now. Maybe there was some disk async reason for `/DietPi/dietpi/.dietpi-process_tool` not being updated, or patch went wrong. However if you face no error messages any more, process tool seems to have successfully overwritten it's config file for fix.
________________
There were two issues:

The fix above should have solved only this part of your posted terminal output:
```
pgrep: only one pattern can be provided
Try `pgrep --help' for more information.
[ OK ] DietPi-Process_tool | Plex Media Server () : Nice 0
[FAILED] DietPi-Process_tool | Plex Media Server () : Affinity 0-3
```
- This is known for v6.16 and fixed with above for v6.17.

The first errors you got:
```
cron: can't lock /var/run/crond.pid, otherpid may be 1478: Resource temporarily unavailable
/DietPi/dietpi/.dietpi-process_tool: line 5: JackettConsole.exe: command not found
2018-10-05 13:21:06,604 proftpd[2331]: fatal: unknown parameter: '0'
/DietPi/dietpi/.dietpi-process_tool: line 7: Radarr.exe: command not found
/DietPi/dietpi/.dietpi-process_tool: line 8: NzbDrone.exe: command not found
...
```
are due to `cat /DietPi/dietpi/.dietpi-process_tool` being in a style that we used some versions ago, but should not look like this on version v6.16, specially not after removing this file and rerunning dietpi-process_tool.
If this errors still appear, I would suggest to rerun `dietpi-update -1`, as something would have gone wrong with the last update.

Fourdee, 2018-10-07:
@MichaIng 

> Whole loop into need to be in double quotes, otherwise if process name matches multiple processes, their IDs will be in one line.

Works fine for me on Stretch:
```
root@DietPi:~# dietpi-process_tool 1
[  OK  ] Root access verified.

 DietPi-Process_tool
─────────────────────────────────────────────────────
 Mode: Apply

[  OK  ] DietPi-Process_tool | Cron (4764) : Nice      0
[  OK  ] DietPi-Process_tool | Cron (4764) : Affinity  0-5
[  OK  ] DietPi-Process_tool | Cron (4764) : Scheduler SCHED_OTHER 0
[  OK  ] DietPi-Process_tool | OpenSSH Server (591) : Nice      0
[  OK  ] DietPi-Process_tool | OpenSSH Server (591) : Affinity  0-5
[  OK  ] DietPi-Process_tool | OpenSSH Server (591) : Scheduler SCHED_OTHER 0
[  OK  ] DietPi-Process_tool | OpenSSH Server (6444) : Nice      0
[  OK  ] DietPi-Process_tool | OpenSSH Server (6444) : Affinity  0-5
[  OK  ] DietPi-Process_tool | OpenSSH Server (6444) : Scheduler SCHED_OTHER 0
[  OK  ] DietPi-Process_tool | Completed
```

```
root@DietPi:~# pgrep ssh
591
6444
```

You running Jessie?

MichaIng, 2018-10-07:
@Fourdee 
Hmm strange, issue appears on Jessie yes, dropbear process. Hmm actually the error makes sense, since without double quotes, all output is in one line, thus should be "591 6444" in this case 🤔.

However, `"$(pgrep -f "${aPROCESS_NAME[$i]}")"` (doubled double quotes) works in all cases. I am already fixing it, just doing some other stuff as well.

Fourdee, 2018-10-07:
@MichaIng 

> "$(pgrep -f "${aPROCESS_NAME[$i]}")"

So it does 👍  could of sworn I tried that earlier and it didn't work (probably not enough coffee!) 🤣. Working fine now :)

MichaIng, 2018-10-07:
Okay PR up: https://github.com/Fourdee/DietPi/pull/2119
- As well I replaced many awk/sed calls with ${varXXX} manipulations. Should speed up the script.

Fourdee, 2018-10-07:
@MichaIng 

Issues are now resolved, i'll mark this as closed.

closed 2018-10-07 by Fourdee