MichaIng/DietPi #2115: Min-swap 128mb

mrmajik45, 2018-10-05:
## Creating a feature request:

#### Is your feature request related to a problem? Please describe:
- Ram shortage

#### Describe the solution you'd like:
- The system has a swap size of 128mb

MichaIng, 2018-10-05:
@mrmajik45
By default DietPi creates a swap file with a size, that the system has 2 GB overall RAM available. But you can easily increase it in dietpi-config or drive_manager.

Besides that I think it's okay, if we leave control of that to end user, what you think?

mrmajik45, 2018-10-05:
Yeah, I didn't see a swap changer so I thought we didn't have one.

mrmajik45, 2018-10-06:
Ok how about this: When the user wants to install a program and there's no space it will warn the user and ask if they want to reduce the swap size and install the program. (Warn that they may have a slowdown in system speed)

MichaIng, 2018-10-06:
@mrmajik45 
Hmm, this is an idea. However I am not sure, if in case you want to install additional software, reducing swap (=available RAM) is a safe suggestion. Free drive space is less than 500M to have `dietpi-software` complaining, where I would always advice to either clean up the system, move parts or whole system to other drive and such.

But maybe we can add this as part of a larger suggestion/error handling menu, if space check fails in general:
Your drive space is insufficient:
- Available: ...
- Needed: ...
You could do the following to increase available drive space:
- ▶️ Clean-up drive via DietPi-Cleaner : Select to open
- ▶️ Move your dietpi_userdata to an external drive : Select to open DietPi-Drive_Manager (only visible, if userdata is located on root partition)
- ▶️ Reduce swap file size : Select to open dietpi-config (only visible, if swapfile is located on root partition)
- ▶️ Other ideas?

mrmajik45, 2018-10-06:
Delete old unused files?