MichaIng/DietPi #2118: Update readme？？

LeeMenHin, 2018-10-07:
about the branch of dev ,beta ,stable

Fourdee, 2018-10-07:
@MichaIng 

> However file name could be updated (BRANCH_SYSTEM.md or something like that?) and/or the included info could be indeed merged into README.md.

Yep 👍 

And we'd need to keep the old file, redirect to new one (as linked in forums etc)

I'll make the change
🈯️ https://github.com/Fourdee/DietPi/blob/dev/BRANCH_SYSTEM.md#steps-to-use-the-dietpi-beta-branch

Fourdee, 2018-10-07:
@LeeMenHin 

As @MichaIng mentioned, we are still developing the beta system for DietPi. We are aiming to release this to the public in the next few days. Once its ready, i'll post on our Twitter page with the info https://twitter.com/DietPi_

I'll mark this as completed.

closed 2018-10-07 by Fourdee

MichaIng, 2018-10-07:
@Fourdee 
We could also search the forum for links and fix them. In case of `TESTING-BRANCH.md` this should not be much, if anything.

MichaIng, 2018-10-10:
We could also add/move those information to the Wiki instead. Perhaps it's better places there instead of being part of the update archive.

Fourdee, 2018-10-11:
@MichaIng 

> We could also add/move those information to the Wiki instead.

If you get the time, go for it 👍