MichaIng/DietPi #2121: General | www-data in sudoers?

MichaIng, 2018-10-08:
@Fourdee 
https://github.com/Fourdee/DietPi/blob/dev/dietpi/func/dietpi-set_software#L320-L327

What for do we need this?
For my impression it's a security issue, especially with NOPASSWD.

Should be never required for www-data to use sudo, besides, Allo GUI needs to access system function?
In case at least we should only add this to Allo GUI images.

Fourdee, 2018-10-09:
@MichaIng 

Agree, needs removing. 

Cant remember the exact reason its been added, however, there would of been a valid reason for its addition.

Lets remove, then debug if a software title/feature fails because of it.

It originally was part of DietPi-Software:
https://github.com/Fourdee/DietPi/commit/ede9c9ca5e48219e58537f07b23c57496e6400b7#diff-75d41fd45b4a2b1cf68db90275ef7394

Fourdee, 2018-10-09:
Patched for removal in v6.17 update.

Completed.

closed 2018-10-09 by Fourdee