MichaIng/DietPi #2122: RE: XU3/4 Autostart Network

DasClarky, 2018-10-08:
## Creating a bug report/issue: Network doesn't come up on start since 6.15.

#### Required Information:
- DietPi version	| 6.14
- Distro version	| `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version	| `uname -a`
- SBC device		| oDroid XU4
- Power supply used	| (EG: 5V 1A RAVpower)
- SDcard used		| (EG: SanDisk ultra)

#### Steps to reproduce:

1. ... Turn the device on after 6.15 
2. ... As above

#### Expected behaviour:

- ... Network should come up on boot as normal

#### Actual behaviour:

- ... Sod all

#### Extra details:

- ... I noticed this on the last version so restored 6.14 and waited for 6.16 as i could see a few people mentioned it, for some reason this was ignored in 6.16...

DasClarky, 2018-10-08:
Made a script to restart it, working now on boot

Fourdee, 2018-10-09:
Reopened for testing.

MichaIng, 2018-10-09:
Should be related to: https://github.com/Fourdee/DietPi/issues/2075

MichaIng, 2018-10-09:
@Fourdee 
Whoopsie found an issue in how we applied the fix:
https://github.com/Fourdee/DietPi/blob/b2d16a6febf35da8be53dea17e5bfe3cff5555de/dietpi/preboot#L255-L257

This is inside: `Apply_DietPi_FirstRun_Settings()`
Of course it needs to be done on every boot.
- Also I will open PR to move this to boot instead of pre-boot, after all other possible network steps are finished.
- `killall -w dhclient &> /dev/null` should have no effect, since dhclient is started together with interfaces, but if it had, it would lead to dhclient not starting at all, since we do not handle it inside `dietpi-services`. @Fourdee do you remember the reason for this?

https://github.com/Fourdee/DietPi/blob/b2d16a6febf35da8be53dea17e5bfe3cff5555de/dietpi/preboot#L270-L279
Not sure if this is not needed any more when running after networking service start. But actually I remember that networking start did not always bring up networks, that are not applied via `auto` (instead of `allow-hotplug` in `/etc/network/interfaces`. Will check what ifup does, if the interface is already up, but might be a good failsafe after networking start.

Fourdee, 2018-10-12:
Unable to replicate on Odroid XU4 with our v6.14 Stretch image, updated to v6.16.

Regardless the network up during boot check added by @MichaIng, will prevent this from occurring in v6.17 for both Jessie and Stretch.

Marking as completed.

closed 2018-10-12 by Fourdee