MichaIng/DietPi #2126: Sickrage fresh install will not start

mdoary, 2018-10-10:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	1.16
- Distro version	stretch or 9.4
- Kernel version	Linux DietPi 4.14.70-v7+ #1144
- SBC device		RPi 3 Model B (armv7l)
- Power supply used	5V 2.5A
- SDcard used		SanDisk ultra)

#### Additional Information (if applicable):
- Software title	Sickrage
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?
- `dietpi-bugreport` ID 49ed237b and 261c817f

#### Steps to reproduce:

1. install latest image 
2. select Sickrage from install software menu 

#### Expected behaviour:

- Should Start with dietpi-service start

#### Actual behaviour:

- Fails to start. 

#### Extra details:

- I am having the same problem on an older setup i was using and create a fresh setup to troubleshoot.

MichaIng, 2018-10-10:
@mdoary 
Thanks for your report.

We will try to replicate.

Fourdee, 2018-10-11:
@MichaIng 

> Capitalization of Sickrage repo changed: https://github.com/SiCKRAGE/SiCKRAGE

👍 

Been thinking we should rename our repo to DiETPi? 😉 😉 🤣

Fourdee, 2018-10-11:
Test:
```
 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Configuring SickRage: automatically download TV shows

[ INFO ] DietPi-Software | Configuring SickRage, please wait...
Job for sickrage.service failed because the control process exited with error code.
See "systemctl status sickrage.service" and "journalctl -xe" for details.
[ INFO ] DietPi-Software | Waiting for SickRage to finish 1st run prep, please wait...
...
...
[FAILED] DietPi-Software | Timed-out waiting for SickRage to generate config.ini
cp: cannot stat '/mnt/dietpi_userdata/sickrage/config.ini': No such file or directory
--2018-10-11 02:35:12--  https://dietpi.com/downloads/conf/sickrage_dietpi_config.ini
Resolving dietpi.com (dietpi.com)... 104.27.179.199, 104.27.178.199, 2606:4700:30::681b:b3c7, ...
Connecting to dietpi.com (dietpi.com)|104.27.179.199|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 23917 (23K)
Saving to: ‘/mnt/dietpi_userdata/sickrage/config.ini’
```

Fourdee, 2018-10-11:
Re-test:
- Install + setup torrent client + online docs check
- 🈯️ ARMv8
- 🈯️ x86_64

Completed. Resolved for v6.17

mdoary, 2018-10-11:
What is the workaround until 6.17?

Fourdee, 2018-10-12:
@mdoary 

> What is the workaround until 6.17?

None currently, due to the vast changes required to install and configure this software. You'll need to wait for v6.17 beta release (ETA few days).

ezekini, 2018-10-14:
Beware of the problems sickrage is having right now. It may or may not be worth it to maintain this software option in Dietpi.....
https://www.reddit.com/r/sickrage/comments/9ll427/echelon_attempting_hostile_takeover_again/

Fourdee, 2018-10-14:
@ezekini 

> Beware of the problems sickrage is having right now. It may or may not be worth it to maintain this software option in Dietpi.....

Not our concern or agenda. The software stays and Sonarr is also a choice 😉 😉

ezekini, 2018-10-15:
Ok... so I updated today to beta v6.17.5 which describes this as fixed in the changelog (great addition to see the changelogs from the terminal), but after trying to install Sickrage, I recieved this error. I don't know if it has to do with a bad previous uninstall (Sorry for the weird format, I couldn't find this error in the logs)

DietPi Error Handler:
 /mnt/dietpi_userdata/sickrage/requirements.txt (line 46)) (2.3.1)
 Requirement already satisfied: enum34; python_version < "3" in
 /usr/lib/python2.7/dist-packages (from
 cryptography>=2.2.1->pyopenssl>=0.12->service_identity==17.0.0->-r
 /mnt/dietpi_userdata/sickrage/requirements.txt (line 46)) (1.1.6)
 Requirement already satisfied: asn1crypto>=0.21.0 in
 /usr/local/lib/python2.7/dist-packages (from
 cryptography>=2.2.1->pyopenssl>=0.12->service_identity==17.0.0->-r
 /mnt/dietpi_userdata/sickrage/requirements.txt (line 46)) (0.24.0)
 pyasn1-modules 0.2.2 has requirement pyasn1<0.5.0,>=0.4.1, but you'll have pyasn1
 0.1.9 which is incompatible.
 Installing collected packages: simplejson, pyasn1-modules, attrs, service-identity,
 psutil
   Found existing installation: simplejson 3.10.0
 Cannot uninstall 'simplejson'. It is a distutils installed project and thus we cannot
 accurately determine which files belong to it which would lead to only a partial
 uninstall.
 You are using pip version 18.0, however version 18.1 is available.
 You should consider upgrading via the 'pip install --upgrade pip' command.

 Unable to continue, DietPi-Software will now terminate.

                Retry         : Re-run the last command that failed
                DietPi-Config : Edit network, APT/NTP mirror settings etc

                        <Ok>                            <Exit>

Fourdee, 2018-10-16:
🈯️ Interesting, fresh install on ASUS TB S, working fine.

Fourdee, 2018-10-16:
@ezekini 

Which SBC device are you getting the error on?

> has to do with a bad previous uninstall 

Was this a manual uninstall you did yourself, or through ```dietpi-software```?

ezekini, 2018-10-16:
Hi @Fourdee. It's an _Odroid C2_. the uninstall/install process was always using `dietpi-software`

Fourdee, 2018-10-17:
@ezekini 

Thanks, 

Try removing the offending packages:
```
pip uninstall simplejson pyasn1
#Say yes to both
```

Then reinstall SiCKRAGE via ```dietpi-software```, see if it passes.

---

🈯️ Confirmed fresh install on C2 is fine, must be limited to users current system.

ezekini, 2018-10-17:
pyasn1 went ok, but the other on returned:  `Cannot uninstall 'simplejson'. It is a distutils installed project and thus we cannot accurately determine which files belong to it which would lead to only a partial uninstall.`

Fourdee, 2018-10-17:
@ezekini 

Feeling the python love 😄 

Ok, will take a look tomorrow:

REF:
- https://github.com/pypa/pip/issues/5247#issuecomment-381550610

Fourdee, 2018-10-23:
@ezekini 

Apologies for the delay. Is this still an issue on your system?

My knowledge of Python is minimal at best, so, to avoid unnecessary time wasted at your end, does the issue still occur on a fresh installation of DietPi using our current Beta with the revised SiCKRAGE install?
https://github.com/Fourdee/DietPi/blob/dev/BRANCH_SYSTEM.md#steps-to-use-the-dietpi-beta-branch

ezekini, 2018-10-23:
@Fourdee , I am a bit busy this week, but I will try a fresh installation whenever I have an hour or so. Will get to you with the results.

Fourdee, 2018-10-23:
@ezekini 

Thanks 👍 

As the original issue in this ticket is resolved, i'll mark this as closed.

@ezekini 

When you come to do the test, if issues still reoccur, please open a new ticket on our github page.

closed 2018-10-23 by Fourdee