MichaIng/DietPi #2130: Xorg failing on XU4 with Stretch image

MichaIng, 2018-10-11:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| `v6.16`
- Distro version	| `Stretch`
- Kernel version	| `uname -a`
- SBC device		| `Odroid XU4`
- Power supply used	| (EG: 5V 1A RAVpower)
- SDcard used		| (EG: SanDisk ultra)

#### Additional Information (if applicable):
- Software title	| Xorg (related MATE desktop and Kodi failing)
- Was the software title installed freshly or updated/migrated? `fresh`
- Can this issue be replicated on a fresh installation of DietPi? `yes`
- `95f9ad89-a855-4288-8f66-0da02081dfeb`

#### Steps to reproduce:

1. Install MATE desktop, or any other Xorg using software
2. `startx`

#### Expected behaviour:

- Desktop/Software starting

#### Actual behaviour:

- Desktop/Software failing with error messages

#### Extra details:

- Via forum: https://dietpi.com/phpbb/viewtopic.php?f=11&t=5127

Fourdee, 2018-10-12:
@MichaIng 

Thanks 👍 

I'll setup a XU4 and run tests for this and https://github.com/Fourdee/DietPi/issues/2122

Fourdee, 2018-10-12:
Confirmed:
```
root@DietPi:~# cat /var/log/Xorg.0.log
[    24.890]
X.Org X Server 1.19.2
Release Date: 2017-03-02
[    24.890] X Protocol Version 11, Revision 0
[    24.890] Build Operating System: Linux 4.9.0-4-armmp-lpae armv7l Debian
[    24.890] Current Operating System: Linux DietPi 4.14.66+ #1 SMP PREEMPT Thu Aug 23 05:59:33 UTC 2018 armv7l
[    24.890] Kernel command line: console=tty1 console=ttySAC2,115200n8 root=UUID=3032ee2a-4efc-40b6-9c1f-2839a840144d rootwait ro fsck.repair=yes net.ifnames=0 drm_kms_helper.edid_firmware=edid/1920x1080.bin smsc95xx.macaddr=00:1e:06:61:7a:72 governor=performance hdmi_tx_amp_lvl=31 hdmi_tx_lvl_ch0=3 hdmi_tx_lvl_ch1=3 hdmi_tx_lvl_ch2=3 hdmi_tx_emp_lvl=6 hdmi_clk_amp_lvl=31 hdmi_tx_res=0 HPD=true vout=hdmi  s5p_mfc.mem=16M
[    24.891] Build Date: 16 October 2017  09:48:39AM
[    24.891] xorg-server 2:1.19.2-1+deb9u2 (https://www.debian.org/support)
[    24.891] Current version of pixman: 0.34.0
[    24.891]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[    24.891] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    24.891] (==) Log file: "/var/log/Xorg.0.log", Time: Fri Oct 12 12:05:27 2018
[    24.896] (==) Using config file: "/etc/X11/xorg.conf"
[    24.896] (==) Using config directory: "/etc/X11/xorg.conf.d"
[    24.896] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    24.897] (==) ServerLayout "ServerLayout0"
[    24.897] (**) |-->Screen "Default Screen" (0)
[    24.897] (**) |   |-->Monitor "<default monitor>"
[    24.898] (**) |   |-->Device "Mali-Fbdev"
[    24.898] (==) No monitor specified for screen "Default Screen".
        Using a default monitor configuration.
[    24.898] (**) Option "BlankTime" "0"
[    24.898] (**) Option "StandbyTime" "0"
[    24.898] (**) Option "SuspendTime" "0"
[    24.898] (**) Option "OffTime" "0"
[    24.898] (==) Automatically adding devices
[    24.898] (==) Automatically enabling devices
[    24.898] (==) Automatically adding GPU devices
[    24.898] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    24.901] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    24.901]    Entry deleted from font path.
[    24.901] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    24.901]    Entry deleted from font path.
[    24.901] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    24.901]    Entry deleted from font path.
[    24.901] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[    24.901]    Entry deleted from font path.
[    24.901] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    24.901]    Entry deleted from font path.
[    24.901] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    24.901]    Entry deleted from font path.
[    24.901] (==) FontPath set to:
        /usr/share/fonts/X11/misc,
        built-ins
[    24.901] (==) ModulePath set to "/usr/lib/xorg/modules"
[    24.901] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[    24.901] (II) Loader magic: 0x660f60
[    24.901] (II) Module ABI versions:
[    24.901]    X.Org ANSI C Emulation: 0.4
[    24.901]    X.Org Video Driver: 23.0
[    24.902]    X.Org XInput driver : 24.1
[    24.902]    X.Org Server Extension : 10.0
[    24.903] (++) using VT number 1

[    24.907] (II) systemd-logind: took control of session /org/freedesktop/login1/session/_31
[    24.908] (II) xfree86: Adding drm device (/dev/dri/card0)
[    24.909] (II) systemd-logind: got fd for /dev/dri/card0 226:0 fd 11 paused 0
[    24.909] (II) no primary bus or device found
[    24.909]    falling back to /sys/devices/platform/exynos-drm/drm/card0
[    24.909] (II) LoadModule: "glx"
[    24.911] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    24.926] (II) Module glx: vendor="X.Org Foundation"
[    24.926]    compiled for 1.19.2, module version = 1.0.0
[    24.926]    ABI class: X.Org Server Extension, version 10.0
[    24.926] (II) LoadModule: "armsoc"
[    24.926] (II) Loading /usr/lib/xorg/modules/drivers/armsoc_drv.so
[    24.928] (II) Module armsoc: vendor="X.Org Foundation"
[    24.928]    compiled for 1.19.2, module version = 1.4.1
[    24.928]    Module class: X.Org Video Driver
[    24.928]    ABI class: X.Org Video Driver, version 23.0
[    24.928] (II) ARMSOC: Driver for ARM Mali compatible chipsets
[    24.928] (WW) Falling back to old probe method for armsoc
[    24.928] (II) No BusID or DriverName specified - opening /dev/dri/card0
[    24.928] (EE) ERROR: Cannot set the DRM interface version.
[    24.928] (EE) ERROR: Cannot open a connection with the DRM - Permission denied
[    24.928] (EE) No devices detected.
[    24.928] (EE)
Fatal server error:
[    24.928] (EE) no screens found(EE)
[    24.928] (EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
[    24.928] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[    24.928] (EE)
[    24.952] (EE) Server terminated with error (1). Closing log file.
```

Fourdee, 2018-10-12:
Ok, appears all drivers are installed correctly.

Tried everything in this thread:
https://forum.odroid.com/viewtopic.php?f=9&t=7101

No joy, ```startx -- :0 vt2``` starts X, but sticks in a loop with mouse flicker and blank screen.

This was working previously: https://github.com/Fourdee/DietPi/issues/1552#issue-299707234

No changes to xorg.conf or installed packages as far as I know. Possible a recent meveric/Xorg package update is causing the issue?

---

Kodi is working 🈯️. Mate specific issue?

elleto, 2018-10-26:
Hi, i am facing the same problem as mentioned above.
however, I found a workaround online guiding me to start MATE desktop.
I deleted xorg.conf and MATE desktop was able to start. Next, I installed KODI from dietpi-software and the xorg unable to start error repeats.

Repeating the same workaround on the generated new xorg.conf, MATE desktop is able to start again but KODI in MATE does not start.
it seems either the XORG.conf is having some problem or that Xserver is having the problem.

Hope this additional information helps.

Fourdee, 2018-10-27:
User reported same issue with C2:
https://dietpi.com/phpbb/viewtopic.php?f=11&t=5188&p=15209#p15209

---

@elleto 

Thanks for the report. Removing the ```xorg.conf``` will allow for desktop usage. Kodi will not function until we can find a solution to the core issue.

Installing Kodi/Desktop in dietpi-software will recreate the ```xorg.conf``` automatically, simply remove it should the issue reappear.

Fourdee, 2018-10-27:
Confirmation XU4:
- 🈯️ Kodi | Appears to revert to software rendering, yet, removal of ```xorg.conf``` breaks it.
```
root@DietPi:~# cat /var/log/Xorg.0.log | grep EE
[    51.909] Current Operating System: Linux DietPi 4.14.66+ #1 SMP PREEMPT Thu Aug 23 05:59:33 UTC 2018 armv7l
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    52.134] (EE) AIGLX error: dlopen of /usr/lib/arm-linux-gnueabihf/dri/armsoc_dri.so failed (/usr/lib/arm-linux-gnueabihf/dri/armsoc_dri.so: cannot open shared object file: No such file or directory)
[    52.134] (EE) AIGLX: reverting to software rendering
```
- 🈴 Desktop
```
root@DietPi:~# cat /var/log/Xorg.0.log | grep EE
[    22.422] Current Operating System: Linux DietPi 4.14.66+ #1 SMP PREEMPT Thu Aug 23 05:59:33 UTC 2018 armv7l
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    22.461] (EE) ERROR: Cannot set the DRM interface version.
[    22.461] (EE) ERROR: Cannot open a connection with the DRM - Permission denied
[    22.461] (EE) No devices detected.
[    22.461] (EE)
[    22.461] (EE) no screens found(EE)
[    22.461] (EE)
[    22.461] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[    22.461] (EE)
[    22.477] (EE) Server terminated with error (1). Closing log file.
```

elleto, 2018-10-29:
Hi Fourdee,
Upon trying more things, only the desktop works when applying the workaround.
everything else that is run through the MATE desktop, like KODI and even firefox does not work.

[image]

according to this image: log that i get when i close the MATE desktop. there seems to be a driver issue.
i tried searching for the folder, but i cannot find it.

hope this helps.

wangcow, 2018-11-06:
Running an ODroid XU4 as well. When running dietpi I get this xorg no screen found issue as well. Tried out debian jessie and stretch, both with the same issues. May be a debian or xorg issue. Guess I'll run arch to see if I get any luck with that

Fourdee, 2018-12-02:
Strange, desktop
- ```startx -- :1``` 🈯️ over SSH
- Fails in autostart 🈴

Fourdee, 2018-12-02:
Resolved in v6.19 in ```login``` script when using ```dietpi-autostart``` under desktop login.

Workaround to launch via local term is:
```
startx &
```

closed 2018-12-02 by Fourdee