MichaIng/DietPi #2134: DietPi for Kirkwood (armv5tel) devices

michelnas, 2018-10-12:
Hi gents,
what a great Job already has been done by all contributers.
DietPi works excellent on my RasPi 2 device.

I wonder, why it is not possible to install it on my old Zyxel NSA310 NAS (CPU Kirkwood armv5tel).
I already installed Debian 9 (Stretch) successfully following the guide from Bodhi in the Doozan-Forum [()](https://forum.doozan.com/read.php?2,12096).

When I then tried to install DietPi with your installer script, I got the message that DietPi can not be installed on armv5tel devices.

I'm pretty sure that many of these old Kirkwood devices could have a chance of a second life if a DietPi installation would be possible.

Please investigate whether an installation might be possible somehow.

Cheers
Michael

michelnas, 2018-10-12:
Hi MichaIng, 
many thanks for your very fast feedback on that topic.
I can completely understand it in terms of supporting "outdated" hardware.
I posted a request of assist at the [Doozan Forum](https://forum.doozan.com/read.php?2,69754).
As I'm just a user (noob) I hope somebody of them can support.

I think that the Kirkwood hardware is strong enough to support most of the DietPi capabilities/opprtunities.

As per my understanding the aim of DietPi is to bring low ressourced hardware working with up-to-date software solutions.
Why not bringing these teams together?

Cheers
Michael

MichaIng, 2020-02-05:
I mark this issue as closed. ToDo list grows faster then we're currently able to solve/implement with current devices/architecture list. Adding a completely new architecture is hence way out of scope. If someone wants to work with it, I'll assist on pull requests and information, but I'm not gonna start working with this myself :wink:.

closed 2020-02-05 by MichaIng