MichaIng/DietPi #2135: G_* functions do (again/still) not run with "sudo -i/-E"

MichaIng, 2018-10-13:
🈴 Recheck:
```
root@VM-Stretch:~# sudo G_CHECK_URL google.com
sudo: G_CHECK_URL: command not found
root@VM-Stretch:~# sudo -i G_CHECK_URL google.com
-bash: G_CHECK_URL: command not found
root@VM-Stretch:~# sudo -E G_CHECK_URL google.com
sudo: G_CHECK_URL: command not found
root@VM-Stretch:~# G_SUDO G_CHECK_URL google.com
-bash: G_CHECK_URL: command not found
```
🈯️ But:
```
root@VM-Stretch:~# sudo -s # or "su" or "sudo -i"
 ───────────────────────────────────────
 DietPi     | 03:02 | Sat 13/10/18
 ───────────────────────────────────────
 v6.17.4    | Virtual Machine (x86_64)
 ───────────────────────────────────────
 eth0       | 192.168.178.29
 ───────────────────────────────────────
 DietPi Team     : Daniel Knight (founder), MichaIng, K-Plan
 Image           : DietPi Core Team (pre-image: Debian netboot)
 Web             : https://DietPi.com | https://twitter.com/dietpi_
 Patreon Legends : PINE64 community
 Donate          : https://DietPi.com/#donate
 DietPi Hosting  : Powered by https://MyVirtualServer.com

 dietpi-launcher  = All the DietPi programs in one place.
 dietpi-config    = Feature rich configuration tool for your device.
 dietpi-software  = Select optimized software for installation.
 htop             = Resource monitor.
 cpu              = Shows CPU information and stats.

root@VM-Stretch:~# G_CHECK_URL google.com
[  OK  ] Root access verified.
[  OK  ] Connection test: google.com
```

```
	G_SUDO(){

		sudo -i $@

	}
```
- `-i` at least is not needed any more, since `dietpi-globals` are loaded now via bashrc instead of profile, thus on every interactive shell, not just login shells. `-i` makes sudo open a login shell (calls profile[.d/], `-s` just a normal interactive one (calls bashrc[.d/] only).
- But it seems a real interactive shell needs to open, which is not the case if a command is invoked directly via `sudo`.

`sudo dietpi-software` etc. btw works fine with `alias sudo='sudo '`, so it's really just about the global G_* functions.

Research:
- Since no interactive shell is opened when a command is invoked directly with `sudo`, `dietpi-globals` are not loaded, and thanks god the `dietpi-banner` does not always prompt, which is the best indicator 😅.
- From within `sudo <command>` just exported variables are available with `-E` option:
```
root@VM-Stretch:~# echo $G_DISTRO
4
root@VM-Stretch:~# sudo bash -c 'echo $G_DISTRO'

root@VM-Stretch:~# sudo -E bash -c 'echo $G_DISTRO'

root@VM-Stretch:~# export G_DISTRO
root@VM-Stretch:~# sudo bash -c 'echo $G_DISTRO'

root@VM-Stretch:~# sudo -E bash -c 'echo $G_DISTRO'
4
```
- So we would need to export G_* functions:
```
root@VM-Stretch:~# export --help
export: export [-fn] [name[=value] ...] or export -p
    Set export attribute for shell variables.

    Marks each NAME for automatic export to the environment of subsequently
    executed commands.  If VALUE is supplied, assign VALUE before exporting.

    Options:
      -f        refer to shell functions
      -n        remove the export property from each NAME
      -p        display a list of all exported variables and functions

    An argument of `--' disables further option processing.

    Exit Status:
    Returns success unless an invalid option is given or NAME is invalid.
root@VM-Stretch:~# G_CHECK_URL google.com
[  OK  ] Connection test: google.com
root@VM-Stretch:~# export -f G_CHECK_URL
root@VM-Stretch:~# sudo -E G_CHECK_URL google.com
sudo: G_CHECK_URL: command not found
```
- Hmm the method does not work... 🤔. Even that `export -fp` clearly shows that `G_CHECK_URL` is exported as function.

Further reading:
- https://stackoverflow.com/questions/9448920/how-can-i-execute-a-bash-function-using-sudo
- https://serverfault.com/questions/177699/how-can-i-execute-a-bash-function-with-sudo/177764#177764

All this appears to be ugly workarounds to me...

Fourdee, 2018-11-25:
@MichaIng 

Is solution here?
```
G_SUDO
```
```
	#DietPi scripts, moved from /etc/bash.bashrc
	# - Sudo command, that ensures profile.d is reloaded (ensures ENV with our globals)
	G_SUDO(){

		sudo -i $@

	}
```

MichaIng, 2019-02-09:
Other approach: `sudo -i; G_CHECK_URL google.com; exit`
🈴 Does not work. `sudo -i` creates a new login shell with DietPi banner, terminal clearing etc. But the other two commands are queued for the parent shell, thus will execute after the sudo shell exits...

Ah:
```
2019-02-09 23:00:49 root@micha:~# sudo -s sleep 1 && G_CHECK_URL dietpi.com
[  OK  ] Connection test: dietpi.com
```
- The command is simply executed too early, before globals have been loaded.
- Obviously the subshell command execution does not wait for profile/bashrc scripts.

Hmm, adding a random sleep time might fail as well on slow devices or when heavy used by parallel processes. Actually better then is to skip interactive shell but only source globals:
```
2019-02-09 23:07:15 root@micha:~# sudo source /DietPi/dietpi/func/dietpi-globals && G_CHECK_URL dietpi.com
sudo: source: command not found
2019-02-09 23:11:35 root@micha:~# sudo . /DietPi/dietpi/func/dietpi-globals && G_CHECK_URL dietpi.com
sudo: .: command not found
```
- Damn, sourcing not possible within sudo 😢.

But in interactive shell:
```
2019-02-09 23:13:12 root@micha:~# sudo -s source /DietPi/dietpi/func/dietpi-globals && G_CHECK_URL dietpi.com
[  OK  ] Connection test: dietpi.com
```

Hmm, globals are sourced then two times/in parallel, but at least they are loaded for sure and usually faster then waiting for a failsafe time of 1 second...

Adding this to `G_SUDO` and print this as only option when G_CHECK_ROOT_USER fails? `sudo -s source /DietPi/dietpi/func/dietpi-globals && <cmd>` is a bid long/confusing as alternative 😉.

MichaIng, 2019-02-24:
Solved as above: https://github.com/Fourdee/DietPi/commit/2cd6ac139156aade8e041b4c5f84e3da23a1f2dd
- Not beautiful but I didn't find any better way to make `sudo -s` waiting for all bashrc/environment scripts being sourced before running the command...

closed 2019-02-24 by MichaIng