MichaIng/DietPi #2140: Open beta v6.17 | Please help to improve stability before next release

MichaIng, 2018-10-14:
We just started public beta testing:
- https://twitter.com/DietPi_/status/1051493658971779072
- https://github.com/Fourdee/DietPi/blob/beta/BRANCH_SYSTEM.md

We hope much of you can help to test v6.17 before we push it to release branch. Note that the new "beta" branch can be considered to be as stable as the prior "master" (release) branch. We internally run the same careful tests before pushing changes from active development into beta branch. So the purpose of the new system is not to outsource in-dev testing, but to further improve release stability and avoid hotfix releases, like sometimes needed in the past.
The vast possible combinations of SBC devices, distro versions, configurations and installed software titles has grown too large to internally test every single of them.

Of course we appreciate, if testers have a special eye on the features we add/change with this release. Check out the current changelog: https://github.com/Fourdee/DietPi/blob/beta/CHANGELOG.txt

If you find any bugs, please open a separate issue, mentioning "beta" in the header, e.g.
`Beta v6.17 | Software title foo fails to install`

Use this thread for general feedback or questions about our new beta program.

Happy testing 🥂!

Millichrome, 2018-10-15:
I will this a go on my test 3B+ board.

MichaIng, 2018-10-15:
@Millichrome 
Thanks for this report.

Indeed I imagine an issue when installing a fresh image and directly switch to beta branch:
- Currently when updating pre-v6.17 to v6.17, `dietpi-update` needs to be rerun manually to apply the new versioning system.
- The new DietPi code is already in place then, but the incremental patches were not yet done. So we manually set version string to v6.16.0 to allow second `dietpi-update` run applying the incremental patches as well.
- When flashing a pre-v6.17 image, on first boot `dietpi-update` will automatically run, but of course the second manual run is still outstanding. Since `dietpi-software` will pop up already and offer configuration/software install, even that incremental patches are missing, this can lead to issues.

@Fourdee 
We need to do a version check on first run after initial `dietpi-update` run. If subversion string is (still) lower than `17`, we either need to automatically do a second update or prompt info to user to do so.
- I would prefer to do that automatically. Actually first run update could simply be a while loop that rechecks available update until none available any more. Only risk is for some reason a unlimited loop, if something goes wrong with version string/var update 🤔.

Millichrome, 2018-10-15:
I see, so it's a purely visual thing. I do remember thinking that the installation process felt a bit more involved than a stable image. :)

MichaIng, 2018-10-15:
@Millichrome 
Jep issues are unlikely, but theoretically possible, especially when you choose several software titles for install on first login and/or the image has a very old DietPi version. Also if I checked to code right, currently a reboot after first run update is missing.
So theoretically code with missing incremental patches can lead to issues during dietpi-software applying initial setup and installs.

But I found a solution I guess, that even allows to skip the manual second update run, when applying beta on fully installed systems. Will play around with this now 😃.

MichaIng, 2018-10-16:
@Millichrome 
As of what I mentioned above and added to OT: Did you try to install via WiFi but master branch first, then apply beta update after first run setup?

It's too late now for me to go through it, but since we did some changes about WiFi with the last two releases, I think the missing reboot after initial update might lead to code inconsistencies.

About second Ethernet attempt:
- Did you use DietPi-Automation actually (configure dietpi.txt to automatically install software on first boot)?
- Do you have a chance to follow the init process on screen? Would be helpful to know which step exactly takes so long.
________
Btw, I finished a PR to apply the second dietpi-update execution automatically: https://github.com/Fourdee/DietPi/pull/2145
This should avoid mentioned potential issues.

I want to finish another visual issue with whiptail menus inside patch_file and maybe we manage to fix one or another bug, listed in OT, before merging next RC into beta.

Millichrome, 2018-10-20:
@MichaIng 

I was able to do WiFi install using the master branch, but not the beta branch. 

The ethernet install was for a clean image. No additional applications or modification other than setting the channel to beta. 

I will try out the 6.17.7 version advertised by the dietpi-update and report back.

Fourdee, 2018-10-22:
Beta in final stages of testing.

Pending 2+ days of additional testing, we'll release v6.17. Outstanding issues without a quick/viable fix will be moved to v6.18 to prevent delaying v6.17 further.

dynobot, 2018-10-25:
Tried to run the beta on Odroid C2 in order to see if the Ethernet issues were resolved.  No luck.
Once after the install and reboot I lost my network connection via Ethernet. Dead in the water now.

dynobot, 2018-10-25:
> Tried to run the beta on Odroid C2 in order to see if the Ethernet issues were resolved. No luck.
> Once after the install and reboot I lost my network connection via Ethernet. Dead in the water now.

Found the problem.....at least for me.

The Ethernet was not broken....it IP address changed !!

I looked on my router and saw a new IP address, I ssh'd into it and behold.....it was the Odroid C2.

Next I tried connecting to wifi using a dongle.....worked fine.

Rebooted and wifi connected no problem
Rebooted again just for kicks and it worked without a hitch.

dynobot, 2018-10-25:
Spoke too soon....wifi worked long enough for me to ssh into the Odroid then it died. It seems to be coming on and off....not only the wifi but the USB ports. I can see my connected DAC turning on and off too.

Fourdee, 2018-10-25:
@dynobot 

> Looks like the Odroid has some power issues concerning its USB ports.

Interesting, if you are already running a 5V/2A PSU (official HK, or, RPi), might be worth investing in a mains powered USB hub, eg:
https://www.amazon.co.uk/Belkin-Ultra-Powered-Power-Supply/dp/B0061RSACG

Also, wen instability occurs, check ```dmesg``` for any errors/issues at the end lines. Might give some more info.

MichaIng, 2018-10-25:
@dynobot 
> IP address changed

Interesting, you receive our IP via DHCP I believe? This is something the DHCP server decides and there is nothing you can do about it on the client, besides changing the MAC address. but this is nothing we do DietPi side. What we did with v6.13 -> v6.14 patch is resetting the device ID, but this is nothing a DHCP server should even be able to recognize (?).

Usually a DHCP server reserves to IP for a certain lease time. If the DHCP client is up the same IP should be extended. So not sure what could have caused a change here. But to be sure you should, if possible, configure your router to assign a fixed IP to the Odroid. In many cases it is as well possible to simply disable the DHCP client and configure the system use a static IP within the DHCP IP range. In my case this works great at least, so one background process less on the machine 🙂.

Fourdee, 2018-10-25:
Thanks to everyone for helping with the Beta test, we really appreciate your support!

Marking this as completed, pending v6.17 release today.

closed 2018-10-25 by Fourdee

MichaIng, 2018-10-25:
NB: The last beta is identical to current master (release), so you don't need to actively change branches and reapply the update.
You will then see an update notification, once we open up the next beta phase. Feel free to re-join next beta phase then by applying the update or switch to master branch to stay with current release and mute beta update notifications.