MichaIng/DietPi #2142: Can't use DietPi with NanoPC-T4

LieDanG, 2018-10-15:
#### Required Information:
- SBC device		|  NanoPC-T4
- Power supply used	|  12V/2A

#### Steps to reproduce:
Downloaded and flashed image from https://github.com/Fourdee/DietPi/issues/1829#issuecomment-398324420 

#### Expected behaviour:
Begin 1st run setup

#### Actual behaviour:
Only get a localhost login screen, tried login with root:dietpi and I get the MOTD from Debian that quickly dissapears and returns again to login screen

#### Extra details:
Flashed the friendlydesktop-bionic image from Friendlyarm and it boots fine to desktop, I was planning to replace my Odroid C2 that I'm currently using with DietPi but I can't figure out why it's not working with my board

MichaIng, 2018-10-15:
@LieDanG 
Thanks for your report. Good to open new issue to bring to attention.

I added v6.17 milestone so we do not forget to check, even it seems an image/kernel issue. Read in the other thread, that we can create a new image based on new available custom kernel.

Fourdee, 2018-10-15:
@LieDanG 

> and I get the MOTD from Debian that quickly dissapears and returns again to login screen

Thanks for the report, very strange, unable to replicate this here.

Could be a bad image write, any errors during write? Which tool are you using to write the images (eg: Android Tool provided in our download).

Regardless, I'am updating this image with new kernel by carlosp, will let you know when its ready for use.

Fourdee, 2018-10-15:
@LieDanG 

Image updated, tested working on my NanoPC-T4, please let me know if this resolves the issue:
https://dietpi.com/downloads/images/DietPi_NanoPCT4-ARMv8-Stretch.7z

LieDanG, 2018-10-15:
There are no errors and I'm writing the image using the Android Tool from your download

This time it says 

Dietpi tty1
DiePi Login:

But it happens the same, login with root:dietpi does nothing it returns again to login screen

MichaIng, 2018-10-15:
@LieDanG 
Hmm, bad that we cannot check autostart index or login script issues, since you cannot access local console easily...

Does login via SSH work, as it looks like you are trying on local console (tty1)?

LieDanG, 2018-10-15:
@MichaIng 
I can't use SSH since the board doesn't get an IP address from the router

It's weird, with the official Friendlyarm image everything works fine, could be a board revision unsupported with current kernel config?

I could try to use it to get some info if that helps

Fourdee, 2018-10-15:
@LieDanG 

> I can't use SSH since the board doesn't get an IP address from the router
> could be a board revision unsupported with current kernel config?

Interesting, could be a different ethernet chipset, however, carlosp uses the latest FriendlyARM kernel, which, should be the same as FriendlyARM image.

Are you able to connect serial, dump the following command once logged in?
```
dmesg
```

LieDanG, 2018-10-15:
I found an old usb serial adapter, this is what I get

`[    3.444881] ohci-platform fe3a0000.usb: Generic Platform OHCI controller
[    3.445644] ohci-platform fe3a0000.usb: new USB bus registered, assigned bus number 6
[    3.446415] ohci-platform fe3a0000.usb: irq 29, io mem 0xfe3a0000
[    3.502488] usb usb6: New USB device found, idVendor=1d6b, idProduct=0001
[    3.503135] usb usb6: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    3.503790] usb usb6: Product: Generic Platform OHCI controller
[    3.504342] usb usb6: Manufacturer: Linux 4.4.154 ohci_hcd
[    3.504838] usb usb6: SerialNumber: fe3a0000.usb
[    3.506629] hub 6-0:1.0: USB hub found
[    3.507055] hub 6-0:1.0: 1 port detected
[    3.508693] input: adc-keys as /devices/platform/adc-keys/input/input3
[    3.511591] dwmmc_rockchip fe310000.dwmmc: IDMAC supports 32-bit address mode.
[    3.512320] dwmmc_rockchip fe310000.dwmmc: Using internal DMA controller.
[    3.512957] dwmmc_rockchip fe310000.dwmmc: Version ID is 270a
[    3.513510] dwmmc_rockchip fe310000.dwmmc: DW MMC controller at irq 25,32 bit host data width,256 deep fifo
[    3.514424] dwmmc_rockchip fe310000.dwmmc: 'clock-freq-min-max' property was deprecated.
[    3.515300] dwmmc_rockchip fe310000.dwmmc: No vmmc regulator found
[    3.515855] dwmmc_rockchip fe310000.dwmmc: No vqmmc regulator found
[    3.516976] dwmmc_rockchip fe310000.dwmmc: allocated mmc-pwrseq
[    3.528502] mmc_host mmc2: Bus speed (slot 0) = 400000Hz (slot req 400000Hz, actual 400000HZ div = 0)
[    3.540261] dwmmc_rockchip fe310000.dwmmc: 1 slots initialized
[    3.542573] rockchip-dmc dmc: Failed to get leakage
[    3.543245] rockchip-dmc dmc: Failed to get pvtm
[    3.545462] rockchip-dmc dmc: could not find power_model node
[    3.562228] asoc-simple-card hdmi-sound: i2s-hifi <-> ff8a0000.i2s mapping ok
[    3.569507] asoc-simple-card spdif-sound: dit-hifi <-> ff870000.spdif mapping ok
[    3.573221]
[    3.573376] =======================================================
[    3.573968] ==== Launching Wi-Fi driver! (Powered by Rockchip) ====
[    3.574531] =======================================================
[    3.575113] Realtek 8188EU USB WiFi driver (Powered by Rockchip,Ver v5.2.2.3_24783.20171030) init.
[    3.575928] [WLAN_RFKILL]: rockchip_wifi_power: 1
[    3.576352] [WLAN_RFKILL]: wifi turn on power. -1
[    3.576772] RTW: module init start
[    3.577104] RTW: rtl8188eu v5.2.2.3_24783.20171030
[    3.577538] [WLAN_RFKILL]: rockchip_wifi_get_oob_irq: Enter
[    3.578321] usbcore: registered new interface driver rtl8188eu
[    3.578838] RTW: module init ret=0
[    3.580642] input: gpio-keys as /devices/platform/gpio-keys/input/input4
[    3.581830] dhd_module_init: in Dongle Host Driver, version 1.579.77.41.2 (r)
[    3.582509] ======== dhd_wlan_init_plat_data ========
[    3.582989] dhd_wlan_init_gpio: GPIO(WL_REG_ON) = -1
[    3.583433] dhd_wlan_init_gpio: GPIO(WL_HOST_WAKE) = -1
[    3.583900] [WLAN_RFKILL]: rockchip_wifi_get_oob_irq: Enter
[    3.584424] dhd_wlan_init_gpio: host_oob_irq: 73
[    3.584845] dhd_wlan_init_gpio: host_oob_irq_flags=0x414
[    3.585354] dhd_wifi_platform_load: Enter
[    3.585746] Power-up adapter 'DHD generic adapter'
[    3.586250] wifi_platform_set_power = 1
[    3.586602] ======== PULL WL_REG_ON(-1) HIGH! ========
[    3.587076] [WLAN_RFKILL]: rockchip_wifi_power: 1
[    3.587496] [WLAN_RFKILL]: wifi turn on power. -1
[    3.589106] mmc2: queuing unknown CIS tuple 0x80 (2 bytes)
[    3.591314] mmc2: queuing unknown CIS tuple 0x80 (3 bytes)
[    3.593484] mmc2: queuing unknown CIS tuple 0x80 (3 bytes)
[    3.597044] mmc2: queuing unknown CIS tuple 0x80 (7 bytes)
[    3.657035] mmc_host mmc2: Bus speed (slot 0) = 150000000Hz (slot req 150000000Hz, actual 150000000HZ div = 0)
[    3.696817] rk808-rtc rk808-rtc: setting system clock to 2013-01-18 09:10:15 UTC (1358500215)
[    3.720057] vcc_sd: disabling
[    3.720729] I : [File] : drivers/gpu/arm/mali400/mali/linux/mali_kernel_linux.c; [Line] : 417; [Func] : mali_module_init(); svn_rev_string_from_arm of this mali_ko is '', rk_ko_ver is '5', built at '07:49:49', on 'Oct 12 2018'.
[    3.723193] Mali: Mali device driver loaded
[    3.724441] ALSA device list:
[    3.724713]   #0: rockchip,hdmi
[    3.725046]   #1: ROCKCHIP,SPDIF
[    3.754712] EXT4-fs (mmcblk1p6): recovery complete
[    3.757441] EXT4-fs (mmcblk1p6): mounted filesystem with ordered data mode. Opts: (null)
[    3.758425] VFS: Mounted root (ext4 filesystem) on device 179:6.
[    3.763195] devtmpfs: mounted
[... 1659 lines cut ...]
[  OK  ] Started Serial Getty on ttyS1.
[    7.721045] random: nonblocking pool is initialized
[    8.421407] cdn-dp fec00000.dp: [drm:cdn_dp_pd_event_work] Not connected. Disabling cdn
[  OK  ] Started Set console font and keymap.
[2:m9 s
        [6:m9 s
                [2raor2
                       [3 e u
                             [9 t s0e

DebianGNU/Lnux 9 Di y0

                      Di  [4:m9 s
                                  [0:m9 s
                                          [1:m9 s
                                                  [5:m9 s
                                                          [0lh0a0
                                                                 [3:m9 s
                                                                         [1:m.(.
 [3:m9 s
         [8:m9 s0
                 [0lh0a4
`

After that I get some random characters but I can't type anything

Fourdee, 2018-10-16:
@LieDanG 

Thanks 👍 

Appears related to display port, I get the same errors.
```
[    3.005191] rockchip-dp ff970000.edp: eDP aining failed (-121)
[    3.008178] rockchip-dp ff970000.edp: Aand reply failed!
[    3.013858] rockchip-dp ff970000.edp: AUX eply failed!
[    3.016666] rockchip-dp ff970000.edp: AUX CH coailed!
[    3.016673] rockchip-dp ff970000.edp: Rx Max Link Rat :ff !
[    3.019520] rockchip-dp ff970000.edp: AUX CH command 
```

Appears the resize script is failing ```dietpi-fs_partition_resize.service```
```
[�[0;32m  OK  �[0m] Reached target Basic System.
         Starting dietpi-fs_partition_resize...
         Starting LSB: Lightweight SSH server...
[�[0;32m  OK  �[0m] Started Daily Cleanup of Temporary Directories.
[�[0;32m  OK  �[0m] Reached target Timers.
         Starting DietPi-RAMlog...
[�[0;1;31mFAILED�[0m] Failed to start dietpi-fs_partition_resize.
See 'systemctl status dietpi-fs_partition_resize.service' for details.
         Starting DietPi-RAMdisk...
[�[0;1;31mFAILED�[0m] Failed to start DietPi-RAMlog.
See 'systemctl status dietpi-ramlog.service' for details.
[�[0;1;31mFAILED�[0m] Failed to start DietPi-RAMdisk.
See 'systemctl status dietpi-ramdisk.service' for details.
[�[0;1;33mDEPEND�[0m] Dependency failed for DietPi-PostBoot.
[�[0;1;33mDEPEND�[0m] Dependency failed for DietPi-Boot.
[�[0;1;33mDEPEND�[0m] Dependency failed for DietPi-PreBoot.
```

Looks like a bad image write. Try the following in android tool:
- Connect the device is MASKROM mode (Hold down the boot and power button's on the NanoPC, for roughly 2-5 seconds.)
- Lower level format
- Run (write the image)

If problems persist:

> After that I get some random characters but I can't type anything, it's from a fresh boot

When this occurs, reconnect the serial cable (USB side), and serial client. Press enter and see if you can log in. If you can log in, check:
```
systemctl status dietpi-fs_partition_resize -l
cat /var/tmp/dietpi/logs/fs_partition_resize.log
```

LieDanG, 2018-10-16:
I can't use the lower leve format option, I get an error

> Match device type failed, stop running

Already tried reconnecting the adapter but can't login

LieDanG, 2018-10-17:
Ok I found the problem

I noticed that the Android Tool by default has the address 0x000B2000 for their rootfs.img but in the  FriendlyDesktop image the parameter.txt file has the address 0x00030000 and I forgot that I had to manually change it when I flashed their image to make it work with my board

I changed the address for the DietPi rootfs and it worked, could you check if that address also works with yours?

Also it's note related with this but I noticed some things with the current DietPi image:

I enabled the shared desktop mode but it doesn't work and it seems that vnc doesn't start with that option as it works fine if it's disabled

The qBitorrent WebUI doesn't seems to work with qbittorrent:dietpi (note sure what are the correct credentials)

Fourdee, 2018-10-17:
@LieDanG 

> I changed the address for the DietPi rootfs and it worked, could you check if that address also works with yours?

Yep, i'll run a test.

> I enabled the shared desktop mode for the vnc but it doesn't work and it seems that vnc doesn't start with that option as it works fine if it's disabled

Thanks, i'll check.

Fourdee, 2018-10-17:
@LieDanG 

For shared desktop mode with VNC to work, a desktop must be running on the system.

🈴 Ah there is an issue with our VNC script under shared mode. Will be resolved for next release.

Fourdee, 2018-10-18:
@LieDanG 

> 0x00030000

This did not work for my system, failed to boot.

As such, i'll mark this as possible info, should any else experience the same issues you did.

LieDanG, 2018-10-20:
@Fourdee 

Did you changed the address both in the Android Tool and in the parameter.txt file?

Fourdee, 2018-10-21:
@LieDanG 

>  parameter.txt

Missed that one, will redo the test 👍

thurloat, 2019-01-15:
I'm running into this same issue in linux as my host, I made the change to parameter.txt but I'm still getting the login loop.

Is there a change that can be made in the config for the `upgrade_tool` as well? I'm not seeing much in terms of docs for that tool. Really hoping to avoid installing windows to do the flashing.

carlosedp, 2019-01-18:
Guys, I'm currently running a Kubernetes Cluster on my NanoPC-T4 boards and while running an ElasticSearch application I found out the board with the 4.4.154 Kernel crashes frequently with Kernel panics. Also the Kernel used by Ambian crashed.

The Kernel 4.4.138 is very stable and I reverted back to it. Also I'm building FriendlyArm 4.4.143 to check if it's stable.

Another improvement I made is that the Kernel partition now has 112MB and can hold multiple Kernel files and provides a boot menu with a newer uboot (have to choose over serial or attached monitor/keyboard). This makes the life easier to test new kernels.

I can provide my files that I used to flash and we can go on from there.

What do you think @MichaIng @Fourdee ?

thurloat, 2019-01-20:
@carlosedp i'd love the files to try out, going to be running a similar setup using Kubernetes and i'd love to avoid the random shutdowns.

LieDanG, 2019-02-13:
@carlosedp do you mind sharing your kernel? It's been a while since the last update for the NanoPC, would like to test that boot menu

carlosedp, 2019-02-14:
Hey guys, sorry for the delay. I packaged the files in a similar flasher that is already in use but I haven't tested flashing with it since I use a Mac and flash on a Linux VM with rkdeveloptool. It might need adjusts.

The download is on https://ufile.io/drlk7 valid for 30 days. Please report back to me if it worked or needed adjusts. The rootfs I used is a little outdated, might need update by @MichaIng or @Fourdee.

MichaIng, 2019-09-02:
Sorry for forgotten response. I think we'll create a new Buster image with current kernel, most likely based on ARMbian Buster with Linux kernel 5.3: https://dl.armbian.com/nanopct4/Debian_buster_dev_minimal_nightly.7z

I mark this issue as closed now. Added the image request to the Buster image collection thread: https://github.com/MichaIng/DietPi/issues/2979

closed 2019-09-02 by MichaIng