MichaIng/DietPi #2144: Beta v6.17 | Autorun 2nd "dietpi-update" to apply v6.17

MichaIng, 2018-10-16:
Ref: https://github.com/Fourdee/DietPi/issues/2140#issuecomment-430005986
- When `v6.17` update runs on first boot, `dietpi-software` only applies update until `patch_file` terminates for manual re-run. Since terminated, `.update_stage` is not updated, thus `dietpi-software` does not know about applied update and does not force `reboot`. Also it offers to configure system, install software and finish first run setup.
- All this might cause issues with half applied v6.17, especially as software is installed with old `dietpi-software` code, and/or many incremental patches are missing due to low DietPi version on image.
- We somehow need to force the 2nd update run and a reboot (after 2nd update), before `dietpi-software` allows to/does install software.

Without doing much hacks on `.update_stage` and or `.install_stage`, somehow force an additional `dietpi-update` run during boot scripts or something, the easiest solution would be an automated update re-run.
_____________________
I experimented around with G_EXIT_CUSTOM trap to somehow call `dietpi-update 1` after terminating 1st dietpi-update instance and patch_file itself.
The problem is, that scripts don't like, if you kill their (grant) parents. Input is broken, leading to ugly errors `_pSLsys_getkey: EIO error`, whiptails does not take input, terminal looks somehow ugly, screen a patchwork of whiptail in background, those error messages and input cursor somewhere in the middle of the screen.
Simple test scripts to replicate this behaviour:
```
root@VM-Stretch:/tmp# cat parent
#!/bin/bash
whiptail --yesno 'Hit "yes" to call child' 12 60 && ./child
root@VM-Stretch:/tmp# cat child
#!/bin/bash
trap './parent' EXIT
kill $PPID
exit
root@VM-Stretch:/tmp# ./parent
```
_______________________
Testing now to simply allow concurrent `dietpi-update` run by `rm -R /tmp/DietPi-Update /tmp/DietPi-Patchfile` first. Then after `dietpi-update 1` has finished inside `patch_file`, prompt reboot question inside and in every case `kill $PPID; exit`.
Branch to test this: https://github.com/Fourdee/DietPi/blob/patch_test/dietpi/patch_file

MichaIng, 2018-10-16:
🈯️ Passed test
[image]
```
root@VM-Stretch:/tmp# dietpi-update -1
[  OK  ] DietPi-Update | Root access verified.
[  OK  ] DietPi-Update | RootFS R/W access verified.

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates

[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/Fourdee/DietPi/patch_test/dietpi/server_v                                                                                                                        ersion-6
[  OK  ] DietPi-Update | Using update server: https://raw.githubusercontent.com/Fourdee/DietPi/patch_test/dietpi/serv                                                                                                                        er_version-6
[  OK  ] DietPi-Update | Repatch was requested: Update will reapply the current subversion patch

[  OK  ] DietPi-Update | Update available
[ INFO ] DietPi-Update | Current version : v6.17.5
[ INFO ] DietPi-Update | Latest version  : v6.17.5
[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=5957 MB | required=100 MB
[  OK  ] DietPi-Update | Update is being applied, please wait...
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Update | wget https://raw.githubusercontent.com/Fourdee/DietPi/patch_test/dietpi/pre-patch_file -O pr                                                                                                                        e-patch_file

 DietPi-Pre-patch
─────────────────────────────────────────────────────
[ INFO ] Applying critical pre-patches

 DietPi-Pre-patch
─────────────────────────────────────────────────────
[ INFO ] Completed pre-patches with exit code: 0

[  OK  ] DietPi-Update | APT update, please wait...
Ign:1 https://cdn-aws.deb.debian.org/debian stretch InRelease
Hit:2 https://cdn-aws.deb.debian.org/debian stretch-updates InRelease
Hit:3 https://cdn-aws.deb.debian.org/debian-security stretch/updates InRelease
Hit:4 https://cdn-aws.deb.debian.org/debian stretch-backports InRelease
Hit:5 https://cdn-aws.deb.debian.org/debian stretch Release
Reading package lists...

[  OK  ] DietPi-Update | G_AGUP
[  OK  ] DietPi-Update | APT upgrade, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

[  OK  ] DietPi-Update | G_AGUG
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   123    0   123    0     0    251      0 --:--:-- --:--:-- --:--:--   252
100  408k    0  408k    0     0   221k      0 --:--:--  0:00:01 --:--:--  335k
[  OK  ] DietPi-Update | Unpack update archive
[  OK  ] DietPi-Update | Copy DietPi core files to RAMdisk
[  OK  ] DietPi-Update | Copy rootfs files in place
[  OK  ] DietPi-Update | Set execute permissions for DietPi scripts
[ SUB1 ] DietPi-Set_software > verify_dietpi.txt ()
[  OK  ] DietPi-Set_software | Connection test: https://raw.githubusercontent.com/Fourdee/DietPi/patch_test/dietpi.tx                                                                                                                        t
[  OK  ] DietPi-Set_software | Patching dietpi.txt
[  OK  ] DietPi-Set_software | Downloading current dietpi.txt
[  OK  ] DietPi-Set_software | dietpi.txt verification completed
[  OK  ] verify_dietpi.txt  | Completed

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Running incremental patching

[ INFO ] DietPi-Update | Current version : v6.17.5
[... 54 lines cut ...]
[  OK  ] DietPi-Set_software | Patching dietpi.txt
[  OK  ] DietPi-Set_software | Downloading current dietpi.txt
[  OK  ] DietPi-Set_software | dietpi.txt verification completed
[  OK  ] verify_dietpi.txt  | Completed
[ SUB2 ] DietPi-Update > Running incremental patching
[ INFO ] DietPi-Update | Current version : v6.16.5
[ INFO ] DietPi-Update | Latest version  : v6.17.5
[ INFO ] DietPi-Update | Patching 6.16.5 to 6.17.5
[  OK  ] DietPi-Patchfile | Desired setting in /root/.ssh/known_hosts was already set: ssh.dietpi.com ssh-rsa AAAAB3N                                                                                                                        zaC1yc2EAAAADAQABAAABAQDE6aw3r6aOEqendNu376iiCHr9tGBIWPgfrLkzjXjEsHGyVSUFNnZt6pftrDeK7UX+qX4FxOwQlugG4fymOHbimRCFiv6c                                                                                                                        f7VpYg1Ednquq9TLb7/cIIbX8a6AuRmX4fjdGuqwmBq3OG7ZksFcYEFKt5U4mAJIaL8hXiM2iXjgY02LqiQY/QWATsHI4ie9ZOnwrQE+Rr6mASN1BVFuI                                                                                                                        gyHIbwX54jsFSnZ/7CdBMkuAd9B8JkxppWVYpYIFHE9oWNfjh/epdK8yv9Oo6r0w5Rb+4qaAc5g+RAaknHeV6Gp75d2lxBdCm5XknKKbGma2+/DfoE8WZ                                                                                                                        TSgzXrYcRlStYN
[  OK  ] DietPi-Software | Initialized database
[  OK  ] DietPi-Software | Reading database completed
[ SUB4 ] DietPi-Software > Automated reinstall
[ INFO ] DietPi-Software | 148: myMPD is not currently installed
[ INFO ] DietPi-Software | The program must be installed, before reinstall can be used
[  OK  ] DietPi-Software | No changes applied for: myMPD
[  OK  ] DietPi-Update | Patching to v6.17.5 completed

[  OK  ] DietPi-Survey | Connection test: ssh.dietpi.com
[  OK  ] DietPi-Survey | Successfully sent survey data
[ SUB2 ] DietPi-Update > Completed
[ INFO ] DietPi-Update | Current version : v6.17.5
[ INFO ] DietPi-Update | Latest version  : v6.17.5
Terminated
root@VM-Stretch:/tmp#
```

MichaIng, 2018-10-16:
Little error:
```
shell-init: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
chdir: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
```

Fourdee, 2018-10-18:
@MichaIng 

Works a charm, great work 👍 

Will mark as completed.

closed 2018-10-18 by Fourdee