MichaIng/DietPi #2153: PineA64 | No sound?

Fourdee, 2018-10-18:
> https://dietpi.com/phpbb/viewtopic.php?f=11&p=15103&sid=5a3203419614f347745ddfcee212bb39#p15103

Fourdee, 2018-10-21:
🈯️ HDMI = hw:1,0
3.5mm = Should be hw:0,0. No sound regardless of unmutes/toggles

Fourdee, 2018-10-21:
Looks like a kernel bug/issue:

```
speaker-test -c2 -D hw:0,0 -r 44100
```

This never changes, always the same values, regardless of unmuted headphone out
```
[  128.577037] HDMI set ptbl: 16 2 0 96 7 5 4 1 4 0 0 128 24 88 44 56 45 1 1
```

Fourdee, 2018-10-21:
REF: https://github.com/ayufan-pine64/linux-build/issues/65#issue-372353454

Fourdee, 2019-01-28:
Will update to ARMbian based image, more stable:
https://github.com/Fourdee/DietPi/issues/2026

MichaIng, 2019-02-24:
Image updated, will mark as closed.

closed 2019-02-24 by MichaIng