MichaIng/DietPi #2154: General | "proc" mount needed inside /etc/fstab?

MichaIng, 2018-10-18:
Just recognized on fresh Debian system that `/etc/fstab` contains the rootfs only, especially no proc mount declaration.

Verified that this is not just since Stretch, but on Jessie it's the same:
```
root@VM-Jessie:~# cat /etc/fstab
#-----------------------------------------------------------
#NETWORK
#-----------------------------------------------------------
#Please use DietPi-Drive_Manager to setup network mounts

#-----------------------------------------------------------
#TMPFS, MISC
#-----------------------------------------------------------
#proc /proc proc defaults 0 0
tmpfs /tmp tmpfs defaults,size=1023M,noatime,nodev,nosuid,mode=1777 0 0
tmpfs                   /DietPi                 tmpfs   defaults,size=10m,noatime,nodev,nosuid,mode=1777        0 0
tmpfs /var/log tmpfs defaults,size=20m,noatime,nodev,nosuid,mode=1777 0 0

#-----------------------------------------------------------
#SWAPFILE
#-----------------------------------------------------------
/var/swap    none    swap    sw    0   0

#-----------------------------------------------------------
#PHYSICAL DRIVES
#-----------------------------------------------------------
UUID=22de6a4b-1a05-4f68-a773-9da30669e494 / auto defaults,noatime,rw 0 1

root@VM-Jessie:~# ls /proc/
1     119   127  154  162  2   239  25   278  31  36   4   49   519  8    94    buddyinfo  cpuinfo    driver       interrupts  kcore       kpageflags  modules       partitions   stat           timer_list   vmallocinfo
10    12    13   156  17   20  24   253  28   32  37   40  5    52   9    95    bus        crypto     execdomains  iomem       keys        loadavg     mounts        sched_debug  swaps          timer_stats  vmstat
1086  121   135  157  18   21  240  254  29   33  372  41  50   6    91   98    cgroups    devices    fb           ioports     key-users   locks       mtrr          self         sys            tty          zoneinfo
1087  122   14   16   181  22  241  26   3    34  38   42  505  7    918  99    cmdline    diskstats  filesystems  irq         kmsg        meminfo     net           slabinfo     sysrq-trigger  uptime
11    1259  15   160  19   23  249  27   30   35  39   43  51   727  93   acpi  consoles   dma        fs           kallsyms    kpagecount  misc        pagetypeinfo  softirqs     sysvipc        version
```

Seems not to hurt, but do we want to clean DietPi fstab a bit, making `#TMPFS, MISC` to `#TMPFS` only? 🙂

Fourdee, 2018-10-20:
@MichaIng 

I took it from the Raspbian fstab, a long long time ago. Will remove it and test.

Fourdee, 2018-10-20:
@MichaIng 

proc is not required, it can removed.

MichaIng, 2018-10-21:
@Fourdee 
You verified that on RPi? Just checked on fresh Raspbian Lite image, and it is still inside:
```
root@VM-Stretch:/mnt/e9646bf0-ef1f-4e8b-983b-c9f97f60e931/etc# cat fstab
proc            /proc           proc    defaults          0       0
PARTUUID=ee25660b-01  /boot           vfat    defaults          0       2
PARTUUID=ee25660b-02  /               ext4    defaults,noatime  0       1
```

Didn't found mich on quick search: https://askubuntu.com/questions/181429/is-a-proc-mount-necessary-in-etc-fstab
But seems indeed a remain from old days, not required and if it works with Debian, then it should definitely work with Raspbian as well.

Fourdee, 2018-10-21:
@MichaIng 

> You verified that on RPi? Just checked on fresh Raspbian Lite image, and it is still inside:

Yep.

> But seems indeed a remain from old days, not required and if it works with Debian, then it should definitely work with Raspbian as well.

Yep, I believe thats the case 👍, will send commit to prevent adding it in future.

MichaIng, 2018-10-21:
Checked a bid about the other mounts, how all this is actually handled in systemd as well:
Systemd actually generates e.g. /run/systemd/generator/tmp.mount based on fstab entries. So theoretically fstab is not required any more, instead systemd mounts could be added to /etc/systemd/system. But of course users expect it to be present, so no advice to remove it, nor does it hurt to use it.

Our fstab /tmp entry leads to:
```
root@VM-Stretch:~# cat /run/systemd/generator/tmp.mount
# Automatically generated by systemd-fstab-generator

[Unit]
SourcePath=/etc/fstab
Documentation=man:fstab(5) man:systemd-fstab-generator(8)
Before=local-fs.target

[Mount]
What=tmpfs
Where=/tmp
Type=tmpfs
Options=defaults,size=1023M,noatime,nodev,nosuid,mode=1777
```
The Debian Stretch default is:
```
root@VM-Stretch:~# cat /usr/share/systemd/tmp.mount
#  This file is part of systemd.
#
#  systemd is free software; you can redistribute it and/or modify it
#  under the terms of the GNU Lesser General Public License as published by
#  the Free Software Foundation; either version 2.1 of the License, or
#  (at your option) any later version.

[Unit]
Description=Temporary Directory
Documentation=man:hier(7)
Documentation=http://www.freedesktop.org/wiki/Software/systemd/APIFileSystems
ConditionPathIsSymbolicLink=!/tmp
DefaultDependencies=no
Conflicts=umount.target
Before=local-fs.target umount.target
After=swap.target

[Mount]
What=tmpfs
Where=/tmp
Type=tmpfs
Options=mode=1777,strictatime,nosuid,nodev

[Install]
WantedBy=local-fs.target
```
- On Jessie it was pre-installed in /lib/systemd/system/tmp.mount, but due to some unknown reason it got auto-enabled there in some setups, which was not wanted, thus the mount was moved to the share folder with Stretch: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=787542#74
- So no `defaults`, `DefaultDependencies=no` should lead to it being loaded a bid earlier. In case of /tmp this is not bad, I think.
- `strictatime` is moreless the oposite of noatime, forcing atime updates in every case. Since it is in RAM, should not slow down much, but I would still leave `noatime`, as long as it is not required for anything.

Generally using `<mnt-point>.mount` files instead of fstab allows to define some more conditions, checks, attachment to systemd targets etc. The `swap` entry at least by default is attached to swap.target, so no need here. Just something to keep in mind, in case required in the future or we simply want to more consequently use the native systemd methods.

Found as well a new option for mounts: https://kernelnewbies.org/Linux_4.0#A.22lazytime.22_option_for_better_update_of_file_timestamps
- `lazytime` if for some reason/software access times are required.
- Since this also affects modified and changed times, we could add it by default to our mounts. However we need to check for Linux kernel 4.X first, e.g. `uname -r`, but I remember the output though devices was not very consistent... 🤔.

Fourdee, 2018-10-21:
@MichaIng 

> lazytime if for some reason/software access times are required.

Interesting, but might cause more issues than its worth:

> This behaviour, however, breaks some programs that rely on accurate access time tracking to work, and it's also against the POSIX standard.

Not something we should add by default I believe?

MichaIng, 2018-10-21:
@Fourdee 
> > This behaviour, however, breaks some programs that rely on accurate access time tracking to work, and it's also against the POSIX standard.
> 
> Not something we should add by default I believe?

This is about the `noatime`, which we already add by default. The `lazytime` alone seems to be safe, was created as reaction to many users use `noatime` 😄.

Since we use `noatime` for a very long time, it seems to be required in rare cases. But we could theoretically remove it (`relatime` is default then) and use `lazytime` to keep disk I/O reduced. Still, minimal additional writes to RAM + on `sync` and every 24h to fisk (on file access) will be overall result 🤔.

Fourdee, 2018-10-21:
@MichaIng 

Lazytime still has to write the access times to disk as some point? If so, this will vastly increase write cycles and reduce flash storage lifespan compared to noatime.

I'd be more inclined to leave noatime as it is?

MichaIng, 2018-10-21:
@Fourdee 
They do not exclude each other:
- `noatime` disables access time completely while
- `lazytime` reduces change/modified timestamps to be done less often. Although I wondering a bid, if the `**if the inode needs to be updated anyway**` condition is not anyway true on every file write, at least if not prevented by `async`. But right now the timestamp on disk is as well not updated, if the file change still kept in cache due to async I believe. Everything else would be nonsense...

Fourdee, 2018-10-21:
@MichaIng 

> lazytime reduces change/modified timestamps to be done less often

But also enables access times again? Or ```noatime``` can be used with ```lazytime```? 

> https://lwn.net/Articles/620086/
Add a new mount option which enables a new "lazytime" mode.  This mode
causes atime, mtime, and ctime updates to only be made to the
in-memory version of the inode.  The on-disk times will only get
updated when (a) when the inode table block for the inode needs to be
updated for some non-time related change involving any inode in the
block, (b) if userspace calls fsync(), or (c) the refcount on an
undeleted inode goes to zero (in most cases, when the last file
descriptor assoicated with the inode is closed).

Fourdee, 2018-10-22:
Original ticket issue now resolved. Will mark as closed.

```lazytime``` new ticket if noatime can be used with lazytime.

closed 2018-10-22 by Fourdee