MichaIng/DietPi #2157: Add open-iscsi

dezren39, 2018-10-21:
open-iscsi is one of the most common iscsi initiators. I plan to hook some pi's up to a nas running a raid array using iscsi for block level access, and so it'd be convenient to install from configuration. I'll likely setup a post-install install script that includes it for now. It would be a useful addition to nfs samba ftp already available.

dezren39, 2018-10-21:
Actually, an iscsi target wouldn't be a bad idea either.

Fourdee, 2018-10-21:
@dezren39 

Many thanks for the request 👍 

We'll look into this when possible.

lwqcz, 2019-11-27:
@MichaIng I've just finished the manual setup of my iSCSI initiator and I could drop some lines to `dietpi-software` if this feature request issue is still valid and unresolved.

MichaIng, 2019-11-27:
@lwqcz 
Would be great. If you find time to raise some PR, I'll assist quickly to get it read for v6.28 (Aim to release v6.27 in a few days).

This might help: https://github.com/MichaIng/DietPi/wiki/How-to-add-a-new-software-title