MichaIng/DietPi #2158: Official 7'' Touchdisplay autostart resolution failure

potter-91, 2018-10-21:
Dietpi (6.16)     PI3B+

Installed Chromium and chose it at autostart.
The resolution (1290x720) is to big for the Display (800x480). See pictures ^^
Could it be possible that the dietpi-autostart chromium windowsize-flag use the resolution which is chosen at dietpi-config -> Display Octions?

[image]
[image]

Greetings from Germany^^

Fourdee, 2018-10-21:
@potter-91 

> Could it be possible that the dietpi-autostart chromium windowsize-flag use the resolution which is chosen at dietpi-config -> Display Octions?

Yep, great idea, we can apply to Chromium if installed, when setting display resolution.

Fourdee, 2018-10-21:
For RPi, we'll auto match current resolution, dietpi.txt entry.
For other devices, users can change the dietpi.txt entries. As, this would require a vast rewrite to support all devices via above method.

Fourdee, 2018-10-21:
Ordered a 7inch touchscreen HDMI, will do some testing when it arrives.

Fourdee, 2018-10-22:
@potter-91 

Fixed in v6.17, you can test this now using our Beta program:
https://github.com/Fourdee/DietPi/blob/dev/BRANCH_SYSTEM.md#steps-to-use-the-dietpi-beta-branch

To apply the new settings, after patch, please re-select the RPi touch screen from ```dietpi-config```.

potter-91, 2018-10-22:
I've got another idea:
A integration or your own script to change the touchscreens brightness and power.
I use this: https://github.com/linusg/rpi-backlight

Chromium now works perfect.
Could be "unclutter" also a good idea to chose it in dietpi.software or -config to hide the mouse?

Fourdee, 2018-10-22:
Notes:
- hide mouse https://stackoverflow.com/a/39518893
- Brightness https://raspberrypi.stackexchange.com/a/46229

@potter-91 
We should be able to do both in bash script, however, i'll need you to test it (lack the device, went for a cheaper IPS screen with higher resolution 😄)

Leave it with me, will let you know when its available in beta to test.

Fourdee, 2018-10-23:
This ticket is completed.

Will open a new ticket for feature requests:
https://github.com/Fourdee/DietPi/issues/2169

closed 2018-10-23 by Fourdee