MichaIng/DietPi #2165: DietPi Virtualbox VM Root Filesystem

louisefindlay23, 2018-10-22:
Using the DietPi .ova file, I can't allocate more than 8 GBs to the root filesystem so I get out of space errors.

MichaIng, 2018-10-22:
You need to adjust it in the virtual machines settings, where you can also adjust RAM, CPU count (e.g. match your host machines CPU/thread count) network etc.

louisefindlay23, 2018-10-22:
I don't see where because any new storage device I add is a different drive and where the option is to alter the DietPi_VirtualBox drive size.

[image]

MichaIng, 2018-10-22:
Ah sorry I was mistaken, it's in the virtual drive manager, so not the VM tab of the GUI. Switch at the top right, there you see a tree structure of the virtual drives and when selecting one way the bottom you can adjust their sizes.
I am not at home currently, still post some screens later, if you didn't find it yet.

louisefindlay23, 2018-10-22:
I ended up converting it to .vdi by this [guide](https://stackoverflow.com/questions/11659005/how-to-resize-a-virtualbox-vmdk-file) which then let me resize it and then dietpi-drive_manager let me resize the drive. Thanks for your help

closed 2018-10-22 by louisefindlay23