MichaIng/DietPi #2169: RPi Touchscreen | Support brightness change + auto hide mouse in X11

Fourdee, 2018-10-23:
> https://github.com/Fourdee/DietPi/issues/2158#issuecomment-431978189

potter-91, 2018-10-23:
Another idea
A new section in dietpi-software, named Chromium default page (or something else)
There you could choose which site would be the start page of chromium.
And you could choose between the dietpi-optimized-softwares and a custum way.

Atavic, 2018-10-23:
But the homepage is set inside chromium.

Fourdee, 2018-10-23:
@potter-91 

Currently, you can set the homepage by changing ```URL=${CHROMIUM_URL:-"http://google.com"}``` in:
```
/var/lib/dietpi/dietpi-software/installed/chromium-autostart.sh
```

But not a bad idea for us to add an option to set this during autostart 👍

Fourdee, 2018-11-25:
Completed.

Autohide option will require a new ticket + enough user interest to warrant our implantation.

closed 2018-11-25 by Fourdee