MichaIng/DietPi #2171: Unable to install openssh

tcalil, 2018-10-24:
#### Details:
- Date           | Wed 24 Oct 08:36:14 BST 2018
- Bug report     | N/A
- DietPi version | v6.16
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | #1145 SMP Fri Sep 21 15:38:35 BST 2018
- Distro         | stretch (index=4)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. launch dietpi-software
2. change ssh server to openssh instead of dropbear

#### Expected behaviour:

- installation should complete

#### Actual behaviour:

- error messages and is unable to continue

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
2 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up openssh-server (1:7.4p1-10+deb9u4) ...
Job for ssh.service failed because the control process exited with error code.
See "systemctl status ssh.service" and "journalctl -xe" for details.
invoke-rc.d: initscript ssh, action "start" failed.

MichaIng, 2018-10-24:
@tcalil 
Thanks for your report. Hmm, I am thinking if we either stop dropbear service in this case or uninstall it, before attempting to install openssh. Since otherwise port 22 is blocked, ssh.service must fail to attach to it.

To resolve it now:
- Of course note that you need to do this from local terminal (instead of SSH session).
- apt purge -y dropbear-run
- rerun dietpi-software to choose openssh
__________
How to resolve this within dietpi-software our side:
- APT install must fail, as long as existing SSH server is at least running.
- So we must uninstall or stop existing SSH server process before attempting to install the new one.
- Stopping SSH server leads to breaking SSH session.
- Thus either we need to only allow this in case of current session is no SSH session (e.g. `[[ $SSH_TTY ]]` check), or implement a oneshot service, that does the switch on next reboot, in case of active SSH session.

Or do we have a chance to force install while ignoring service start errors? Theoretically by ignoring APT error, expecting it is due to above and the next reboot (after auto uninstall of existing SSH server) will resolve it?

Fourdee, 2018-10-24:
@MichaIng @tcalil 

I've never experienced this error using SSH to make the change. @tcalil are you doing this on local system with keyboard+HDMI?

MichaIng, 2019-02-27:
Mark as closed, due to lack of user reply. Feel free to reopen, if issue persists.

closed 2019-02-27 by MichaIng