MichaIng/DietPi #2173: No sound output except in Roon.

ezekini, 2018-10-24:
## Creating a bug report/issue:
#### Required Information:
- DietPi version: 6.17.9
- Distro version: 9.4
- Kernel version: Linux DietPi 4.14.71-v7+ #1145 SMP Fri Sep 21 15:38:35 BST 2018 armv7l GNU/Linux
- SBC device: RPi 3 Model B+ (armv7l)
- Power supply used: 5V@2A
- SDcard used: SanDisk ultra

#### Additional Information (if applicable):
- Software title: Shairport, Raspotify, Roon bridge.
- Was the software title installed freshly or updated/migrated? Yes
- Can this issue be replicated on a fresh installation of DietPi? I guess

#### Steps to reproduce:

1. Install raspotify, Roon bridge, shairport.

#### Expected behaviour:

- Sound should output from _rPI Allo Boss Dac_.

#### Actual behaviour:

I can only get an output using Roon.

#### Extra details:

I have selected Allo Boss Dac in dietpi-config. What is strange is that if I enter alsamixer, I have to manually select the soundcard, which, to my memory, wasn't neccesary in the past.

Fourdee, 2018-10-24:
@ezekini 

Thanks for the report, lets check a few things, please paste results:
```
cat /etc/asound.conf
aplay -l
```

ezekini, 2018-10-24:
Here are results @Fourdee :

```
root@DietPi:~# cat /etc/asound.conf
pcm.!default {
    type hw
    card 0
    device 0
}
ctl.!default {
    type hw
    card 0
}
root@DietPi:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi [vc4-hdmi], device 0: MAI PCM vc4-hdmi-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: BossDAC [BossDAC], device 0: Boss DAC HiFi [Master] pcm512x-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
root@DietPi:~#
```

MichaIng, 2018-10-25:
@Fourdee
Is reboot needed to (re)load a kernel module? Perhaps it is possible to use modprobe for this?

Fourdee, 2018-10-25:
@MichaIng 

> Is reboot needed to (re)load a kernel module? Perhaps it is possible to use modprobe for this?

Issue is with dt overlay, which needs to be loaded during boot time via kernel. Can't be done on the fly as far as i'am aware ☹️ 
https://www.raspberrypi.org/forums/viewtopic.php?t=99401

MichaIng, 2018-10-25:
@Fourdee 
https://www.raspberrypi.org/documentation/configuration/device-tree.md

Very interesting on general, according to changing dtoverlays on runtime level: `3.5.1 THE DTOVERLAY COMMAND`

Worth to try to keep configuration handy.

MichaIng, 2019-01-05:
@Fourdee 
Yep, basically this should work:
```
# Check if desired overlay is available
dtoverlay -a | grep 'target_overlay'
# Load desired overlay to running system
dtoverlay 'target_overlay'
# Remove old overlay
dtpverlay -r 'old_overlay'
```
Source: https://www.raspberrypi.org/documentation/configuration/device-tree.md#part3.5.1

Fourdee, 2019-06-27:
@MichaIng 

Applied a simple workaround for now, if vc4, sound card target index++.

Only downside, user will need to re-select sound card after changing vc4 on/off. msg?

MichaIng, 2019-06-27:
@Fourdee 
Is good for now. I planned to rewrite our sound card selection menu that way:
Main menu:
1. enable/disable audio capability (if disabled, purge alsa-utils, hide below entries)
2. Allow to select additional dtoverlay/kernel module to load (only shown on HW models that support/require this, e.g. RPi)
3. Select sound card+device based on `aplay -l` auto detection.
4. DietPi-Justboom as it was
5. Optionally, see if we can come up with a reliable option: Select microphone (capture device) based on `arecord -l`.

Btw. regarding mails some days ago: I found a solution for the Mycroft + microphone issue with default (non equaliser) asound.conf: There is an option to switch capture sample rate within mycroft.conf, that is by default 16000 (does not work) but works well with 48000. This was the case on all my x86 devices (all VMs with different sound cards) and different notebook. So it seems the "plug" module allows this sample rate conversion or something but that totally breaks sound quality.
With equaliser enabled things are broken by error message I wrote you, about pa_lib_alsa (or similar). Looks like in that case Mycroft looks for ("pa" = ?) PulseAudio which is not available, thus fails? However this is all about Mycroft and not about our sound settings, so nothing to change in this direction.

closed 2019-07-03 by MichaIng