MichaIng/DietPi #2182: can't install sonarr

omar93, 2018-10-25:
#### Details:
- Date           | Thu 25 Oct 18:07:31 BST 2018
- Bug report     | N/A
- DietPi version | v6.16
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 2 Model B (armv7l) (index=2)
- Kernel version | #1144 SMP Tue Sep 18 17:34:46 BST 2018
- Distro         | stretch (index=4)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1)dietpi-software
2)optimized-software
3)check sonarr
4)install

#### Expected behaviour:

sonarr should install and work

#### Actual behaviour:
i get an error saying it could not and tells me to report the issue here

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
E: Could not get lock /var/lib/dpkg/lock - open (11: Resource temporarily unavailable)
E: Unable to lock the administration directory (/var/lib/dpkg/), is another process using it?
```

MichaIng, 2018-10-25:
@omar93 
Thanks for your report.

> ```
> Log file contents:
> E: Could not get lock /var/lib/dpkg/lock - open (11: Resource temporarily unavailable)
> E: Unable to lock the administration directory (/var/lib/dpkg/), is another process using it?
> ```

This error usually shows up, if APT is called while already running.

Could you please check via `htop` if there is any other script running APT concurrently. Otherwise try to manually run `apt-get -y upgrade`, if this works.

omar93, 2018-10-25:
[image]

this is what i get

omar93, 2018-10-25:
the problem was fixed by your command thx

closed 2018-10-25 by omar93