MichaIng/DietPi #2183: Greg - A Podcast manager

wallace11, 2018-10-25:
## Creating a software request:

#### Give us some formal software information:
- Software title | Greg
- Short description | A command-line podcast aggregator (or manager)
- GitHub URL (if available) | https://github.com/manolomartinez/greg
- (Official) install documentation | https://github.com/manolomartinez/greg

#### Are there similar/alternative software titles available with DietPi-Software?
No.

#### What makes your requested software better than the above solutions, if available?
- It is fully open source, without any (feature) limitation to free users.
- There's currently no Podcast apps available for the DietPi.

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?
- It probably won't, actually, even though there may be some problems with Python versions etc.

#### Can you provide the installation steps that you would suggest DietPi-Software to do?
Nothing out of the ordinary. Maybe it should require `python-pip` package?

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.
If there's a base for that I could do some changes as needed along the way. It's a Python program so there shouldn't be anything special needed anyway.

MichaIng, 2018-12-26:
No development on greg for more than two years. Isn't there a solution that is still in active development?

wallace11, 2018-12-26:
Wow, Greg has been serving me so well I haven't noticed that...

MichaIng, 2018-12-26:
@wallace11 
It is not necessarily an knockout argument, especially for using the software. More problems usually appear when building the software, due to outdated function calls, old library dependencies and such. There are already some open issues and PRs about obsolete python3 calls and such 🤔.

However I added it to our new FeatHub page, let's see it there is more interest for it: https://feathub.com/Fourdee/DietPi/+27

MichaIng, 2019-02-27:
No votes so far or down == up. Will mark as closed for now due to general cleanup. However the FeatHub request stays open and as fast as it reaches closer to the top, we reopen this issue to further work on it.

closed 2019-02-27 by MichaIng