MichaIng/DietPi #2184: Crash while installing NextCloud

webtroter, 2018-10-25:
#### Details:
- Date           | Fri 26 Oct 00:55:52 BST 2018
- Bug report     | N/A
- DietPi version | v6.17.12 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1145 SMP Fri Sep 21 15:38:35 BST 2018
- Distro         | stretch (index=4)
- Command        | ncc maintenance:install
- Exit code      | 1
- Software title | DietPi-Software

#### Steps to reproduce:
I formated a USB key, moved the user dir, then tried to install nextcloud only.

#### Expected behaviour:
Nextcloud to install

#### Actual behaviour:
it crash after those console lines : 
```
Enabling dietpi-nextcloud: ok
Run "service lighttpd force-reload" to enable changes
Enabling dietpi-dav_redirect: ok
Run "service lighttpd force-reload" to enable changes
[ INFO ] DietPi-Software | Enable 4-byte support for MariaDB
[  OK  ] DietPi-Software | systemctl restart mysql
[FAILED] DietPi-Software | ncc maintenance:install
```

#### Additional logs:
```
Log file contents:
An unhandled exception has been thrown:
Error: Class 'Memcached' not found in /var/www/nextcloud/config/config.php:1156
Stack trace:
#0 /var/www/nextcloud/lib/private/Config.php(214): include()
#1 /var/www/nextcloud/lib/private/Config.php(63): OC\Config->readData()
#2 /var/www/nextcloud/lib/base.php(140): OC\Config->__construct('/var/www/nextcl...')
#3 /var/www/nextcloud/lib/base.php(591): OC::initPaths()
#4 /var/www/nextcloud/lib/base.php(1068): OC::init()
#5 /var/www/nextcloud/console.php(46): require_once('/var/www/nextcl...')
#6 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#7 {main}
```

webtroter, 2018-10-26:
After more tests and tries :

A failed NextCloud installation attempt breaks DietPi-Services.

I tried with a freshly flashed SD card on a RasPi 3B. I can even install owncloud before and have it working.

webtroter, 2018-10-26:
Is there a way to roll back the DietPi stack on a previous version?

MichaIng, 2018-10-26:
@webtroter 
Thanks for your report.

Could you please send a bugreport: `dietpi-bugreport 1`
And paste the content of your config.php: `cat /var/www/nextcloud/config/config.php`

> Error: Class 'Memcached' not found in

🤔 Hmm, we configure APCu and Redis, not Memcached, was this a fresh Nextcloud install or did you migrate from another instance/server?

MichaIng, 2018-10-26:
```
'memcache.distributed' => '\OC\Memcache\Memcached',
...
/**
 * Server details for one or more memcached servers to use for memory caching.
 */
'memcached_servers' => array(
        // hostname, port and optional weight. Also see:
        // http://www.php.net/manual/en/memcached.addservers.php
        // http://www.php.net/manual/en/memcached.addserver.php
        array('localhost', 11211),
        //array('other.host.local', 11211),
),

/**
 * Connection options for memcached, see http://apprize.info/php/scaling/15.html
 */
'memcached_options' => array(
        // Set timeouts to 50ms
        \Memcached::OPT_CONNECT_TIMEOUT => 50,
        \Memcached::OPT_RETRY_TIMEOUT =>   50,
        \Memcached::OPT_SEND_TIMEOUT =>    50,
        \Memcached::OPT_RECV_TIMEOUT =>    50,
        \Memcached::OPT_POLL_TIMEOUT =>    50,

        // Enable compression
        \Memcached::OPT_COMPRESSION =>          true,

        // Turn on consistent hashing
        \Memcached::OPT_LIBKETAMA_COMPATIBLE => true,

        // Enable Binary Protocol
        \Memcached::OPT_BINARY_PROTOCOL =>      true,

        // Binary serializer vill be enabled if the igbinary PECL module is available
        //\Memcached::OPT_SERIALIZER => \Memcached::SERIALIZER_IGBINARY,
),
```
This must fail, if `php-memcached` is not installed. And we don't want it, since we don't need it, since we don't have a distributed server usually. And if, Redis still makes more sense...
Also the other pre-existing entries break our install and configuration effort.
```
/**
 * This example shows how to configure Nextcloud to store all files in a
 * swift object storage.
 *
 * It is important to note that Nextcloud in object store mode will expect
 * exclusive access to the object store container because it only stores the
 * binary data for each file. The metadata is currently kept in the local
 * database for performance reasons.
 *
 * WARNING: The current implementation is incompatible with any app that uses
 * direct file IO and circumvents our virtual filesystem. That includes
 * Encryption and Gallery. Gallery will store thumbnails directly in the
 * filesystem and encryption will cause severe overhead because key files need
 * to be fetched in addition to any requested file.
 *
 * One way to test is applying for a trystack account at http://trystack.org/
 */
'objectstore' => [
        'class' => 'OC\\Files\\ObjectStore\\Swift',
        'arguments' => [
                // trystack will use your facebook id as the user name
                'username' => 'facebook100000123456789',
                // in the trystack dashboard go to user -> settings -> API Password to
                // generate a password
                'password' => 'Secr3tPaSSWoRdt7',
                // must already exist in the objectstore, name can be different
                'container' => 'nextcloud',
                // prefix to prepend to the fileid, default is 'oid:urn:'
                'objectPrefix' => 'oid:urn:',
                // create the container if it does not exist. default is false
                'autocreate' => true,
                // required, dev-/trystack defaults to 'RegionOne'
                'region' => 'RegionOne',
                // The Identity / Keystone endpoint
                'url' => 'http://8.21.28.222:5000/v2.0',
                // required on dev-/trystack
                'tenantName' => 'facebook100000123456789',
                // dev-/trystack uses swift by default, the lib defaults to 'cloudFiles'
                // if omitted
                'serviceName' => 'swift',
                // The Interface / url Type, optional
                'urlType' => 'internal'
        ],
],
```
This is an `example` in how and what you can configure, but no working default file. This and many following entries must fail, if nothing related set up before.

Trying to remove this wrong default file before attempting install.

Twilight0, 2018-10-26:
Where is this file located? I do not understand. What does it have to do with nextcloud installation?

webtroter, 2018-10-26:
This was a fresh install on a fresh SD card. On a rpi 3B. I will get back
with the info you ask, it might take me some time since I would have to
reflash everything.

MichaIng, 2018-10-26:
Okay guys, at least a simple solution:

We need to remove the (why ever) wrong placed example config.php before attempting the Nextcloud install command:
`sed -i '/G_ERROR_HANDLER_COMMAND=\'ncc maintenance:install\'/a\rm $config_php' /DietPi/dietpi/dietpi-software`

To stop ncc maintenance mode related error messages, remove existing nextcloud/occ file:
`rm /var/www/nextcloud/occ`

Then rerun the dietpi-software to install Nextcloud. Short command: `dietpi-software install 114`

MichaIng, 2018-10-26:
Okay found the reason: https://github.com/Fourdee/DietPi/commit/e0f4c883611cb413d0be9c4669981cf4ca584acf#r31060077

Accidental wrong fix in this case.

So the better fix is: `sed -i '\|cp \'/var/www/nextcloud/config/config.sample.php\'|d' /DietPi/dietpi/dietpi-software`

webtroter, 2018-10-26:
the sed line seems not complete. Bash asks me to close something, and I don't really know sed.
[image]

webtroter, 2018-10-26:
I could not reinstall NextCloud on the Pi. I will reflash and retry later tonight

Since the sed query didn't work, I simply removed the line in the dietpi-software script.

MichaIng, 2018-10-26:
@webtroter 
Strange, escaping the single quote `'` somehow does not work. `sed -i "\|cp '/var/www/nextcloud/config/config.sample.php'|d" /DietPi/dietpi/dietpi-software` works then. Jep of course manually remove the line works as well, I just wanted to prevent the manual edit + searching for correct line.

MichaIng, 2018-10-26:
The initial reason for the config.php copy was a rare issue that can occur during install:
- If the Nextcloud MariaDB database somehow already exists, but the install folder /var/www/nextcloud does not, then, because of existing database, the install script expects the install folder with already finished Nextcloud install (+config.php) to exist as well.
- It is dangerous to use a fresh Nextcloud install folder together with an existing database. If install version and database version do not exactly match, they could produce follow up errors or even database corruption.
- So either both need to exist or both need to be fresh.
- I will add a meaningful warning in this rare case, that user need to either find and place the (old) install directory or otherwise remove the existing nextcloud database. Then rerun the install. Install will then exit.

MichaIng, 2018-10-26:
Okay PR is up: https://github.com/Fourdee/DietPi/pull/2186

webtroter, 2018-10-26:
> 
> 
> Okay PR is up: #2186
> Testing/testers required 🔽

How can I test on fresh installs? 

I think this is a better question :
How can I merge the PR locally on the pi and then update dietpi?

Fourdee, 2018-10-27:
@webtroter 

Once we've ran local testing, we'll release to Beta for public testing. Will let you know when its ready + instructions to install Beta.

Fourdee, 2018-10-27:
Test:
- 🈯️ NC + reinstall + GUI tests
- 🈯️ OC + reinstall + GUI tests

Fourdee, 2018-10-27:
@webtroter 

Fix is available in our Beta version, please see following link for more information:
https://github.com/Fourdee/DietPi/blob/master/BRANCH_SYSTEM.md#steps-to-use-the-dietpi-beta-branch

MichaIng, 2018-10-27:
In most cases the quick fix above will work as well, simply removing the faulty config.sample.php copy:
```
sed -i "\|cp '/var/www/nextcloud/config/config.sample.php'|d" /DietPi/dietpi/dietpi-software
# Below only, if you already ran into the broken install:
rm /var/www/nextcloud/config/config.php
rm /var/www/nextcloud/occ
dietpi-software install 114
```
The other parts of the PR only deal/workaround the rare situation that was the reason to add the config.sample.php copy.