MichaIng/DietPi #2185: Error applying dietpi-update from 6.14.0 to 6.17.12 on Rock64

jncl, 2018-10-26:
#### Details:
- Date           | Fri 26 Oct 14:27:54 BST 2018
- Bug report     | N/A
- DietPi version | v6.14.0 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | ayufan-rock64
- SBC device     | Rock64 (aarch64) (index=43)
- Kernel version | #1 SMP Thu Oct 12 09:14:48 UTC 2017
- Distro         | stretch (index=4)
- Command        | wget https://dietpi.com/downloads/dietpi-update_mirror/master/pre-patch_file -O pre-patch_file
- Exit code      | 8
- Software title | DietPi-Update

#### Steps to reproduce:

1. Run dietpi-update 1 as root user
2. ...

#### Expected behaviour:

- Complete successfully

#### Actual behaviour:

- Fails with missing file

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
--2018-10-26 14:27:46--  https://dietpi.com/downloads/dietpi-update_mirror/master/pre-patch_file
Resolving dietpi.com (dietpi.com)... 104.27.179.199, 104.27.178.199, 2606:4700:30::681b:b2c7, ...
Connecting to dietpi.com (dietpi.com)|104.27.179.199|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2018-10-26 14:27:46 ERROR 404: Not Found.
```

MichaIng, 2018-10-26:
@jncl 
Thanks for the report.

Jep we missed to place the pre-patch on our server and until v6.17 dietpi.com repo is used first. I will place it right now, wait a moment.

Fourdee, 2018-10-26:
@MichaIng 

👍 

Added to my notes:
https://github.com/Fourdee/DietPi/issues/548#issue-181241147

MichaIng, 2018-10-26:
@Fourdee 
I edited it a bid. Since v6.17 we use GitHub mirror as first try, but pre v6.17 we used dietpi.com mirror as first try. So especially for pre-v6.17 updates + in case GitHub is down, this is required.

Shall we also upload the beta branch?

Fourdee, 2018-10-26:
@MichaIng 

> Shall we also upload the beta branch? At least when starting next beta phase?

Good idea, but requires us to ensure we manually update mirror.

Might be worth us investing in a cron job to do this for us, every hour?

Fourdee, 2018-10-27:
@MichaIng 

Original issue is now resolved, thanking you, will mark as completed 👍

closed 2018-10-27 by Fourdee