MichaIng/DietPi #2192: issue with install NextCloud on fresh DietPi 6.17.12

777ilya777, 2018-10-28:
#### Details:
- Date           | Sun 28 Oct 14:33:05 GMT 2018
- Bug report     | N/A
- DietPi version | v6.17.12 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1145 SMP Fri Sep 21 15:38:35 BST 2018
- Distro         | stretch (index=4)
- Command        | ncc maintenance:mode --on
- Exit code      | 1
- Software title | DietPi-Services

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
An unhandled exception has been thrown:
Error: Class 'Memcached' not found in /var/www/nextcloud/config/config.php:1156
Stack trace:
#0 /var/www/nextcloud/lib/private/Config.php(214): include()
#1 /var/www/nextcloud/lib/private/Config.php(63): OC\Config->readData()
#2 /var/www/nextcloud/lib/base.php(140): OC\Config->__construct('/var/www/nextcl...')
#3 /var/www/nextcloud/lib/base.php(591): OC::initPaths()
#4 /var/www/nextcloud/lib/base.php(1068): OC::init()
#5 /var/www/nextcloud/console.php(46): require_once('/var/www/nextcl...')
#6 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#7 {main}
```

MichaIng, 2018-10-28:
@777ilya777 
Thanks for your report.

However check existing issues please 😉: https://github.com/Fourdee/DietPi/issues/2184

To fix it quickly, run the following:
```
sed -i "\|cp '/var/www/nextcloud/config/config.sample.php'|d" /DietPi/dietpi/dietpi-software
rm /var/www/nextcloud/config/config.php
rm /var/www/nextcloud/occ
dietpi-software install 114
```
- The last command will rerun Nextcloud installation.

Sorry for the breakfix our side, it is resolved for v6.18, which you can also already apply via beta: https://github.com/Fourdee/DietPi/blob/master/BRANCH_SYSTEM.md

I mark this as closed, if still any issues, please reply to the other opened issue.

closed 2018-10-28 by MichaIng

777ilya777, 2018-10-28:
fix work tnx for rocked speed answer

MichaIng, 2018-10-28:
@777ilya777 
Thanks for feedback 🙂.

> rocked speed
Thanks to Push-IMAP email notifications 😄.