MichaIng/DietPi #2198: DietPi-Config | WiFi: Error message due to "wireless-power off"

MichaIng, 2018-10-29:
```
root@NativePC:~# ifup wlan0
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not supported.
```
Due to https://github.com/Fourdee/DietPi/blob/dev/dietpi/dietpi-config#L2556

Error message shows up but does not have an effect on interface/adapter start and function, during my tests.

What is this setting for, perhaps we can add it conditionally to prevent this error from showing up?
**€: Ah, disable power savings mode. Makes sense. However the setting seems not to be supported by all WiFi adapters 🤔.**
Reading: https://askubuntu.com/questions/85214/how-can-i-prevent-iwconfig-power-management-from-being-turned-on

Fourdee, 2018-10-29:
@MichaIng 

> However the setting seems not to be supported by all WiFi adapters 

Yep 👍 Its a "cover all supported" method of disabling power saving. However, minor visual issue with some adaptors which do not support the feature.

MichaIng, 2019-01-21:
Hmm, not too easy to check if the adapter/module supports the power feature respectively it's enabled: https://www.linuxquestions.org/questions/linux-laptop-and-netbook-25/update-to-kernel-3-13-removed-ability-to-set-powersave-on-ar9462-ath9k-4175499101-print/

Would need to find the related kernel module first and then check if it has the `ps_enable` option and set to 1. Not even sure if this is consistent throughout all devices+kernels 🤔.

Easiest is to test `iw dev $WIFI_DEV set power_save off &> /dev/null` and add the setting based on result.

MichaIng, 2019-01-21:
Fixed with: https://github.com/Fourdee/DietPi/commit/17a1a064533630dbd4393f90f6f10de1c03cea7f

Would be actually great, if we could have this as choice in WiFi adapter settings.
- Show the option, if supported
- Prompt a warning, that it might lead to instability and should be disabled if any connection issues appear
- Add value, if supported, based on user choice
- But I am not sure about the values. `iwconfig` does not take `on` as argument, but an integer:` iwconfig wlan0 power saving N` and has other options `period N` and `timeout N`.
- `iw wlan0 set power_save` on the other hand has `<on|off>` argument.
- Finally there seem to be different options, perhaps not consistent/all supported on all devices. Thus too much testing required to find an all compatible implementation.
- Only thing we could do is leave choice to user whether to disable power savings via `wireless-power off` or not and thus leave choice to system/own `iw`/`iwconfig` calls.

MichaIng, 2019-01-28:
@Fourdee 
I tested it here successfully with onboard WiFi without power savings support. I mark as closed.

closed 2019-01-28 by MichaIng