MichaIng/DietPi #2200: Wifi RTL8188CUS works, but RA Link RT5370 and Atheros AR9271 does not

Gord1, 2018-10-29:
Using Dietpi 6.17 on Nanopi Neo

I have 3 wifi dongles. 1 works out of the box, while 2 others don't.
RTL8188CUS works fine, but RA Link RT5370 and Atheros AR9271 does not

They all show up properly in lsusb, eg
root@DietPi:~# lsusb
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 006: ID 148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter
Bus 004 Device 001: ID 1d6

----------------------------------------------------------------------
for the RTL8188CUS shows correctly
root@DietPi:~# lsmod
Module                  Size  Used by
rtl8192cu              81920  0
rtl_usb                20480  1 rtl8192cu
rtl8192c_common        49152  1 rtl8192cu
rtlwifi                73728  3 rtl_usb,rtl8192c_common,rtl8192cu
nls_ascii              16384  1
rtc_ds1307             24576  0
g_mass_storage         16384  0
ip_tables              24576  0
x_tables               20480  1 ip_tables

But the 2 that don't work show:
root@DietPi:~# lsmod
Module                  Size  Used by
nls_ascii              16384  1
rtc_ds1307             24576  0
g_mass_storage         16384  0
ip_tables              24576  0
x_tables               20480  1 ip_tables

-----------------------------------------------------------------------
And iwconfig shows no wlan0
root@DietPi:~# iwconfig
lo        no wireless extensions.
eth0      no wireless extensions.

----------------------------------------------------------------------
I have the latest firmware for Atheros and Ralink

root@DietPi:~# apt-get upgrade firmware-atheros
Reading package lists... Done
Building dependency tree       
Reading state information... Done
firmware-atheros is already the newest version (20161130-3).

root@DietPi:~# apt-get install firmware-misc-nonfree
Reading package lists... Done
Building dependency tree       
Reading state information... Done
firmware-misc-nonfree is already the newest version (20161130-3).

I don't know why these 2 wifi adapters are not loading and working.

Gordon Williams

MichaIng, 2018-10-29:
@Gord1 
Thanks for your report. Did we already talk in the forum? 🙂 
However, for completeness I rechecked theoretical installed driver support:

**Ralink RT5370** according to web search requires `rt2870.bin` driver which is part of pre-installed `firmware-misc-nonfree` package (on Stretch, on Jessie: `firmware-ralink`): https://packages.debian.org/stretch/all/firmware-misc-nonfree/filelist
- Verify: https://wiki.debian.org/rt2800usb

**Atheros AR9271** clearly covered by pre-installed `firmware-atheros` package: https://packages.debian.org/stretch/firmware-atheros
- Verify: https://wiki.debian.org/ath9k_htc

So both should work out of the box. I read about possible conflicting drivers, but this should only be the case if drivers are compiled/installed manually besides other pre-installed package drivers. When disabling WiFi, DietPi disables known WiFi kernel modules, but this is reverted when enabling WiFi and the here mentioned Atheros and Ralink modules are not even included 😉: https://github.com/Fourdee/DietPi/blob/master/dietpi/func/dietpi-set_hardware#L1176-L1219

On Raspberry Pi the `raspberrypi-sys-mods` by default disables some Realtek drivers for some reason:
```
2018-10-29 22:32:27 root@micha:/var/log# cat /etc/modprobe.d/blacklist-rtl8*
blacklist rtl8192cu
blacklist rtl8xxxu
```

Perhaps NanoPi also comes with some default blacklisting? Could be checked: `cat /etc/modprobe.d/*`

MichaIng, 2018-10-30:
@Gord1 
Jep, required modules are not blocked.

Could you also check `dmesg` for any kernel/voltage related errors?

Fourdee, 2018-10-30:
@Gord1 

Hi Gordon,

> RA Link RT5370 and Atheros AR9271

I suspect these modules are not compiled in the provided FriendlyARM kernel used on the device.

We can confirm this, for the 'RA Link RT5370':
- https://wiki.debian.org/rt2800usb
```
modprobe rt2800usb
``` 

For the AR9271:
- https://wiki.debian.org/ath9k_htc
```
modprobe ath9k_htc
```

Gord1, 2018-10-30:
I tried purging the firmware-atheros and firmware-misc-nonfree drivers and installing them and no improvement.

The funny thing is if I unplug the atheros or ralink dongle and plug in the realtek, the realtek's light comes on and wifi starts up immediately without any fuss or fiddle.  It is just plug in and go.

Gordon

Gord1, 2018-10-30:
after ralink installed and reboot
root@DietPi:~# lsmod
Module                  Size  Used by
nls_ascii              16384  1
rtc_ds1307             24576  0
g_mass_storage         16384  0
ip_tables              24576  0
x_tables               20480  1 ip_tables
root@DietPi:~# modprobe rt2800usb
root@DietPi:~# lsmod
Module                  Size  Used by
rt2800usb              24576  0
rt2800lib              94208  1 rt2800usb
crc_ccitt              16384  1 rt2800lib
rt2x00usb              20480  1 rt2800usb
rt2x00lib              40960  3 rt2800lib,rt2800usb,rt2x00usb
nls_ascii              16384  1
rtc_ds1307             24576  0
g_mass_storage         16384  0
ip_tables              24576  0
x_tables               20480  1 ip_tables
root@DietPi:~# 

So that is different.

but

root@DietPi:~# iwconfig
lo        no wireless extensions.

eth0      no wireless extensions.

dmesg shows
[   46.043587] usbcore: registered new interface driver rt2800usb

I will try the same for the Atheros.
Gordon

Gord1, 2018-10-30:
Got an error with the Atheros:

root@DietPi:~# lsmod
Module                  Size  Used by
nls_ascii              16384  1
rtc_ds1307             24576  0
g_mass_storage         16384  0
ip_tables              24576  0
x_tables               20480  1 ip_tables

root@DietPi:~# modprobe ath9k_htc
modprobe: FATAL: Module ath9k_htc not found in directory /lib/modules/4.14.0

root@DietPi:~# ls /lib/modules/4.14.0
build	modules.alias	   modules.builtin.bin	modules.devname  modules.symbols
extra	modules.alias.bin  modules.dep		modules.order	 modules.symbols.bin
kernel	modules.builtin    modules.dep.bin	modules.softdep  source
root@DietPi:~#

Gord1, 2018-10-31:
Hi,
Just looking at the Ralink 5370 dongle

root@DietPi:~# lsusb
gives
Bus 004 Device 003: ID 148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter

For 148F devices shows that the 5370 is not listed
root@DietPi:~# modinfo rt2800usb |grep 148F
alias:          usb:v148Fp3572d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v148Fp8070d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v148Fp3370d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v148Fp3072d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v148Fp3071d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v148Fp3070d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v148Fp2870d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v148Fp2770d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v148Fp2070d*dc*dsc*dp*ic*isc*ip*in*

But
https://wiki.debian.org/rt2800usb  ***FOR Debian 3.16 (3.16.7-ckt9-2) kernel images
shows a larger number of devices where mine is listed:
USB: 148F:2070 Ralink Technology, Corp. RT2070 Wireless Adapter
USB: 148F:2770 Ralink Technology, Corp. RT2770 Wireless Adapter
USB: 148F:2870 Ralink Technology, Corp. RT2870 Wireless Adapter
USB: 148F:3070 Ralink Technology, Corp. RT2870/RT3070 Wireless Adapter
USB: 148F:3071 Ralink Technology, Corp. RT3071 Wireless Adapter
USB: 148F:3072 Ralink Technology, Corp. RT3072 Wireless Adapter
USB: 148F:3370 Ralink Technology, Corp. RT3370 Wireless Adapter
USB: 148F:3572 Ralink Technology, Corp. RT3572 Wireless Adapter
USB: 148F:3573 Ralink Technology, Corp. RT3573 Wireless Adapter
USB: 148F:5370 Ralink Technology, Corp. RT5370 Wireless Adapter   <---- 5370 ********
USB: 148F:5372 Ralink Technology, Corp. RT5372 Wireless Adapter
USB: 148F:5572 Ralink Technology, Corp. RT5572 Wireless Adapter
USB: 148F:8070 Ralink Technology, Corp. (Device name unknown)
USB: 148F:F301 Ralink Technology, Corp. (Device name unknown)

root@DietPi:~# cat /etc/debian_version
9.5

Is this the reason that the dongle is not working.  The 5370 is missing from the current module?

Is there a way to get it to work?
Gordon Williams

Fourdee, 2018-10-31:
@Gord1 

Hi Gordon,

> For 148F devices shows that the 5370 is not listed

I suspect a disabled kernel compile option:
https://github.com/torvalds/linux/blob/master/drivers/net/wireless/ralink/rt2x00/Kconfig#L175

The above must be enabled for ```rt2800usb``` module to support the 5370.

Unfortunately, we are unable to offer custom kernels, due to time restraints and you would need to compile the kernel, with that option enabled.

Gord1, 2018-10-31:
Hi,
OK, thanks for all your work.  

I've tried Armbian and both ralink and realtek dongles work.

You can close this.
Gordon

closed 2018-11-09 by Fourdee