MichaIng/DietPi #2202: NTFS harddrive as cloud storage

Phil1988, 2018-10-30:
In the past I used Raspbian on a Raspberry to run Nextcloud.
I had an external USB HDD attached to it where all data was stored.
This was very handy as my Raspberry once got a corrupted filesystem and couldnt boot and I was able to restore all the media in a few seconds by attaching this HDD to my (Windows) PC.

Now I wanted something more powerfull and using an MiniPC running Windows 10 and VirtualBox - and of couse dietpi ;)

I tried to use a similar setup with an attached HDD to store all the data to it and included it as shared folder.
In the dietpi-software I wanted to change the "User Data Location" and chose my HDD on the "List" but I get the error, that the filesystem doesnt support permissions (which is correct and I know that but I want to have the benefits of using an NTFS HDD).

The error I get is this https://user-images.githubusercontent.com/7077026/41327801-c19fdb12-6ebc-11e8-91b5-fa7041b9d7bc.png which was included in this enhancement:
https://github.com/Fourdee/DietPi/issues/1846

I know the resons for implementing this but is there a way that I can use a windows readable storage as data storage for my nextcloud data?

Might be a very special question but maybe also a usefull one for the non linux users (I know... kinda weird to write this on the dietpi github :D )

Thanks!

Fourdee, 2018-10-30:
@Phil1988 

Hi,

We added filesystem permissions support for NTFS in v6.15/16 I believe.

Check if following commands gives results:
```
cat /etc/fstab | grep permissions
```

Should get
```
UUID=69EC495234BC97FC /mnt/ssd auto defaults,noatime,rw,nofail,permissions 0 0
```

If not, try unmount + mount the drive again in ```dietpi-drive_manager```. Once you see a results (as above), try moving the userdata.

Phil1988, 2018-10-30:
Hmm...
Looks like things are more complicated.

As I am using VirtualBox my "storage" is a virtualDisk file, which will destroy the direct access benefit from using ths NTFS drive.
So I thought I could use a shared folder that links to the root of my HDD.
The shared folder is working but in the drive-manager I cant see it.

the `cat /etc/fstab | grep permissions`
gives me
[image]

and the driver-manager shows the same:
[image]

Do you have an idea how I can use one drive simultaneously on the guest and host in a filesystem windows and linux can read?
VirtualBox and VMware are ok for me, but as far as I saw the VMware only allows auto start (on windows boot) as a shared-vm, which doesnt allow shared folders :D 

So I'm a bit stuck right now ;)

MichaIng, 2018-10-30:
@Phil1988 
Not sure if I fully understood everything.

When you set up a samba/NFS/whatever server, this will not show up in drive manager. Only when you mount an external network drive, this can be done via drive manager and shows up there as well.

Our VirtualBox images come with vmdk disk image file jep, but this does not influence the contained file system nor the permissions you can set. You can create an additional virtual disk, attach it to the VM and format it as NTFS. It should be possible to set up this NTFS drive as network (e.g. samba for Windows access) share and move dietpi_userdata to this drive.

Do it the other way round, adding a shared folder from host system to mount on guest should be not the preferred way, since the directory is mainly used by the guest, or at least should be when containing it's dietpi_userdata.
I have no idea how these shared folders appear within the Linux guest. this is something we could investigate and see if we can somehow recognize and handle them with drive manager.

Phil1988, 2018-10-30:
Currently I have no samba/NFS running.

The "problem" of these vmdk is, that it is a contained file where I have no direct access to the data (only at running the VM) and therefore a backup if something happens (vmdk currupted, VM destroyed...) is not possible.
So if I create a virtual disk and attach it to the VM then I have no real benefit.

The goal I am trying to reach is, to have a windows accessable folder (a real one where I can see the files, not a virtual container) that is the storage of the cloud.
So if I or a user upload an important file and there is any problem I can just take this file from the "server" aka shared folder/drive from windows and restore it.

The only thing I can tell is that the shared folders are mounted in the /media directory.

So what would be your advice for a solution?
Maybe you already mentioned it but I didnt understand :)

Thanks for you response and help!

MichaIng, 2018-11-01:
@Phil1988
Okay jep direct access to a vmdk/vdi image is not possible.

You have several options to achieve your need:
- As a full VM backup I would regularly create a snapshot of the whole VM. So if it refuses to start at all, you can recover thus snapshot.
- You could plug an external USB drive to the host and attach this through VirtualBox UI to the VM. Format it with NTFS to allow posix permissions. Then you e.g. can move userdata to it and store backups there. When to VM breaks, files are then accessible from host.
- Create a Syncthing, Nextcloud or ownCloud server on the VM and use their clients to sync data to the host. The userdata directory can be added e.g. as external storage the Nextcloud and ownCloud.
- An e.g. Samba network drive makes the files accessible from host as well, but the files are not automatically stored on the host and the VM must run to have access. But at least you could do regular backups from the host side.

So yeah it depends a bid on what exactly you want to backup and how current the files need to be.
For system backup, I think the snapshot solution is very handy.

Phil1988, 2018-11-06:
Thanks @MichaIng for your answer.

- The Snapchot thing is not what I had in mind. It should be pretty much "bullet proof" and if the host system breaks, then the data is lost. (therefore the dedicaded drive)

- You said I can plug in an external USB drive and move the userdata to it.
By attaching it to the VM I have no access to it from the host system right? So it somehow is bullet proof but not really accessable at any time :)

- The client on the host system is a nice workaround which will fullfill my goals but also double the data.
So my storage will drain twice as fast :D  

- The Samba network drive will (in my knowledge) not be working if there is something happening with the VM right? So the VM must work to make the drive accessable and therefore its not really a solution.

So the best solution so far is to "mirror" the data beween the VM and the host by using the nextcloud client. But this is not really a long term solution as it doubles the data or better: it cuts the usable storage to 50%.

Have you had any success with the investigations on shared folders for your drive manager?

So far this is the only really usable solution I see because it has the following benefits:
Data accessable during the use of the cloud (which is planned to be 24/7 ;) )
Data accessable if the VM breaks
Data accessable if the complete host system breaks

For the complete "bullet proof" I will consider a backup to an external drive so the files are also accessable it the HDD breaks :D

MichaIng, 2018-11-24:
Checked a bid:
- Shared folder possible after installing vbox guest additions. Works a charm then.
- `dietpi-drive_manager` does not recognize them, although I though it will catch it as network drive, since mount "source" is just the custom name of the shared folder.
- I didn't find a way to explicitly check for such mounts (mounted or attached), since lsblk/blkid does not list them. Similar to `encryptfs`.
- mount type: `vboxsf`

So currently it is okay. At least drive manager does not destroy the mount of somehow adds it (wrong) to fstab.

Next tests, to check if userdata or at least backups will work, fully preserving all file types and attributes:
- [ ] POSIX permissions
- [ ] Symlinks

MichaIng, 2019-04-22:
POSIX permissions and symlink not supported, or, it depends on host OS and the related file system of the shared folder. In my case Windows (NTFS => no permissions support).

However we handle it similar to bind mounts:
- Detect by fs type (vboxsf), re-add to fstab as is.
- Remove source/target from df output, so assure it is not shown doubled
- Temporary mounts are not shown most likely since $1 is the name of the share which most likely does not contain as slash.

closed 2019-05-12 by MichaIng