MichaIng/DietPi #2206: missing files after diet-pi update to v6.17

DoaneAS, 2018-10-31:
## Creating a bug report/issue:
 Missing system files following update.

#### Required Information:
- DietPi version	| 
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=17
G_DIETPI_VERSION_RC=12
G_GITBRANCH=master
G_GITOWNER=Fourdee

- Distro version	| stretch 
- Kernel version	| Linux PC196431 4.14.71-v7+ #1145 SMP Fri Sep 21 15:38:35 BST 2018 armv7l GNU/Linux
- SBC device		| RPi 3 Model B (armv7l)
- Power supply used	| standard
- SDcard used		|  SanDisk ultra

#### Additional Information (if applicable):
- Software title	| Diet-pi update

#### Steps to reproduce:
1. deitpi-update
2. update does not complete.
3. No record of system backups after update.
4. Missing files in /usr/bin after update (ldconfig).  
#### Expected behaviour:
updates successfully.
- ...

#### Actual behaviour:
update breaks installation; missing files. 
- ...

#### Extra details:
Maybe a hardware issue?  I/O to sd card appears normal.  
This is strange, but after updating and the reboot failed, I tried to restore from a backup, and I'm missing all backups, and basic system binaries like ldconfig. I made a backup right before updating.    Here is log, from attempt to install samba after diet-pi update.  Error occurs with any installs, samba just example.
```
DietPi-Software: G_AGI: samba
                          │  - Exit code: 100
                          │  - DietPi version: v6.17.12 (Fourdee/master) | HW_MODEL:3 | HW_ARCH:2 | DISTRO:4
                          │  - Image creator: n/a
                          │  - Pre-image: n/a
                          │
                          │ Log file contents:
                          │ Preconfiguring packages ...
                          │ dpkg: warning: 'ldconfig' not found in PATH or not executable^M
                          │ dpkg: warning: 'start-stop-daemon' not found in PATH or not executable^M
                          │ dpkg: error: 2 expected programs not found in PATH or not executable^M
                          │ Note: root's PATH should usually contain /usr/local/sbin, /usr/sbin and /sbin^M
                          │ E: Sub-process /usr/bin/dpkg returned an error code (2)
                          │

```

Btw, `echo $PATH` returns `/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin` so I'm not quite what is going on.

Running `diet-pi update -1` pulls down an update, but after confirming the reboot, I get `reboot command not found`.   After a manual reboot, system cannot open ssh connections.  

Maybe possible for diet-pi to back up system and roll-back if update unsuccessful?    This is the third time in a year that an update has broken the system and required reflashing sd card.

thanks a lot.

MichaIng, 2018-10-31:
Very strange. Did you check `/bin/which -a ldconfig start-stop-daemon` and `ls -l /sbin/ldconfig /sbin/start-stop-daemon` if those binaries are really missing?

From which version of DietPi did you update to v6.17? We did a dietpi-backup rework some versions ago, after which older backups were not supported any more. But of course those can be restored manually. Verify that the actual backup files are still available at your backup location.

Please also check `dmesg` (hopefully binary available, perhaps /bin/dmesg works otherwise) for kernel/boot issues.

DoaneAS, 2018-11-02:
Thanks.  All is working fine after a fresh install.  Unfortunately I don't have a copy of the old system to check your suggestions.  I think the last time I updated, before this most recent attempt, was not more than a few weeks ago.

My best guess is a hardware issue with the SD card could be to blame, or something related to mounting that volume.  In any case I'll close this and appreciate your response and help.

Best,
Ashley

closed 2018-11-02 by DoaneAS