MichaIng/DietPi #2210: PREP_SYSTEM_FOR_DIETPI fails on NanoPi Neo Air with latest Armbian

troyane, 2018-11-01:
#### Details:
- Date           | Thu  1 Nov 13:18:39 UTC 2018
- Bug report     | Looks like it is related to https://github.com/Fourdee/DietPi/issues/2054
- DietPi version | v6.16.0 (Fourdee/master)
- Pre-image      | Armbian_5.65_Nanopiair_Debian_stretch_next_4.14.78
- SBC device     | NULL (index=64)
- Kernel version | # 412 SMP Fri Oct 26 11:37:04 CEST 2018
- Distro         | stretch (index=4)
- Command        | `G_AGI: apt-transport-https apt-utils bash-completion bc bzip2 ca-certificates console-setup cron curl debconf dirmngr ethtool fake-hwclock gnupg htop iputils-ping isc-dhcp-client kmod locales nano p7zip-full parted psmisc resolvconf sudo systemd-sysv tzdata udev unzip usbutils wget whiptail crda firmware-atheros firmware-brcm80211 firmware-iwlwifi iw rfkill wireless-tools wpasupplicant firmware-misc-nonfree dropbear-run initramfs-tools dosfstools hdparm`
- Exit code      | 100
- Software title | DietPi-PREP
- SBC device		| NanoPi Neo Air (64) 
- SDcard used		| SanDisk ultra

#### Steps to reproduce:

1. Use Etcher to prepare microSD with latest official Armbian from https://dl.armbian.com/nanopiair/Debian_stretch_next.7z
2. Connect NanoPi Neo Air to laptop via serial.
3. Boot into Armbian, setup Wi-Fi, do `apt update/upgrade`.
4. Start `screen` and do all steps from https://github.com/Fourdee/DietPi/issues/1285
5. As a result -- got mentioned error.

#### Few questions:
1. Main -- is there an option to overcome this issue automatically?
2. Secondary -- is there a way to "continue" PREP_SYSTEM_FOR_DIETPI after this error? 

#### Additional logs:
```
[  OK  ] DietPi-PREP | APT dist-upgrade, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

[  OK  ] DietPi-PREP | G_AGDUG
[ INFO ] DietPi-PREP | Installing core DietPi pre-req APT packages
[  OK  ] DietPi-PREP | APT installation for: apt-transport-https apt-utils bash-completion bc bzip2 ca-certificates console-setup cron curl debconf dirmngr ethtool fake-hwclock gnupg htop iputils-ping isc-dhcp-client kmod locales nano p7zip-full parted psmisc resolvconf sudo systemd-sysv tzdata udev unzip usbutils wget whiptail crda firmware-atheros firmware-brcm80211 firmware-iwlwifi iw rfkill wireless-tools wpasupplicant firmware-misc-nonfree dropbear-run initramfs-tools dosfstools hdparm, please wait...
E: Failed to fetch https://deb.debian.org/debian-security/pool/updates/main/p/p7zip/p7zip_16.02+dfsg-3+deb9u1_armhf.deb  Could not resolve host: deb.debian.org
E: Failed to fetch https://deb.debian.org/debian-security/pool/updates/main/p/p7zip/p7zip-full_16.02+dfsg-3+deb9u1_armhf.deb  Could not resolve host: deb.debian.org
E: Failed to fetch https://deb.debian.org/debian/pool/main/d/dropbear/dropbear-bin_2016.74-5_armhf.deb  Could not resolve host: deb.debian.org
E: Failed to fetch https://deb.debian.org/debian/pool/main/d/dropbear/dropbear-run_2016.74-5_all.deb  Could not resolve host: deb.debian.org
E: Failed to fetch https://deb.debian.org/debian/pool/non-free/f/firmware-nonfree/firmware-atheros_20161130-3_all.deb  Could not resolve host: deb.debian.org
E: Failed to fetch https://deb.debian.org/debian/pool/non-free/f/firmware-nonfree/firmware-brcm80211_20161130-3_all.deb  Could not resolve host: deb.debian.org
E: Failed to fetch https://deb.debian.org/debian/pool/non-free/f/firmware-nonfree/firmware-iwlwifi_20161130-3_all.deb  Could not resolve host: deb.debian.org
E: Failed to fetch https://deb.debian.org/debian/pool/non-free/f/firmware-nonfree/firmware-misc-nonfree_20161130-3_all.deb  Could not resolve host: deb.debian.org
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

[FAILED] DietPi-PREP | G_AGI: apt-transport-https apt-utils bash-completion bc bzip2 ca-certificates console-setup cron curl debconf dirmngr ethtool fake-hwclock gnupg htop iputils-ping isc-dhcp-client kmod locales nano p7zip-full parted psmisc resolvconf sudo systemd-sysv tzdata udev unzip usbutils wget whiptail crda firmware-atheros firmware-brcm80211 firmware-iwlwifi iw rfkill wireless-tools wpasupplicant firmware-misc-nonfree dropbear-run initramfs-tools dosfstools hdparm
``

troyane, 2018-11-01:
@MichaIng I was aware of possible problems with connection over SSH, so I connected via UART. Also Nano Pi Neo Air has no Ethernet option, its Wifi-only.

Fourdee, 2018-11-04:
I believe the issue is network-manager. Removing the package breaks current connection over WiFi.

A quick solution is USB > Ethernet then run PREP.

Another possible solution is to do the following before PREP to setup WiFi without network-manager:
- ```apt-get purge network-manager -y```
- copy paste all
```
cat << _EOF_ > /etc/network/interfaces
#/etc/network/interfaces
#Please use DietPi-Config to modify network settings.

# Local
auto lo
iface lo inet loopback

# Ethernet
#allow-hotplug eth0
iface eth0 inet dhcp
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
#dns-nameservers 8.8.8.8 8.8.4.4

# Wifi
allow-hotplug wlan0
iface wlan0 inet dhcp
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
wireless-power off
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
#dns-nameservers 8.8.8.8 8.8.4.4
_EOF_
- edit ```/etc/wpa_supplicant/wpa_supplicant.conf``` and set your WiFi creds, eg:
```
country=GB
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
network={
    ssid="MYSSID"
    scan_ssid=1
    key_mgmt=WPA-PSK
    psk="MYKEY"
}
```
- Reload systemd ```systemctl daemon-reload```
- Restart networking ```ifdown wlan0; systemctl restart networking; ifup wlan0```
- Run DietPi PREP

troyane, 2018-11-04:
@Fourdee thank you for your instructions. IMO it sounds better for me to avoid any USB2Ethernet. So, I'd like to test your second option tomorrow and will write here as soon as I get any results.

troyane, 2018-11-05:
@Fourdee I've already done steps you've provide, and I need to say that DietPi PREP done its job already.

Just in case, I've backed-up microSD image by `dd if=/dev/mmblck | gzip > handmade_dietpi.img.gz` before DietPi first start. You can [download it here](https://drive.google.com/file/d/1bJi3s3GH03Bkr5rx5lw2u09G22qgcgWt/view?usp=sharing). 

<details>
<summary>Here is first run log</summary>

```
** File not found /boot/boot.env **

** Unable to read "/boot/boot.env" from mmc0:1 **
Failed (-5)
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
230454 bytes read in 31 ms (7.1 MiB/s)
Unknown command 'bmp' - try 'help'
starting USB...
No controllers found
Autoboot in 1 seconds, press <Space> to stop
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot/boot.scr
3708 bytes read in 12 ms (301.8 KiB/s)
## Executing script at 43100000
U-boot loaded from SD
Boot script loaded from mmc
228 bytes read in 10 ms (21.5 KiB/s)
5102467 bytes read in 411 ms (11.8 MiB/s)
6984272 bytes read in 554 ms (12 MiB/s)
Found mainline kernel configuration
31408 bytes read in 67 ms (457 KiB/s)
504 bytes read in 106 ms (3.9 KiB/s)
Applying kernel provided DT overlay sun8i-h3-usbhost1.dtbo
504 bytes read in 107 ms (3.9 KiB/s)
Applying kernel provided DT overlay sun8i-h3-usbhost2.dtbo
4179 bytes read in 99 ms (41 KiB/s)
Applying kernel provided DT fixup script (sun8i-h3-fixup.scr)
## Executing script at 44000000
## Loading init Ramdisk from Legacy Image at 43300000 ...
   Image Name:   uInitrd
   Image Type:   ARM Linux RAMDisk Image (gzip compressed)
   Data Size:    5102403 Bytes = 4.9 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 43000000
   Booting using the fdt blob at 0x43000000
   Loading Ramdisk to 49b22000, end 49fffb43 ... OK
   reserving fdt memory region: addr=43000000 size=6e000
   Loading Device Tree to 49ab1000, end 49b21fff ... OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.78-sunxi (root@nightly) (gcc version 7.2.1 20171011 (Linaro GCC 7.2-2017.11)) #412 SMP Fri Oct 26 11:37:04 CEST 2018
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=50c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: FriendlyElec NanoPi NEO Air
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x5bc00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: Using PSCI v0.1 Function IDs from DT
[    0.000000] random: get_random_bytes called from start_kernel+0x8d/0x36a with crng_init=0
[    0.000000] percpu: Embedded 17 pages/cpu @dff5b000 s39884 r8192 d21556 u69632
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129920
[... 501 lines cut ...]
         Starting Update UTMP about System Runlevel Changes...
[  OK  ] Started Update UTMP about System Runlevel Changes.

 ────────────────────────────────────────────────
 DietPi | Mon 05/11/18 - 09:30
 ────────────────────────────────────────────────
 v6.16.0 | NanoPi NEO Air (armv7l)
 ────────────────────────────────────────────────
 NONE | Use dietpi-config to setup a connection
 ────────────────────────────────────────────────
 Default Login:
 Username = root
 Password = dietpi

 Please login to continue
```

</details>

---

I'm connecting via UART, but I can't log in:
[image]

No reaction on any hotkeys, clicks, enter etc.

Fourdee, 2018-11-10:
@troyane 

Managed to get WiFi working with uboot/kernel from nanopi-neo-air_sd_friendlycore-xenial_4.14_armhf_20181011.img

Will try and run PREP + create image.

Fourdee, 2018-11-10:
Wifi dead after PREP. Cant only assuming updating the APT packages is causing the issue, no packages removed.

Grabbed hciuart/hciconfig/brcm_patchram_plus from ubuntu, which works before PREP.

Stuck in loop.
```
 /etc/init.d/brcm_patchram_plus
```

Fourdee, 2018-11-10:
FS resize not working:
```
root@DietPi:~# cat /var/tmp/dietpi/logs/fs_partition_resize.log

Welcome to fdisk (util-linux 2.29.2).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Command (m for help): Disk /dev/mmcblk0: 7.4 GiB, 7948206080 bytes, 15523840 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x83befaea

Device         Boot  Start    End Sectors  Size Id Type
/dev/mmcblk0p1       49152 131071   81920   40M 83 Linux
/dev/mmcblk0p2      131072 950271  819200  400M 83 Linux

Command (m for help): Partition number (1,2, default 2):
Partition 2 has been deleted.

Command (m for help): Disk /dev/mmcblk0: 7.4 GiB, 7948206080 bytes, 15523840 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x83befaea

Device         Boot Start    End Sectors Size Id Type
/dev/mmcblk0p1      49152 131071   81920  40M 83 Linux

Command (m for help): Partition type
   p   primary (1 primary, 0 extended, 3 free)
   e   extended (container for logical partitions)
Select (default p):    p   primary (1 primary, 0 extended, 3 free)
   e   extended (container for logical partitions)
Select (default p):    p   primary (1 primary, 0 extended, 3 free)
   e   extended (container for logical partitions)
Select (default p):
Using default response p.
Partition number (2-4, default 2): Partition number (2-4, default 2): Partition number (2-4, default 2): First sector (2048-15523839, default 2048): 
```

Fourdee, 2018-11-10:
@troyane 

Image completed, verified working WiFi. However, please test and let us know if any issues:
https://dietpi.com/downloads/images/DietPi_NanoPiNEOAir-ARMv7-Stretch.7z

Fourdee, 2018-11-16:
Image passed local testing, WiFi working fine. Will release to dietpi.com.

Fourdee, 2018-11-16:
Image released to http://dietpi.com/download

Will mark as completed. 

PREP workaround available for WiFi only devices with ```network-manager``` installed:
https://github.com/Fourdee/DietPi/issues/2210#issuecomment-435687637

closed 2018-11-16 by Fourdee