MichaIng/DietPi #2211: Open Beta v6.18 | Help testing and hardening the next release

MichaIng, 2018-11-01:
RC version | v6.18.2
---------- | -------
Changelog | https://github.com/Fourdee/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/Fourdee/DietPi/compare/master...beta
How to apply | https://github.com/Fourdee/DietPi/blob/beta/BRANCH_SYSTEM.md

### Known issues:
#### DietPi functionality
- DietPi-Globals | G_* functions are not available via `sudo`: https://github.com/Fourdee/DietPi/issues/2135
- DietPi-Config | Enabling WiFi + Ethernet adapters, both on different subnets, breaks WiFi connection: https://github.com/Fourdee/DietPi/issues/2103
  - Current workaround is to disable the adapter that is not in use (not for internet connection).
#### DietPi visual
- DietPi-Globals | G_AG* functions show some wrong aligned output lines: https://github.com/Fourdee/DietPi/issues/1788
- DietPi-Config | When bringing up WiFi interfaces with some WiFi adapters, a power management error is thrown. However this does not affect functionality: https://github.com/Fourdee/DietPi/issues/2198
#### SBC/device related
- RPi | LXAppearance (on LXDE desktop) hangs on dbus-launch: https://github.com/Fourdee/DietPi/issues/1791
- Pine64 | Enable Bluetooth add-on module: https://github.com/Fourdee/DietPi/issues/528
- Odroid XU4 | Xorg (e.g. desktops) fail to start: https://github.com/Fourdee/DietPi/issues/2130
  - To use desktop, `/etc/X11/xorg.conf` needs to be removed.
  - But to use Kodi, `/etc/X11/xorg.conf` is required, so one will work, one will be broken.
- Odroid C2 | Kodi GPU acceleration does not work on Stretch images: https://github.com/Fourdee/DietPi/issues/1814
- Odroid C2 | HotSpot does not work with (some?) WiFi adapters: https://github.com/Fourdee/DietPi/issues/1955
#### Software title related
- OpenBazaar | Client connection fails: https://github.com/Fourdee/DietPi/issues/1090
- Node-RED | Pre-installed modules cannot be updated via webUI: https://github.com/Fourdee/DietPi/issues/2073
- Raspimjpeg | With Lighttpd, streaming mjpeg does not work: https://github.com/Fourdee/DietPi/issues/1747

Fourdee, 2018-11-17:
Beta testing now live for v6.18 release:
https://t.co/1fEdFPyBya

Will try and wrap up a few more tickets in the next few days.

msdos, 2018-11-20:
I think this process of "open beta" could have a line in README.md, and an information about https://github.com/Fourdee/DietPi/blob/dev/BRANCH_SYSTEM.md#steps-to-use-the-dietpi-beta-branch for devs as well.

MichaIng, 2018-11-20:
@msdos 
How exactly do you want to have it mentioned? Since everyone can freely switch to `beta` branch, it is obviously opened.
We could add an info that we will inform via GitHub, forum and Twitter about new "`Beta phases`", which means feature stop man bugfix concentration?

Note, that we introduced the beta branch just with the last release, so we are learning/testing how to use it best ourselves currently. Thinks might change about how we handle it until we found a way that suites best 😉.

But also for my impression we need to have `dev` and `beta` branch more separate. Keep beta branch untouched (so leave it match `master` branch) until open beta phase is started, so there is no doubt/question about beta phase being "opened"/"started" or not and users on beta branch will stay with current release until open beta has started. Also stay with feature stop then, to avoid implementing more testing cases and concentrate on fixing/stability. I hope we can increase trust into the beta program by this and win more users that are willing to stay with beta branch so it will grow to a real benefit.
@Fourdee 
What do you think?

MichaIng, 2018-11-20:
Referenced Beta in forum: https://dietpi.com/phpbb/viewtopic.php?f=9&t=5254

Fourdee, 2018-11-20:
@MichaIng 

> Keep beta branch untouched (so leave it match master branch) until open beta phase is started, so there is no doubt/question about beta phase being "opened"/"started" or not and users on beta branch [...]

Yep agree 👍 

I've been aiming to only push dev > beta, prior to "open beta (eg: when we twitter post)" when a significant change is made (eg: nextcloud install fix). Aside from that, yep, beta branch should only be merged after our initial testing is passed, not before (which we do already).

Fourdee, 2018-11-26:
Thanks to all Beta testers, v6.18 live, completed.

closed 2018-11-26 by Fourdee

MichaIng, 2018-11-26:
Released, will close 🎆 🍾