MichaIng/DietPi #2212: Jackett reinstall doesn't work.

msdos, 2018-11-01:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| `cat /DietPi/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=17
G_DIETPI_VERSION_RC=12
G_GITBRANCH=master
G_GITOWNER=Fourdee
- Distro version	| `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
9.4
- Kernel version	| `uname -a`
Linux 4.14.71-v7+ #1145 SMP Fri Sep 21 15:38:35 BST 2018 armv7l GNU/Linux
- SBC device		| `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
RPi 3 Model B (armv7l)
- Power supply used	| (EG: 5V 1A RAVpower)
- SDcard used		| (EG: SanDisk ultra)

#### Additional Information (if applicable):
- Software title	| Jackett
- Was the software title installed freshly or updated/migrated?
Updated
- Can this issue be replicated on a fresh installation of DietPi?
I don't know. Probably since I think the error is in the reinstall logic itself.

#### Steps to reproduce:

1. Run `dietpi-software reinstall 147`.

#### Expected behaviour:

- Jackett should be updated.

#### Actual behaviour:

- Jackett is not updated. Instead, a new folder named `Jackett` is now inside `/opt/jackett`.

#### Extra details:

- I think this is related to this snippet. https://github.com/Fourdee/DietPi/blob/5ccb97c5ac9cde53eb31e95b7dba729dcc22b1cc/dietpi/dietpi-software#L6041

When you try to do a simple proof of concept of this in bash:

```bash
root@Dietpi:/tmp# mkdir test
root@Dietpi:/tmp# mkdir Test
root@Dietpi:/tmp/test# touch test/test_file
root@Dietpi:/tmp/Test# touch Test/Test_file
root@Dietpi:/tmp# ls -la test/
total 0
drwxr-xr-x 2 root root  60 Nov  1 17:49 .
drwxrwxrwt 9 root root 180 Nov  1 17:49 ..
-rw-r--r-- 1 root root   0 Nov  1 17:49 test_file
root@Dietpi:/tmp# ls -la Test/
total 0
drwxr-xr-x 2 root root  60 Nov  1 17:49 .
drwxrwxrwt 9 root root 180 Nov  1 17:49 ..
-rw-r--r-- 1 root root   0 Nov  1 17:49 Test_file
# Run a command similar to the Jackett reinstall script
root@Dietpi:/tmp# cp -R /tmp/Test /tmp/test
root@Dietpi:/tmp# ls -la /tmp/test
total 0
drwxr-xr-x 3 root root  80 Nov  1 17:53 .
drwxrwxrwt 9 root root 180 Nov  1 17:49 ..
drwxr-xr-x 2 root root  60 Nov  1 17:53 Test
-rw-r--r-- 1 root root   0 Nov  1 17:49 test_file
root@Dietpi:/tmp# cd /tmp
root@Dietpi:/tmp# ls -la test/Test/
total 0
drwxr-xr-x 2 root root  60 Nov  1 17:53 .
drwxr-xr-x 3 root root 100 Nov  1 17:55 ..
-rw-r--r-- 1 root root   0 Nov  1 17:53 Test_file
root@Dietpi:/tmp#
```

You can see that doint this `cp` `Test` is being put _inside_ `test`. This is exactly what happens with the Jackett reinstallation.

MichaIng, 2018-11-02:
@msdos
Thanks for your report. Jep indeed it needs to be `cp -Rf /opt/Jackett/. /opt/jackett` to support reinstall into existing directory. With asterisk instead of dot, to skip hidden files (with leading dot in name).
For fresh install mkdir -p /opt/jackett required first.

Are there any configuration files or data stored within the install dir? Otherwise we could remove existing dir first to avoid accumulating obsolete files.

I anyway planned to implement a shared method in how to handle existing install dirs. This is not consistent currently.

Allowing following options:
- Clearing existing dir first, if really no custom data/configs are expected inside.
- Merging (via cp method above), if custom files are expected, but definitely not shipped by download dir.
- Skip download/install completely, if the software has an internal update method, which should be preferred then. Use this instead then, if CLI exists?
- Ask user in case of interactive call, skip as default, if we are not sure.

Fourdee, 2018-11-05:
@MichaIng 

Works a charm 👍 updated from 0.9 to 0.10.410.0 via reinstall.

Completed.

closed 2018-11-05 by Fourdee