MichaIng/DietPi #2215: Using ampersand in password breaks transmission-daemon service and webui.

MistahDarcy, 2018-11-03:
#### Information:
- Software title	| Transmission
- Was the software title installed freshly or updated/migrated? Fresh install
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce:
1. Create any password with an & (ampersand) symbol.
2. transmission-daemon completely fails to load on boot, and webui is inaccessible.

#### Extra details:
Manually editing /transmission-daemon/settings.json and updating the password field allows for normal operation. (EX: "rpc-password": "password&123",)

MichaIng, 2018-11-03:
@MistahDarcy 
Thanks for this info. Hmm I guess there are other special characters unsupported then. Hmm not sure how to handle this best.
- Replace/Remove magic characters from GLOBAL_PW, before applying to Transmission and inform user about it? But what about DietPi-Automation installs?
- Only allow non-magic characters for the global pass? No good solution. It is an issue that there are still so much services that do not support special chars and they should be pushed to do so instead the other way round! No modern solution to only support alphanumerical chars...

Do you have a suggestion or idea how to handle this best, keeping modern security standards, allowing uninteractive install but keeping user from running into an unexpected situation? 🤔

Fourdee, 2018-11-03:
Notes:

```
[  OK  ] DietPi-Software | Setting in /etc/transmission-daemon/settings.json adjusted: "rpc-password": "password&123",
```

```
"rpc-password": "password    "rpc-password": "{a9a67b86744c2069f05eb2857f8c9e834ba04d6b.LGo72pW",123",
```

Re-run, G_CONFIG_INJECT:
```
"rpc-password": "password"rpc-password": "password    "rpc-password": "{a9a67b86744c2069f05eb2857f8c9e834ba04d6b.LGo72pW",123",123",
```

Same
```
G_CONFIG_INJECT '"rpc-password"' "\"rpc-password\": \"$GLOBAL_PW\"," /etc/transmission-daemon/settings.json
```

---

🈯️ We need to check ```GLOBAL_PW``` and escape ```\``` in ```G_CONFIG_INJECT```
```
GLOBAL_PW='password\&123'
G_CONFIG_INJECT '"rpc-password"' "\"rpc-password\": \"$GLOBAL_PW\"," /etc/transmission-daemon/settings.json
```

MichaIng, 2018-11-03:
@Fourdee 
Ah okay, so not an issue of Transmission.

Hmm that's bad, `sed -Ei "0,\|^[[:blank:]]*$pattern.*$|s||$setting|" $file` expands `&` even when added via variable.

I anyway planned to do all required escaping within the script.

MichaIng, 2018-11-07:
PR up for G_CONFIG_INJECT internal excaping: https://github.com/Fourdee/DietPi/pull/2227
- `&` added
- Extended regex support removed, as this would have made everything more complicated and benefit is limited. Need to check all function calls for ext regex use, before it can be merged.
- But would be great to test the function bullet prove. sed/grep syntax errors should not be possible any more now, if I haven't forgotten a special char. Debug/error code could then be removed.

Fourdee, 2018-11-08:
Test:
- ```dietpi```
- ```password&123```

🈯️ Nextcloud + Transmission.

@MichaIng 
You legend 👍 Great fix!

MichaIng, 2018-11-11:
@Fourdee 
Solved: https://github.com/Fourdee/DietPi/commit/8b6108107b58758ef141e9054983bbd344db762c

But as said, we need to keep this in mind and solve thoughout all code!

Fourdee, 2018-11-11:
G_ESCAPE_EXPANSION + G_CONFIG_INJECT test:
Password:

> \`"$%&test

Strange, had to escape the ! in DietPiRocks!
```
root@DietPi:~# GLOBAL_PW="$(cat /var/lib/dietpi/dietpi-software/.GLOBAL_PW.bin | openssl enc -base64 -d -aes-256-cbc -nosalt -pass pass:DietPiRocks!)"
-bash: !: event not found

root@DietPi:~# GLOBAL_PW="$(cat /var/lib/dietpi/dietpi-software/.GLOBAL_PW.bin | openssl enc -base64 -d -aes-256-cbc -nosalt -pass pass:DietPiRocks\!)"
root@DietPi:~# echo $GLOBAL_PW
\`"$%&test
```

MichaIng, 2018-11-11:
@Fourdee 
As said, this is very dangerours: `echo -e` will do other expansion now, this will definitely lead to other issues! We need to do this individually!

Fourdee, 2018-11-11:
@MichaIng 

Yep, will remove from G_CONFIG_INJECT, but leave ```G_ESCAPE_EXPANSION ``` in place for testing.

closed 2018-11-20 by MichaIng