MichaIng/DietPi #2216: Plex media server doesn't start

roelfoubert, 2018-11-03:
I have a fully updated system but today my plex server wont start.
when I look for the service status I get a failed status.
[image]
someone knows how to fix this?

MichaIng, 2018-11-03:
@roelfoubert 
Thanks for your report.
Could you please check the full log when attempting to start:
```
systemctl start plexmediaserver
sleep 5
journalctl -r
```
This shows the full syslog output in reverse order, newest at the top, so you don't need to scroll to bottom 😉.

roelfoubert, 2018-11-03:
Nov 03 15:04:15 raspberrypi systemd[1]: plexmediaserver.service: Failed with result 'exit-code'.
Nov 03 15:04:15 raspberrypi systemd[1]: plexmediaserver.service: Unit entered failed state.
Nov 03 15:04:15 raspberrypi systemd[1]: Failed to start Plex Media Server for Linux.
Nov 03 15:04:15 raspberrypi systemd[1]: plexmediaserver.service: Start request repeated too quickly.
Nov 03 15:04:15 raspberrypi systemd[1]: Stopped Plex Media Server for Linux.
Nov 03 15:04:15 raspberrypi systemd[1]: plexmediaserver.service: Service hold-off time over, scheduling restart.
Nov 03 15:04:10 raspberrypi systemd[1]: plexmediaserver.service: Failed with result 'exit-code'.
Nov 03 15:04:10 raspberrypi systemd[1]: plexmediaserver.service: Unit entered failed state.
Nov 03 15:04:10 raspberrypi systemd[1]: plexmediaserver.service: Main process exited, code=exited, status=255/n/a
Nov 03 15:04:09 raspberrypi systemd[1]: Started Plex Media Server for Linux.
Nov 03 15:04:09 raspberrypi systemd[1]: Starting Plex Media Server for Linux...
Nov 03 15:04:09 raspberrypi systemd[1]: Stopped Plex Media Server for Linux.
Nov 03 15:04:09 raspberrypi systemd[1]: plexmediaserver.service: Service hold-off time over, scheduling restart.
Nov 03 15:04:04 raspberrypi systemd[1]: plexmediaserver.service: Failed with result 'exit-code'.
Nov 03 15:04:04 raspberrypi systemd[1]: plexmediaserver.service: Unit entered failed state.
Nov 03 15:04:04 raspberrypi systemd[1]: plexmediaserver.service: Main process exited, code=exited, status=255/n/a
Nov 03 15:04:03 raspberrypi systemd[1]: Started Plex Media Server for Linux.
Nov 03 15:04:03 raspberrypi systemd[1]: Starting Plex Media Server for Linux...
Nov 03 15:04:03 raspberrypi systemd[1]: Stopped Plex Media Server for Linux.
Nov 03 15:04:03 raspberrypi systemd[1]: plexmediaserver.service: Service hold-off time over, scheduling restart.
Nov 03 15:03:58 raspberrypi systemd[1]: plexmediaserver.service: Failed with result 'exit-code'.
Nov 03 15:03:58 raspberrypi systemd[1]: plexmediaserver.service: Unit entered failed state.
Nov 03 15:03:58 raspberrypi systemd[1]: plexmediaserver.service: Main process exited, code=exited, status=255/n/a
Nov 03 15:03:57 raspberrypi systemd[1]: Started Plex Media Server for Linux.
Nov 03 15:03:57 raspberrypi systemd[1]: Starting Plex Media Server for Linux...

MichaIng, 2018-11-03:
@roelfoubert 
Okay this just shows the 5 repeating startup attempts from systemd 🤔.

We need some Plex internal log. If you can find this somewhere would be great, but I will also set it up and check.

MichaIng, 2018-11-03:
Okay works fine here. Logs were hard to find, but here it is:
`/var/lib/plexmediaserver/Library/Application\ Support/Plex\ Media\ Server/Logs/Plex\ Media\ Server.log`

This is the best start, I guess, but the same folder contains more logs for the sub processes. Check for this  for ERROR messages, e.g. `grep 'ERROR' /var/lib/plexmediaserver/Library/Application\ Support/Plex\ Media\ Server/Logs/*`
This then hopefully gives a hint.

roelfoubert, 2018-11-03:
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex DLNA Server.2.log:Mar 28, 2018 01:37:13.414 [0x75357140] ERROR - Couldn't rename file "/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Plug-in Support/Databases/com.plexapp.dlna-new.db-shm" to "/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Plug-in Support/Databases/com.plexapp.dlna.db-shm": No such file or directory
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex DLNA Server.2.log:Mar 28, 2018 01:37:13.414 [0x75357140] ERROR - Couldn't rename file "/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Plug-in Support/Databases/com.plexapp.dlna-new.db-wal" to "/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Plug-in Support/Databases/com.plexapp.dlna.db-wal": No such file or directory
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex Media Scanner Analysis.1.log:Jul 13, 2018 21:15:45.590 [0x73158400] ERROR - [FFMPEG] - Valid timecode frame rate must be specified. Minimum value is 1
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex Media Scanner Analysis.1.log:Jul 13, 2018 21:15:46.865 [0x73158400] ERROR - [FFMPEG] - Valid timecode frame rate must be specified. Minimum value is 1
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex Media Scanner Analysis.1.log:Jul 13, 2018 21:15:48.468 [0x73158400] ERROR - [FFMPEG] - Valid timecode frame rate must be specified. Minimum value is 1
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex Media Scanner Analysis.1.log:Jul 13, 2018 21:15:49.819 [0x73158400] ERROR - [FFMPEG] - Valid timecode frame rate must be specified. Minimum value is 1
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex Media Scanner Analysis.1.log:Jul 13, 2018 21:17:11.487 [0x73158400] ERROR - [FFMPEG] - Valid timecode frame rate must be specified. Minimum value is 1
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex Media Server.1.log:Jul 29, 2018 17:45:12.587 [0x732ff400] ERROR - Error issuing curl_easy_perform(handle): 7
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex Media Server.1.log:Jul 29, 2018 17:45:12.587 [0x732ff400] ERROR - Error parsing content.
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex Media Server.1.log:Jul 29, 2018 17:45:12.588 [0x732ff400] ERROR - Error parsing XML.
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex Media Server.1.log:Jul 29, 2018 17:45:17.443 [0x6daff400] ERROR - Network Service: Error in browser handle read: 125 (Operation canceled) socket=-1
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex Media Server.1.log:Jul 29, 2018 17:45:19.380 [0x6a8ff400] ERROR - QueryParser: Invalid field 'onlyTransient' found, ignoring.
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex Media Server.1.log:Jul 29, 2018 17:45:23.640 [0x698ff400] ERROR - [PlexRelay] kex protocol error: type 7 seq 11
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex Media Server.1.log:Jul 29, 2018 17:45:52.029 [0x706ff400] ERROR - Extras: Preroll file didn't exist or couldn't be read: ''.
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex Media Server.2.log:Jul 29, 2018 17:42:44.869 [0x731ff400] ERROR - Error issuing curl_easy_perform(handle): 7
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex Media Server.2.log:Jul 29, 2018 17:42:44.869 [0x731ff400] ERROR - Error parsing content.
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex Media Server.2.log:Jul 29, 2018 17:42:44.879 [0x731ff400] ERROR - Error parsing XML.
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex Media Server.2.log:Jul 29, 2018 17:42:52.219 [0x6d9ff400] ERROR - Network Service: Error in browser handle read: 125 (Operation canceled) socket=-1
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex Media Server.2.log:Jul 29, 2018 17:43:38.357 [0x6fcff400] ERROR - [PlexRelay] kex protocol error: type 7 seq 11
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex Media Server.3.log:Jul 29, 2018 16:35:26.302 [0x718ff400] ERROR - Error issuing curl_easy_perform(handle): 7
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex Media Server.3.log:Jul 29, 2018 16:35:26.305 [0x718ff400] ERROR - Error parsing content.
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex Media Server.3.log:Jul 29, 2018 16:35:26.305 [0x718ff400] ERROR - Error parsing XML.
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex Media Server.3.log:Jul 29, 2018 16:35:41.998 [0x6d2ff400] ERROR - Network Service: Error in browser handle read: 125 (Operation canceled) socket=-1
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex Media Server.3.log:Jul 29, 2018 16:36:18.372 [0x6d2ff400] ERROR - Network Service: Error in advertiser handle read: 125 (Operation canceled) socket=-1
[... cut ...]
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex Media Server.log:Jul 29, 2018 18:01:40.469 [0x706ff400] ERROR - Network Service: Error in browser handle read: 125 (Operation canceled) socket=-1
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex Media Server.log:Jul 29, 2018 18:02:06.502 [0x706ff400] ERROR - Network Service: Error in advertiser handle read: 125 (Operation canceled) socket=72
grep: /var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/PMS Plugin Logs: Is a directory

MichaIng, 2018-11-03:
@roelfoubert 
Okay last ERROR from end of Juli, so this should not related 🤔:
```
/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/Plex Media Server.log:Jul 29, 2018 18:02:06.502 [0x706ff400] ERROR - Network Service: Error in advertiser handle read: 125 (Operation canceled) socket=72
grep: /var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Logs/PMS Plugin Logs: Is a directory
```

Did the error actually appear after a `dietpi-update`, or do you remember something else that changed?

Also check the very last logs:
`tail /var/lib/plexmediaserver/Library/Application Support/Plex\ Media\ Server/Logs/Plex\ Media\ Server.log`

roelfoubert, 2018-11-03:
yesterday it all works today I had a problem. the only thing it does is auto apt-get upgrade so maybe a bad upgrade?

roelfoubert, 2018-11-03:
tail logs:
Jul 29, 2018 18:02:06.544 [0x745ab0d0] DEBUG - Captured session 14.
Jul 29, 2018 18:02:06.544 [0x745ab0d0] DEBUG - Captured session 15.
Jul 29, 2018 18:02:06.544 [0x745ab0d0] DEBUG - Captured session 16.
Jul 29, 2018 18:02:06.544 [0x745ab0d0] DEBUG - Captured session 17.
Jul 29, 2018 18:02:06.544 [0x745ab0d0] DEBUG - Captured session 18.
Jul 29, 2018 18:02:06.544 [0x745ab0d0] DEBUG - Captured session 19.
Jul 29, 2018 18:02:06.603 [0x706ff400] INFO - Network Service: Abandoning advertise socket, it was closed.
Jul 29, 2018 18:02:06.617 [0x745ab0d0] DEBUG - Database: Shutting down.
Jul 29, 2018 18:02:06.617 [0x745ab0d0] DEBUG - Captured session 0.
Jul 29, 2018 18:02:06.617 [0x745ab0d0] DEBUG - Captured session 1.

MichaIng, 2018-11-03:
@roelfoubert 
Possible. Which logging mode did you choose? If you don't use dietpi-ramlog, could you check which packages were updated during this upgrade step? `/var/log/apt/term.log`
In case of dietpi-ramlog with storing logs before clearing, you would find them in /root/ directory.

Strange, the last log entries are from Jul 29, but your system time is correct according to screen above 🤔.

Just checked out install code. Did you change the locale on your system? Just checked out install code and found that locale needs to be en_US.UTF-8 on ARM devices.
@Fourdee 
We just apply US locale on interactive installs: https://github.com/Fourdee/DietPi/blob/dev/dietpi/dietpi-software#L10286
- I guess we should always apply it? G_WHIP_MSG already turns into simple [ INFO ] when G_USER_INPUTS=0.

roelfoubert, 2018-11-03:
the timezone is correct
[image]

the log file:
root@raspberrypi:~# /var/log/apt/term.log
-bash: /var/log/apt/term.log: Permission denied

MichaIng, 2018-11-03:
@roelfoubert 
```
root@raspberrypi:~# /var/log/apt/term.log
-bash: /var/log/apt/term.log: Permission denied
```
This tries to execute the log file 😉
Use `cat /var/log/apt/term.log` to fully output to terminal. But when not using ramlog this can be veeery much.
I mostly use `nano /var/log/apt/term.log` which opens the log in the editor which allows scrolling up and down.

Hmm your locale is en_GB, but according to our install script en_US is needed. Could you try to change that? Sadly I can't test here, as I use a VM for testing and this locale is only required for ARM devices (which use the Plex repo instead of manual package install).

roelfoubert, 2018-11-03:
When I change to en_US it automatically changes to en_GB

Fourdee, 2018-11-03:
@MichaIng 

> We just apply US locale on interactive installs: https://github.com/Fourdee/DietPi/blob/dev/dietpi/dietpi-software#L10286
I guess we should always apply it?

Yep 👍, + ```G_WHIP_MSG 'Locale will be changed to en_US``` etc?

Fourdee, 2018-11-03:
@roelfoubert @MichaIng 

> When I change to en_US it automatically changes to en_GB

confirmed, looking into it now.

MichaIng, 2018-11-03:
@Fourdee 
Jep, I didn't check the reason for this, but if required then should not depend on interactive/noninteractive call.

roelfoubert, 2018-11-03:
I'm going to reinstall the pi on another micro sd card and restore a backup from before and see if it works

MichaIng, 2018-11-03:
@Fourdee @roelfoubert 
Ah, actually locale changes. Check via `locale`
Just the wrong is shown in dietpi-config, since script internally en_GB is applied. We need to check either `/etc/default/locale` or dietpi.txt to get correct value 😉.

roelfoubert, 2018-11-05:
After more testing it turned out to be a problem in the application support folder. When reinstalling plex and made a new server it works again

MichaIng, 2018-11-06:
@roelfoubert 
Thanks for feedback. So there is noting we need to change with our Plex install code, so we can close the issue right?
Visual issue with locale is solved in current Beta.

Fourdee, 2018-11-09:
> Visual issue with locale is solved in current Beta.

@MichaIng 

Yep, resolved in beta, completed.

closed 2018-11-09 by Fourdee