MichaIng/DietPi #2217: Unable to install Nextcloud

enolp, 2018-11-03:
#### Details:
- Date           | Sat  3 Nov 15:45:33 CET 2018
- Bug report     | N/A
- DietPi version | v6.17.12 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1145 SMP Fri Sep 21 15:38:35 BST 2018
- Distro         | stretch (index=4)
- Command        | ncc maintenance:install
- Exit code      | 1
- Software title | DietPi-Software

#### Steps to reproduce:

1. Open dietpi-software
2. Install Nextcloud

#### Expected behaviour:

- Instalation of Nextcloud without problems

#### Actual behaviour:

- Instalation fails at some point

#### Extra details:

- An USB stick is connected and set up as a drive
- Pihole, fail2ban, Kodi and PiVPN are already installed

#### Additional logs:
```
Log file contents:
An unhandled exception has been thrown:
Error: Class 'Memcached' not found in /var/www/nextcloud/config/config.php:1156
Stack trace:
#0 /var/www/nextcloud/lib/private/Config.php(214): include()
#1 /var/www/nextcloud/lib/private/Config.php(63): OC\Config->readData()
#2 /var/www/nextcloud/lib/base.php(140): OC\Config->__construct('/var/www/nextcl...')
#3 /var/www/nextcloud/lib/base.php(591): OC::initPaths()
#4 /var/www/nextcloud/lib/base.php(1068): OC::init()
#5 /var/www/nextcloud/console.php(46): require_once('/var/www/nextcl...')
#6 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#7 {main}
```

enolp, 2018-11-03:
I checked the changelog from dev branch and this bug is solved. Sorry for the inconveniencie.

MichaIng, 2018-11-03:
@enolp 
No worries 😉. Check existing (closed) issues about it to fix this on your system.

Fix is also already merged to beta branch.

enolp, 2018-11-03:
Well @MichaIng , I've changed to beta branch and I still unable to install Nextcloud. I reopen the issue and copy what Dietpi shows me:

#### Details:
- Date           | Sat  3 Nov 17:03:16 CET 2018
- Bug report     | N/A
- DietPi version | v6.18.2 (Fourdee/beta)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1145 SMP Fri Sep 21 15:38:35 BST 2018
- Distro         | stretch (index=4)
- Command        | ncc maintenance:install
- Exit code      | 1
- Software title | DietPi-Software

#### Steps to reproduce:

1. Change to beta branch
2. Update Dietpi
3. Open dietpi-software
4. Install Nextcloud

#### Expected behaviour:

- Install Nextcloud without problems

#### Actual behaviour:

- Nextcloud installation fails at some point and it is unable to continue

#### Extra details:

- An USB stick is connected and set up as a drive
Pihole, fail2ban, Kodi and PiVPN are already installed

#### Additional logs:
```
Log file contents:

In Install.php line 121:
                                      
  Database <mysql> is not supported.  
                                      

maintenance:install [--database DATABASE] [--database-name DATABASE-NAME] [--database-host DATABASE-HOST] [--database-port DATABASE-PORT] [--database-user DATABASE-USER] [--database-pass [DATABASE-PASS]] [--database-table-prefix [DATABASE-TABLE-PREFIX]] [--database-table-space [DATABASE-TABLE-SPACE]] [--admin-user ADMIN-USER] [--admin-pass ADMIN-PASS] [--data-dir DATA-DIR]
```

MichaIng, 2018-11-03:
@enolp 
Jep you need to remove the by faulty install script wrong placed config file, before reinstall/install does work again:
```
rm /var/www/nextcloud/config/config.php
rm /var/www/nextcloud/occ
dietpi-software install 114
```
Removing occ just prevents dietpi-services from printing errors about it. Although this should also be fixed in the beta.

enolp, 2018-11-03:
@MichaIng, your fix does not work and Dietpi shows me the same error. My second try was on a clean backup which I changed to beta branch (see 4th comment)

MichaIng, 2018-11-03:
@enolp 
Ah sorry didn't check your error log and made assumption too fast.

> Database <mysql> is not supported.

Did they change something about the database argument naming?? I will try to replicate!

enolp, 2018-11-03:
It is strange because Dietpi installs mariadb not mysql so I don't know whats happening

MichaIng, 2018-11-03:
@MichaIng 
Jep, on many distros, MySQL is just an alias for an actual MariaDB install. Since both are mostly compatible most software titles just use "mysql" which does include MariaDB.

MichaIng, 2018-11-03:
@enolp 
Hmm very strange, install works very well here: `[  OK  ] DietPi-Software | ncc maintenance:install`

This is a fresh install of Nextcloud on your system, right? Could you try to remove all Nextcloud related and do a very fresh install attempt:

```
rm -R /var/www/nextcloud
rm -R /mnt/dietpi_userdata/nextcloud_data
mysqladmin drop nextcloud -f
mysql -e "drop user oc_admin@localhost"
mysql -e "drop user oc_admin"
```

Then retry install.

enolp, 2018-11-03:
The second command throws me an error: 
mysqladmin: DROP DATABASE nextcloud failed;
error: 'Can't drop database 'nextcloud'; database doesn't exist'

When I install or update software I always make a dd backup of my SD in case something goes bad. Something I noted is that when you install PiHole, Dietpi installs mysql. Probably there is some conflic there.

MichaIng, 2018-11-03:
@enolp 
Jep expected that all of the above commands succeed/are required, depending on how far the actual installation went. But this is failsafe then, that every old, perhaps faulty trace of Nextcloud is purged before attempting fresh install.

> When I install or update software I always make a dd backup of my SD in case something goes bad. 

Best practice 👍 

> Something I noted is that when you install PiHole, Dietpi installs mysql. Probably there is some conflict there.

??? MariaDB is installed with Pi-hole? Should be not the case, also was not during my tests dome days ago. SQLite should be installed instead, which is what Pi-hole uses.
But even if it was, MariaDB/MySQL is a database server and can handle multiple software clients at once 😉.

enolp, 2018-11-03:
> Jep expected that all of the above commands succeed/are required, depending on how far the actual installation went. But this is failsafe then, that every old, perhaps faulty trace of Nextcloud is purged before attempting fresh install.

Was the first time I installed Nextcloud on this installation and two times (using dd backup) on a clean environment. I don't know what's hapening

> ??? MariaDB is installed with Pi-hole? Should be not the case, also was not during my tests dome days ago. SQLite should be installed instead, which is what Pi-hole uses.
But even if it was, MariaDB/MySQL is a database server and can handle multiple software clients at once

No, Dietpi installs "LLSP" sqlite (sorry for saying mysql) when installing Pi-Hole and Nextcloud installs LLMP (I think). I don't know if it makes conflict or what

MichaIng, 2018-11-03:
@enolp 
> No, Dietpi installs "LLSP" sqlite (sorry for saying mysql) when installing Pi-Hole and Nextcloud installs LLMP (I think). I don't know if it makes conflict or what

Okay, jep as it should. Nope no conflicts by this. They share the webserver + PHP and use different databases.

So if you recovered a backup where no Nextcloud install was attempted yet, then
`sed -i "\|cp '/var/www/nextcloud/config/config.sample.php'|d" /DietPi/dietpi/dietpi-software`
should be the only step required before installing Nextcloud.

If you already run into the error in OT, then do the steps above to remove all traces, before trying the reinstall.

If the `Database <mysql> is not supported.` error still occurs, then please send a `dietpi-bugreport`, as I never faced this and would need some deeper investigation.

enolp, 2018-11-03:
I restored my previous backup and then run the sed command but it doesn't work, so i uploaded a bug report: f489c9d1-7964-4348-b501-c2de9bd420e1

MichaIng, 2018-11-03:
@enolp 
Did you try what I mentioned above: `apt-get install -y php-mysql` then rerun installation.

MichaIng, 2018-11-03:
@enolp 
Jep sorry that I didn't think about this earlier and that's for your effort 👍.

enolp, 2018-11-03:
Ok, backup restored, branch changed to beta, php-mysql installed before installing Nextcloud, Nextcloud installed and all is working as expected. Nice!!!

Thanks for the support and the great job.

MichaIng, 2018-11-04:
@enolp 
Great, thanks for testing!

closed 2018-11-04 by MichaIng