MichaIng/DietPi #2218: RPi.GPIO | Broken by current kernel release (APT)?

MichaIng, 2018-11-03:
Ref: https://dietpi.com/phpbb/viewtopic.php?f=9&t=5198
- [ ] Replicate
- [ ] Check if current master resolves it: `rpi-update`
- [ ] Hope for fast package/release update 😉.

MichaIng, 2018-11-15:
@Fourdee 
Found your answer on the forum. Have you been able to replicate, is it expected on current kernel/misconfiguration?

Fourdee, 2018-11-15:
@MichaIng 

Will re-test, moved to milestone.

Fourdee, 2018-11-16:
RPi install test:
- 🈯️ 
```
root@DietPi:~# uname -a
Linux DietPi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux
```

Unable to test user specific script with reported issue, as, lacking the script itself.

Waiting for user response:
https://dietpi.com/phpbb/viewtopic.php?f=9&t=5198&p=15398#p15398

Fourdee, 2018-11-25:
No response from user, will mark as closed. Will reopen if user replies.

closed 2018-11-25 by Fourdee