MichaIng/DietPi #2219: DietPi-Software | Mono: Current version fails on ARM

MichaIng, 2018-11-03:
Forum ref: https://dietpi.com/phpbb/viewtopic.php?f=11&t=5206
Related Sonarr issue: https://github.com/Sonarr/Sonarr/issues/2769
Related Mono issue: https://github.com/mono/mono/issues/11095

Current workaround is to add `-O=-aot` option to mono call, e.g.:
`/usr/bin/mono -O=-aot /opt/NzbDrone/NzbDrone.exe -nobrowser -data=/mnt/dietpi_userdata/sonarr`

Fourdee, 2018-11-04:
ARMv8:
- 🈯️ With -O=-aot
- 🈯️ Without -O=-aot

```
root@DietPi:~# mono --version
Mono JIT compiler version 5.16.0.179 (tarball Thu Oct  4 12:05:24 UTC 2018)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
        TLS:           __thread
        SIGSEGV:       normal
        Notifications: epoll
        Architecture:  armel,vfp+hard
        Disabled:      none
        Misc:          softdebug
        Interpreter:   yes
        LLVM:          yes(3.6.0svn-mono-/)
        GC:            sgen (concurrent by default)
```

Fourdee, 2018-11-04:
aot: https://www.mono-project.com/docs/advanced/runtime/docs/aot/

Fourdee, 2018-11-04:
Interesting:
🈯️ ```-O=-aot ```
🈴 ```--aot```
```
root@DietPi:~# /usr/bin/mono --aot -O=all /opt/jackett/JackettConsole.exe
Mono Ahead of Time compiler - compiling assembly /opt/jackett/JackettConsole.exe
AOTID E4A8AB7F-467D-A405-C2A6-BE94BA6BB23F
Code: 108772(78%) Info: 2896(2%) Ex Info: 8396(6%) Unwind Info: 3743(2%) Class Info: 5542(3%) PLT: 433(0%) GOT Info: 7908(5%) Offsets: 1829(1%) GOT: 4896
Compiled: 223/223 (100%), No GOT slots: 71 (31%), Direct calls: 0 (0%)
Executing the native assembler: "as"   -mfpu=vfp3 -o /tmp/mono_aot_ELtFYk.o /tmp/mono_aot_ELtFYk
Executing the native linker: gcc --shared -o /opt/jackett/JackettConsole.exe.so.tmp  /tmp/mono_aot_ELtFYk.o
Stripping the binary: "strip" --strip-symbol=\$a --strip-symbol=\$d /opt/jackett/JackettConsole.exe.so.tmp
JIT time: 191 ms, Generation time: 63 ms, Assembly+Link time: 263 ms.
```

Fourdee, 2018-11-04:
Ok, so can't replicate on ARMv7, only difference is time of tarball
```
Mono JIT compiler version 5.16.0.179 (tarball Thu Oct 4 12:22:11 UTC 2018)
Mono JIT compiler version 5.16.0.179 (tarball Thu Oct  4 12:05:24 UTC 2018)
```

Generator, 2018-11-10:
Tested on RPi3 with Radarr 

Without `-O=-aot`
```
$ sudo -u radarr mono /opt/Radarr/Radarr.exe -nobrowser -data=/mnt/Storage/Dietpi/var/Radarr

Unhandled Exception:
System.TypeInitializationException: The type initializer for 'NzbDrone.Console.ConsoleApp' threw an exception. ---> System.TypeInitializationException: The type initializer for 'NzbDrone.Common.Instrumentation.NzbDroneLogger' threw an exception. ---> System.TypeInitializationException: The type initializer for 'NLog.LogManager' threw an exception. ---> System.TypeInitializationException: The type initializer for 'NLog.LogFactory' threw an exception. ---> System.OverflowException: TimeSpan overflowed because the duration is too long.
  at System.TimeSpan.Interval (System.Double value, System.Int32 scale) <0x74a03bb8 + 0x0010c> in <c3c5f4bb011a4af8b925b0d39ee12396#b6d93984cc5312a550b66bac10e075f9>:0
  at System.TimeSpan.FromSeconds (System.Double value) <0x74a03e88 + 0x0001b> in <c3c5f4bb011a4af8b925b0d39ee12396#b6d93984cc5312a550b66bac10e075f9>:0
  at NLog.LogFactory..cctor () [0x00000] in <a273cb27438940d6830e58cd57866c35>:0
   --- End of inner exception stack trace ---
  at (wrapper managed-to-native) System.Object.__icall_wrapper_mono_generic_class_init(intptr)
  at NLog.LogManager..cctor () [0x00000] in <a273cb27438940d6830e58cd57866c35>:0
   --- End of inner exception stack trace ---
  at NzbDrone.Common.Instrumentation.NzbDroneLogger..cctor () [0x00005] in <445f1ed19d7d40fc8d1f35855cc618bc>:0
   --- End of inner exception stack trace ---
  at NzbDrone.Console.ConsoleApp..cctor () [0x00000] in <b6e6e83031db400582b8bc076d3bc359>:0
   --- End of inner exception stack trace ---
[ERROR] FATAL UNHANDLED EXCEPTION: System.TypeInitializationException: The type initializer for 'NzbDrone.Console.ConsoleApp' threw an exception. ---> System.TypeInitializationException: The type initializer for 'NzbDrone.Common.Instrumentation.NzbDroneLogger' threw an exception. ---> System.TypeInitializationException: The type initializer for 'NLog.LogManager' threw an exception. ---> System.TypeInitializationException: The type initializer for 'NLog.LogFactory' threw an exception. ---> System.OverflowException: TimeSpan overflowed because the duration is too long.
  at System.TimeSpan.Interval (System.Double value, System.Int32 scale) <0x74a03bb8 + 0x0010c> in <c3c5f4bb011a4af8b925b0d39ee12396#b6d93984cc5312a550b66bac10e075f9>:0
  at System.TimeSpan.FromSeconds (System.Double value) <0x74a03e88 + 0x0001b> in <c3c5f4bb011a4af8b925b0d39ee12396#b6d93984cc5312a550b66bac10e075f9>:0
  at NLog.LogFactory..cctor () [0x00000] in <a273cb27438940d6830e58cd57866c35>:0
   --- End of inner exception stack trace ---
  at (wrapper managed-to-native) System.Object.__icall_wrapper_mono_generic_class_init(intptr)
  at NLog.LogManager..cctor () [0x00000] in <a273cb27438940d6830e58cd57866c35>:0
   --- End of inner exception stack trace ---
  at NzbDrone.Common.Instrumentation.NzbDroneLogger..cctor () [0x00005] in <445f1ed19d7d40fc8d1f35855cc618bc>:0
   --- End of inner exception stack trace ---
  at NzbDrone.Console.ConsoleApp..cctor () [0x00000] in <b6e6e83031db400582b8bc076d3bc359>:0
   --- End of inner exception stack trace ---
```
With `-O=-aot`
```
$ MONO_ENV_OPTIONS='-O=-aot'  sudo -u radarr mono /opt/Radarr/Radarr.exe -nobrowser -data=/mnt/Storage/Dietpi/var/Radarr
[Info] Bootstrap: Starting Radarr - /opt/Radarr/Radarr.exe - Version 0.2.0.1120
[Info] AppFolderInfo: Data directory is being overridden to [/mnt/Storage/Dietpi/var/Radarr]
[Info] Router: Application mode: Interactive
[Info] MigrationLogger: *** Migrating data source=/mnt/Storage/Dietpi/var/Radarr/nzbdrone.db;cache size=-10485760;datetimekind=Utc;journal mode=Wal;pooling=True;version=3 ***
[Info] MigrationLogger: *** Migrating data source=/mnt/Storage/Dietpi/var/Radarr/logs.db;cache size=-10485760;datetimekind=Utc;journal mode=Wal;pooling=True;version=3 ***
[Info] OwinHostController: Listening on the following URLs:
[Info] OwinHostController:   http://*:7878/radarr/
[Info] OwinHostController:   http://*:7878/
[Info] NancyBootstrapper: Starting NzbDrone API
```
**Software version:**
Mono
```
Mono JIT compiler version 5.16.0.179 (tarball Thu Oct  4 12:22:11 UTC 2018)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
        TLS:           __thread
        SIGSEGV:       normal
        Notifications: epoll
        Architecture:  armel,vfp+hard
        Disabled:      none
        Misc:          softdebug
        Interpreter:   yes
        LLVM:          yes(3.6.0svn-mono-/)
        GC:            sgen (concurrent by default)
```
Dietpi
```
$ cat /DietPi/dietpi/.version
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=17
G_DIETPI_VERSION_RC=12
G_GITBRANCH=master
G_GITOWNER=Fourdee
```

Fourdee, 2018-11-11:
@Generator 

Thanks for the testing + report 👍 

Ok, clearly we need to roll out ```-O=-aot```. I'll make the change to all Mono applications + testing.

```
dietpi-software install 106 144 145 147
```

Fourdee, 2018-11-11:
Latest release link for Radarr not functional:
[image]

https://api.github.com/repos/Radarr/Radarr/releases/latest
```
{
  "message": "Not Found",
  "documentation_url": "https://developer.github.com/v3/repos/releases/#get-the-latest-release"
}
```

Fourdee, 2018-11-11:
Retest:
```
dietpi-software install 106 144 145 147
```
- 🈯️ ARMv7
- 🈯️ ARMv8

Fourdee, 2018-11-11:
Fixed in Beta:
- https://github.com/Fourdee/DietPi/pull/2235

Completed.

closed 2018-11-11 by Fourdee

MichaIng, 2018-11-11:
@Fourdee 
Whoopsie, my fault to not test this for all. `/latest` is not yet code within release, I added this some days ago with the jackett reinstall fix I think. Just found that to be handy and though it is part of the GitHub API to always work, whenever a release is available. Seems not, maybe the maintainer needs to manually mark a release as latest first or such 🤔. But yeah, `grep -m1` should be safe, since latest is always on the top 👍. I planned to use this in more cases where we currently download a certain release (and need to regularly update our URL for this). We have no "chance" to test new releases then before adding to our code, on the other hand there will be less user requests to update to current version. Also we download latest already in most other cases, where possible, via master branch and such.

msdos, 2018-11-20:
I'm using 6.17.2 and all applications that use Mono are broken.

msdos, 2018-11-20:
I used the workaround proposed here (and https://github.com/Fourdee/DietPi/pull/2235/files): edited all `/etc/systemd` service files related to sonarr, lidarr, jackett and radarr to call mono using `-O=-aot`.