MichaIng/DietPi #2220: DietPi-Software | Son/Rad/Lidarr: Create files/dirs with 664/775 permissions

MichaIng, 2018-11-03:
..to allow media software write access and file create permissions (e.g. subtitle downloads via Plex subzero plugin): https://github.com/Fourdee/DietPi/issues/2067#issuecomment-435624393

Hopefully this can be pre-configured, otherwise add info to online doc to change the settings accordingly.

Fourdee, 2018-11-15:
Note to self:
- Under Settings > Media Management > Show advanced > Permissions
[image]

Will try and find in cfg options

MichaIng, 2018-11-15:
@Fourdee 
Searched Sonarr wiki, code and forum for a complete config.xml documentation, but just found articles/issues/code snippets about single settings 🤔.
Generally the settings seem to be stored within the SQLite database and only a few basic settings can be pre-set via config.xml.

We could also try to identify the related database table, or check CLI for config related commands, which would be safer indeed.

Fourdee, 2018-11-18:
Ok, for now lets add this info the online docs:
- Sonarr
- Radarr
- Lidarr

arr arr arr 🤣

Fourdee, 2018-11-18:
Will mark as completed. Cant see a viable way to automate this in our setup currently.

closed 2018-11-18 by Fourdee