MichaIng/DietPi #2221: Nextcloud cannot be installed, again

enolp, 2018-11-04:
I'm here again. Trying to install Nextcloud with dietpi_userspace on another drive, throws me this error. To save time, I also send a bug report just in case: f489c9d1-7964-4348-b501-c2de9bd420e1

#### Details:
- Date           | Sun  4 Nov 12:37:10 CET 2018
- Bug report     | N/A
- DietPi version | v6.18.2 (Fourdee/beta)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1145 SMP Fri Sep 21 15:38:35 BST 2018
- Distro         | stretch (index=4)
- Command        | ncc maintenance:install
- Exit code      | 1
- Software title | DietPi-Software

#### Steps to reproduce:

1. Plug an USB storage
2. Make that USB a drive (formatting or mounting)
3. Point dietpi_userspace to it using dietpi-drive_manager
4. Open dietpi-software
5. Install Nextcloud

#### Expected behaviour:

- Nextcloud should be installed without problems using the external drive

#### Actual behaviour:

- Installation crashes, maybe by a lack of symlink support

#### Extra details:

- The external drive is a USB stick formatted to F2FS

#### Additional logs:
```
Log file contents:

In Local.php line 387:
                                     
  Following symlinks is not allowed  
                                     

maintenance:install [--database DATABASE] [--database-name DATABASE-NAME] [--database-host DATABASE-HOST] [--database-port DATABASE-PORT] [--database-user DATABASE-USER] [--database-pass [DATABASE-PASS]] [--database-table-prefix [DATABASE-TABLE-PREFIX]] [--database-table-space [DATABASE-TABLE-SPACE]] [--admin-user ADMIN-USER] [--admin-pass ADMIN-PASS] [--data-dir DATA-DIR]
```

Fourdee, 2018-11-04:
@enolp 

Thanks for the report 👍 

We have a known issue with Nextcloud install in v6.17 (https://github.com/Fourdee/DietPi/pull/2186), fixed in our beta:
https://github.com/Fourdee/DietPi/blob/beta/BRANCH_SYSTEM.md

F2FS "should" support symlinks:
https://unix.stackexchange.com/questions/323155/does-f2fs-support-permissions-and-symlinks/323156

We'll look into it.

Fourdee, 2018-11-04:
@enolp 

>  I forgot to mention that I was using the beta branch.

> DietPi version | v6.18.2 (Fourdee/beta)

You did mention it 👍 My error 😄

enolp, 2018-11-04:
Also my error because I didn't read my own bug report, just fill the field and clicked send xD

Fourdee, 2018-11-04:
@MichaIng 

PHP not being installed in dev branch.

[image]

```
root@DietPi:~# dietpi-software install 114
[  OK  ] DietPi-Software | RootFS R/W access verified.

[  OK  ] DietPi-Software | Initialized database
[  OK  ] DietPi-Software | Reading database

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Automated install

[  OK  ] DietPi-Software | Installing Nextcloud: A safe home for all your data
[  OK  ] DietPi-Software | Free space check: path=/ | available=11988 MB | requi      red=500 MB
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Software | Connection test: https://deb.debian.org/debian/
[  OK  ] NTPD: time sync | Completed
[ SUB1 ] DietPi-Services > unmask
[  OK  ] DietPi-Services | unmask all: cron
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Update & upgrade APT

[  OK  ] DietPi-Software | APT update, please wait...
Hit:1 http://fuzon.co.uk/meveric all InRelease
Hit:2 http://fuzon.co.uk/meveric stretch InRelease
Ign:3 https://cdn-aws.deb.debian.org/debian stretch InRelease
Hit:4 https://cdn-aws.deb.debian.org/debian stretch-updates InRelease
Hit:5 https://cdn-aws.deb.debian.org/debian-security stretch/updates InRelease
Hit:6 https://cdn-aws.deb.debian.org/debian stretch-backports InRelease
Hit:7 https://cdn-aws.deb.debian.org/debian stretch Release
Reading package lists...

[  OK  ] DietPi-Software | G_AGUP
[ INFO ] DietPi-Software | Running apt simulation to check for errors, please wa      it...
[  OK  ] DietPi-Software | APT installation for: bash-doc -s, please wait...
Inst bash-doc (4.4-5 Debian:9.5/stable [all])
Conf bash-doc (4.4-5 Debian:9.5/stable [all])

[  OK  ] DietPi-Software | G_AGI: bash-doc -s
[  OK  ] DietPi-Software | APT upgrade, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

[  OK  ] DietPi-Software | G_AGUG

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Checking for prerequisite software

[ INFO ] DietPi-Software | Redis will be installed
[ INFO ] DietPi-Software | Lighttpd will be installed
[ INFO ] DietPi-Software | MariaDB will be installed
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron

 DietPi-Software
[... 261 lines cut ...]
 Mode: Configuring MariaDB: database

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Configuring Redis: optional non-sql database store

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Configuring Nextcloud: A safe home for all your data

[ INFO ] DietPi-Software | Enabling needed PHP modules.
[ INFO ] DietPi-Software | Enabling APCu memory cache for PHP command line usage (CLI) as well, including Nextcloud ncc command and cron jobs.
[ INFO ] DietPi-Software | Enabling and configuring OPcache for Nextcloud.
[ INFO ] DietPi-Software | Lighttpd webserver found, enabling Nextcloud specific configuration.
[  OK  ] DietPi-Software | Added setting        "mod_setenv", to /etc/lighttpd/lighttpd.conf after line       "mod_access",
[  OK  ] DietPi-Software | cp -f /DietPi/dietpi/conf/lighttpd.nextcloud.conf /etc/lighttpd/conf-available/99-dietpi-nextcloud.conf
Enabling dietpi-nextcloud: ok
Run "service lighttpd force-reload" to enable changes
Enabling dietpi-dav_redirect: ok
Run "service lighttpd force-reload" to enable changes

[ INFO ] DietPi-Software | Enable 4-byte support for MariaDB
[  OK  ] DietPi-Software | systemctl restart mariadb
[FAILED] DietPi-Software | ncc maintenance:install
```

Fourdee, 2018-11-04:
🈯️ C2:
```
root@DietPi:/mnt/ssd# ln -sf test /mnt/ssd/test2
root@DietPi:/mnt/ssd# ls -lha
total 28K
drwxr-xr-x 2 root root 4.0K Nov  4 13:07 .
drwxr-xr-x 8 root root 4.0K Nov  4 13:06 ..
-rw-r--r-- 1 root root    2 Nov  4 13:07 test
lrwxrwxrwx 1 root root    4 Nov  4 13:07 test2 -> test
```

---

Issue is OC/NC disabled support for symlinked data dir due to "security" concerns:
https://github.com/nextcloud/server/issues/1257#issuecomment-245314285

However, as DietPi controls, manages and installs to symlink only if userdata is located on external drive as per user requirements, and ```www-data``` permissions correctly set on the NC/OC data dir, there is no security concern and we will re-enable this feature.

Fourdee, 2018-11-04:
@enolp 

Fixed in Beta update:
```
dietpi-update
```

closed 2018-11-04 by Fourdee

MichaIng, 2018-11-04:
@Fourdee
Jep I fixed the missing PHP in the open PR, recognized as well. Will compare solutions later and make it ready for merge.

The same is for desktops and xserver.

Fourdee, 2018-11-04:
@MichaIng 

> And you need to revert the symlink fix.

Fails without it 😕 

[image]

MichaIng, 2018-11-04:
@Fourdee
Strange, symbolic links inside the data dir are not supported or let's say not followed. But symbolic links of the data directory itself or it's parents are definitely supported, as this is a common method to move it after install, part of my how-to on Nextcloud forum and so many users following and reporting back it's working, including my own tests not long ago. If it's not working anymore then this must be a very recent change and Nextcloud forum should be flooded with related issues, which is not. I will tested, there must be another issue.

MichaIng, 2018-11-04:
Testing:
Okay, ncc maintenance:install throws an error, but if this is ignored, the instance works fine, uploading new files work and they are placed as expected etc. config.php is fully configured with /mnt/dietpi_userdata/nextcloud_dada with dietpi_userdata being a symlink.
What is missing, is the skeleton directory, so data dir will be empty then. But actually this is anyway how I like it more...

So we could revert to ignoring ncc maintenance:install error code. Not ideal though. I think I open a Nextcloud GitHub issue. Also the comment within the settings code and the 2016 commit/issue clearly states that it is about symlink INSIDE data dir pointing OUTSIDE of data dir. But this is not the case if data dir itself or even parent folder is a symlink. This is handled incorrectly.

MichaIng, 2018-11-04:
@Fourdee 
Reworked the workaround: https://github.com/Fourdee/DietPi/commit/c35b4bbbdf2dbc82942fceca03a2aebeef5beccd
- This preserves Nextcloud integrity and also the reasonable security feature, if a symlink is placed inside the data dir. Otherwise external users could upload a symlink, pointing to sensitive system data somewhere else to have at least read access in most cases!
- Now check for Stretch+ (no issue on Jessie) + datadir path containing a symlink, then skip error handling and manually move datadir.

Retesting:
- [ ] Stretch
- [ ] Jessie

Fourdee, 2018-11-05:
@MichaIng 

Legend 👍 Thank you.