MichaIng/DietPi #2225: Nano Pi Fire 3 - unable to boot

okipol88, 2018-11-06:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| N/A
- Distro version	|  N/A 
- Kernel version	|  N/A
- SBC device		|  Nano Pi Fire 3
- Power supply used	|  5V 3.1 A
- SDcard used		|  SanDisk Ultra 16 GB

#### Additional Information (if applicable):
- Flashed using Etcher 

#### Steps to reproduce:

1. Flash the image in [Diet Pi Image Download Link](https://dietpi.com/downloads/images/DietPi_NanoPiM3-ARMv7-Stretch.7z) to SD card
2. Run on device

#### Expected behaviour:

- FriendlyArm Boot screen appears.
- Diet Pi boots.

#### Actual behaviour:

- FriendlyArm Boot screen appears .
- Black screen appears.
- DietPi does not boot.

#### Extra details:

- Caps lock does not work on the keyboard (no input to switch on the CapsLock LED indicator).
- Using DELL U2414H monitor
- Works fine with Armbian (not SBC fault)

Fourdee, 2018-11-06:
@okipol88 

Thanks for the report, i'll try to replicate.

Please check the PCB for any revision codes, and, list them here if possible.

Fourdee, 2018-11-06:
Issue with tty1.

After the system is booted switch to tty2:
CTRL+ALT+F2

Fourdee, 2018-11-06:
Current:
```
Linux DietPi 4.4.49-s5p6818 #1 SMP PREEMPT Wed May 9 14:01:02 CST 2018 armv7l GNU/Linux
```

Tried the lastest image + dtb's from s5p6818-sd-friendlycore-xenial-4.4-armhf-20180906.img, same issue.

Fourdee, 2018-11-06:
REF: https://github.com/Fourdee/DietPi/issues/1926#issuecomment-405736808

Fourdee, 2018-11-06:
Works after boot, hangs system during boot.
```
systemctl start getty@tty1
chvt 1
```

MichaIng, 2018-11-06:
@Fourdee 
Perhaps we can hardcode TTY2 use when NanoPi Fire 3 is detected (on DietPi-PREP)?
It this `'62' ': NanoPi M3/T3/F3'` or requires new hardware ID (e.g. free 54)?

okipol88, 2018-11-06:
@Fourdee Nice ;)
Would be much appreciated to add this info tfor all the dummies (like me or only me ;() to the download section on [DietPI website](https://dietpi.com) -> Download section.
I Think i did encounter the [REF](https://github.com/Fourdee/DietPi/issues/1926#issuecomment-405736808) for a brief moment. I did not associate it with this :/.

Many thanks ;). Will try it out.

Fourdee, 2018-11-07:
@MichaIng 

Yep, we should be able to do during boot to switch tty2, i'll run some tests:
```
chvt 2
```

Fourdee, 2018-11-08:
🈯️ during boot ```chvt 2```

Issue is no hardware specific info in dmesg or ```lscpu``` that is unique.